kenjdavidson/react-native-bluetooth-classic #1: readUntil* functions are not completed

kenjdavidson, 2019-08-12:
Complete all the readUntil* methods with a decided protocol.  Right now some of the return data while some do not.  I believe that the data returned should be base64 (as thats the data written) but that will need to be decided.

wenchao1020, 2019-08-27:
I have no idea about IOS, but In android, readUntil* works well, the problem is the data send to react-native, in my way, I change the onData function
```javascript
/**
     * Handle read
     * @param data Message
     */
    void onData (String data) {
        mBuffer.append(data);
        String completeData = readUntil(this.delimiter);
        if (completeData != null && completeData.length() > 0) {
              WritableMap params = Arguments.createMap();
              WritableArray writableArray = Arguments.createArray();
              byte[] bytes = completeData.getBytes();
        
              for(int i=0;i<bytes.length;i++){
                  writableArray.pushInt((int)bytes[i]);
              }
              
              params.putArray("arr", writableArray);
             // real data, in debugger mode, you can read it
              params.putString("data", completeData); 
              sendEvent(DEVICE_READ, params);
          }
      }
```
and in js  
```javascript
BluetoothSerial.withDelimiter(delimiter).then(() => {
     BluetoothSerial.on('read', ({ arr }) => {
       let nData = Buffer.from(arr)
       let asciiData = Buffer.from(nData.toString(), 'ascii')
       console.log(asciiData.toString())     // will get the real string data
     })
})
```
I am not good in java, Hope this helps.

kenjdavidson, 2019-10-13:
Created **OnReadEventParamWriter** interface which is called on all OnRead Events in order to convert the BluetoothDevice and String contents into an appropriate WritableMap.  

    public WritableMap write(BluetoothDevice device, String contents);

The default functionality emits the parameter

    {
        device: <BluetoothDevice>,
        data: <String>,
        timestamp: <Date>
    }

When this goes in, the appropriate OnReadEventParamWriter would be 

    public WritableMap write(BluetoothDevice device, String contents) {
          WritableMap params = Arguments.createMap();
          WritableArray writableArray = Arguments.createArray();
          byte[] bytes = contents.getBytes();
        
          for(int i=0;i<bytes.length;i++){
              writableArray.pushInt((int)bytes[i]);
          }
              
          params.putArray("arr", writableArray);
          params.putString("data", completeData);
          return params;
    }

kenjdavidson, 2020-11-10:
Resolved(ish) with `release/1.0.x`

closed 2020-11-10 by kenjdavidson