kenjdavidson/react-native-bluetooth-classic #4: RN 0.60 support

EthanCannoy, 2019-08-29:
This project is quite new, but it's missing v0.60 support (specifically noticed a lack of pods support)? Seems kinda odd. Looks like it works up to 0.59.9, seeing as that's what is used in the example project package.json

kenjdavidson, 2019-09-05:
I'll be honest this is literally the first ios or apple development I've ever done. I had to Google around to see what pod support even did, since I have it all working in my main project I really didn't see the point. I do have a goal to build this out at some point, maybe this will be the first addition.

With regards to 0.60 it looked liked there were breaking changes so I'll probably hold off until I get a version released and my company app released. Then I can worry about upgrading.

Thanks for the info tho. Will Def look into it.

kenjdavidson, 2020-01-06:
Should be resolved with v0.10.1 (published).   @iamandiradu said that his pull request was tested and working with 0.60 podspec.

closed 2020-01-06 by kenjdavidson