kenjdavidson/react-native-bluetooth-classic #7: getConnectedDevice is not a function

bodtx, 2019-10-09:
When I use this:
`let device = await RNBluetoothClassic.getConnectedDevice();`
on android 7.1 it says:

`_reactNativeBluetoothClassic.default.getConnectedDevice is not a function`

whereas I use this, it works:
`await RNBluetoothClassic.disconnect();`

or

`this.connectToDevice(btBadgeurDevice).then(() =>  RNBluetoothClassic.write("XXXX\r\n")).then(() => RNBluetoothClassic.disconnect());`

any idea, did I miss a thing?

kenjdavidson, 2019-10-09:
Ya, because I'm not actually using that function I completely didn't realize it wasn't connected to the NativeModule.

kenjdavidson, 2019-10-09:
I just added the mapping from NativeModule to RNBluetoothClassic.

closed 2019-10-10 by kenjdavidson