kenjdavidson/react-native-bluetooth-classic #8: not found on npmjs.com

bodtx, 2019-10-10:
when I install:
`npm install react-native-bluetooth-classic --save`

I have

```
npm ERR! code E404
npm ERR! 404 Not Found: react-native-bluetooth-classic@latest
```

[image]

kenjdavidson, 2019-10-10:
Hey ya. It's not posted to npm yet. I was going to put up 0.9.1 but then there some issue found with it so I stopped. Not I'm trying to get pods working, which isn't really needed I guess.  I haven't been in a rush since me and the tow other ppl using it at the time we're still finding some issues.

kenjdavidson, 2019-10-10:
You'll have to use the local install for the time being. If you're fairly confident it's good, I haven't heard from the other two in a while, so I could probably post it this weekend.

kenjdavidson, 2019-10-10:
I also got in to github package registry beta and was debating publishing there.

AapoPitkanen, 2019-11-27:
Hi Ken!

Have you thought about posting this library to npm yet? I'm using it in a project and it is definitely an upgrade to react-native-bluetooth-serial as it also implements the iOS external accessory framework. Creating an actual build with the app with this library is currenty a bit difficult as I've yet to figure out how to add a local package to an automated build configuration such as with Microsoft App Center.

Haven't found any issues yet in implementing this to our project, it works a lot better than bluetooth-serial did. Keep up the good work!

kenjdavidson, 2019-11-27:
Yes. I am planning too. I'll be honest I've just been lazy and picky. I also wanted to get some things done before it's posted.  I'll try to get the current implementation to npm when we land and get settled (traveling at the moment). I was just super picky and didn't want to post something with a lot of bugs.

Thanks for the comment.  I appreciate it.

AapoPitkanen, 2019-11-27:
No worries, I totally understand. Great to hear it if you can get the current version posted, I'm looking forward to it 😄 👍

kenjdavidson, 2019-11-28:
Updated the current implementation as **v0.9.2**.  Let me know if you have any issues with it, I'm not at home so I don't have anything to really test it with, but I'm trusting that master is fully functionality (as per your previous comments).

closed 2019-11-28 by kenjdavidson