kenjdavidson/react-native-bluetooth-classic #12: Problem API isConnected()

wiseguysby, 2019-10-29:
I've got a problem with api isConnected(), error :
TypeError: _reactNativeBluetoothClassic.default.isConnected is not a function
TypeError: _reactNativeBluetoothClassic.default.isConnected is not a function

what's solution for this problem?

kenjdavidson, 2019-10-29:
The function just isn't connected in the javascript file, you can add the following line:

**this.isConnected = nativeModule.isConnected;**

to the following section of ./src/index.js

    this.requestEnable = nativeModule.requestEnable;
    this.isEnabled = nativeModule.isEnabled;
    this.list = nativeModule.list;
    this.connect = nativeModule.connect;
    this.disconnect = nativeModule.disconnect;
    this.writeToDevice = nativeModule.writeToDevice;

I just got my laptop back and will be able to continue testing and merging of the two branches I have on the go to resolve a number of these issues.

wiseguysby, 2019-10-30:
Ok, thanks for your support

closed 2019-10-30 by wiseguysby