kenjdavidson/react-native-bluetooth-classic #15: Running example app failed on iOS 12 at Xcode 11

iamcxa, 2019-11-08:
I clone the whole project from github,  runs `npm install` command at `BluetoothClassicExample` folder. Android parts runs perfectly but when I open XCode to runs the app on my iPhone firstly, I got these errors when the app just enter the splash screen:

[image]

After I try to resolve the problem, I got this issue and don't know why:

[image]

does anybody knows how to resolve these issues?

kenjdavidson, 2019-11-08:
The BluetoothClassicExample application needs to have the **UISupportedExternalAccessoryProtocols** configured in a file in your home directory.  Apple MFi program requires this for their devices, you cannot just use this library on IOS without having specific consent from the Apple MFi vendor and their codes.

I know the functionality isn't optimal, I have plans on moving the file from ~/BluetoothClassicExample.plist to somewhere in the project directory and providing an example, but for the time being it is what it is.  Or you can add the **UISupportedExternalAccessoryProtocols** directly to your own plist (making sure you don't commit that back to your own repository).

> **IOS**
> IOS requires that UISupportedExternalAccessoryProtocols are configured within the Info.plist file. As per the MFi these values should not be made available or stored within git unless working with the specific vendor. The IOS build has been updated to include a merging of during the Run Scripts phase:
> 
> Info.plist 
> ~/BluetoothClassicExample.plist
> You're responsible for creating this file, without the file your plist will not be updated with supported protocols and the device will not be available within IOS. You can forgo this and add the protocols directly to the development plist, but if pushed to git this goes against the terms of the Apple MFi program. The following file can be copied to ~/BluetoothClassicExample.plist and then will be merged.
> 
> <?xml version="1.0" encoding="UTF-8"?>
> <!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
> <plist version="1.0">
> <dict>
>   <key>UISupportedExternalAccessoryProtocols</key>
>   <array>
>     <string>com.apple.m1</string>
>   </array>
> </dict>
> </plist>
> 
> Willing to take suggestions on doing this in a better way: environment variables, etc.

With regards to your comment `After I try to resolve the problem`, how exactly did you attempt to resolve the issue?

kenjdavidson, 2019-11-08:
With regards to the `NSError` issue, I believe for new versions of XCode you need to actually use the full `NSError(domain,code,message)`.  I've already started updating all the `NSError` calls to include the full arguments, but I haven't had a chance to complete that work.

closed 2019-11-11 by kenjdavidson