kenjdavidson/react-native-bluetooth-classic #22: How to verify write message while connected to laptop bluetooth?

mjaydeep01, 2019-12-18:
I am trying to run the example given by connecting to laptop bluetooth.
Connection is successful, but when click on Send Data, nothing happens / received on the laptop.

Can you please guide, is this correct way?

kenjdavidson, 2019-12-18:
You'll need to have a ssp service on the computer that will print the incoming messages. All my testing is done with a Bluetooth device that either echos or replies with command results.  A quick good shows this link http://blog.bachi.net/?p=9700

Give it a shot. I don't know when or if I'll have time to look into it for you.

kenjdavidson, 2019-12-18:
Hey, so I just gave this a test and was able to confirm messages were being sent:

1) Turn on and Pair Android to Windows (note - due to the MFi requirement for Apple, I'm not sure if this is possible)

2) Configure your computer with an incoming bluetooth port.  Here is a SS of the final result, to get there just click +Add and then chose `Incoming` and your port will be created.
[image]

3) Download Putty
4) Start putty and connect to COM? (In my case this was COM4)

5) Start the application and connect to the desktop devices
6) Confirm that you're connected and type a message

[image]

kenjdavidson, 2019-12-18:
Note, that you may need to modify the encoding of library for your testing.  I have already started working on a method of customizing encoding through React Native (#9 ), but at this point it's not in the release.  By default the values are utf-8 (ios) and ISO-8859-1 (android).  You probably have to hardcode this to ASCII in your examples, or change the encoding on putty accordingly.

mjaydeep01, 2019-12-19:
Great !!
Thank you, @kenjdavidson.
I managed to follow above instruction and i did received the messages as well on putty.

Can i ask for one favor?  How to send files?

kenjdavidson, 2019-12-19:
That one's on you for now. You should be able to do it by repeatedly reading and sending the bytes of a file. You'll need to write the "server/client" portion to accept said bytes and save them. You'll have to setup your own control language for sending meta data. It might be better to write your own rn module to use the already build in Bluetooth file transfer apis. Note again this will be difficult on ios as you don't have the mfi protocols for Apple. And is therefore out of scope of this project.

closed 2019-12-19 by kenjdavidson