kenjdavidson/react-native-bluetooth-classic #23: Problem with pod install

aymensmurf, 2019-12-24:
I have a problem installing this module on iOS.

**npmPackages**
```json
    "react": "16.9.0",
    "react-native": "0.61.5",
    "react-native-bluetooth-classic": "^0.9.3"
```
**Commands**
```
cd ios
pod install
``` 
**Error**
```
[!] The `RNBluetoothClassic` pod failed to validate due to 1 error:
    - ERROR | attributes: Missing required attribute `homepage`.
    - WARN  | source: The version should be included in the Git tag.
```

kenjdavidson, 2019-12-24:
Hey ya, so I haven't actually updated or tested this using Pods (as per issue #4 ) states.  There have been some forks that have updated the `homepage` attribute, which I haven't tested nor seen any pull requests for.  I would suggest:
- Installing it manually 
- Updated it to work with Pods, and hooking a brother up so that others can partake in your knowledge.

kenjdavidson, 2020-01-06:
Should be resolved with @iamandiradu pull request.  I've added his podspec changes to both 0.10.1 (RN 0.60) and 0.9.6 (RN 0.41.2 which was the original dependency)

closed 2020-01-06 by kenjdavidson