kenjdavidson/react-native-bluetooth-classic #24: Can't read from a device

yatagan, 2019-12-28:
v0.9.3, Android

I can scan devices, connect and write to a device (and the recipient receives my messages),

but I can't read nither with:
* readFromDevice (returns "")
* readUntilDelimiter (returns nil)
* addListener(BTEvents.READ (never calls a callback)

available always returns 0

Other Bluetooth serial terminals can read on the same android from the same device.

Maybe I lost something?
Thanks!

yatagan, 2019-12-28:
"react-native": "0.59.5"

kenjdavidson, 2019-12-28:
Have you attempted to debug the native side? Can you confirm that the buffer is being filled with your scanned data.

As mentioned in issue #1 I don't use those functions in my own app. But they have been confirmed by other people.

You'll need to provide a bit more context and details to determine the issues.

tpettrov, 2020-01-07:
Hi, I can confirm that readFromDevice() is not working due to event being send in the onData method in RNBluetoothClassicModule.java . If you comment out this section from the method -> 

`/*while ((message = readUntil(this.mDelimiter)) != null) {
      BluetoothMessage bluetoothMessage
              = new BluetoothMessage<String>(deviceToWritableMap(mBluetoothService.connectedDevice()), message);
      sendEvent(BluetoothEvent.READ.code, bluetoothMessage.asMap());
    } */`

ReadFromDevice works as expected.

kenjdavidson, 2020-01-07:
Right on, thanks. I'll look at this tonight when I get home. I don't use the manual reading functions so I appreciate you taking a peak at it.

kenjdavidson, 2020-01-07:
This might actually need to be expanded a little more, the removeListener doesn't pass in the name of the listener which was removed, going to need to add in custom functionality across the board on READ so that a specific read flag is managed. 

I'll take another look at this when I'm home tonight.

tpettrov, 2020-01-07:
Yes I'm commenting out the whole while... I'm only pointing out the reason why ReadFromDevice() is not working in my case, otherwise you're right that events wouldn't be send. A good solution will be to check whether there are any subscriptions and only then to send the event. But my Java skills are close to zero and I can't help :/

kenjdavidson, 2020-01-07:
Right on man, I get it and appreciate what you provided, it definitely cut down on the amount of debugging I needed to do to get up and running.  You'll notice that the BluetoothExample doesn't use the read functions (only the READ event) which is why I never noticed it - 100% my bad.

(above comments) I've got some ideas on how to do this, I just need the time - hopefully before the weekend I'll have it resolved across the board.  Real life and day jobs and all.

kenjdavidson, 2020-01-07:
Out of curiosity, what is your use case for using the `read*` functions over the READ listener?  With the devices that I'm working with, I could see it for being something like:
- Getting configuration, where the message I sent expects a return of X delimiter lines

Which in most cases if you're working with a specific device you should know exactly what is coming back, and it should.  Although if something is sending an arbitrary number of lines for the same request then that could be a problem.

I'm just curious, the two things that I do are:
- Pretty much just sit and wait for data to come in
- Go get a specific set of data (config options) in which I know there are always 10 items returned, which makes it easy to just count out 10 in the listener after I make the request.

I'm assuming you're either polling on a loop or a ui event?

kenjdavidson, 2020-01-08:
Was able to do a bunch of refactoring on the Android side to bring it more inline with IOS.
- Removed the extra DeviceEmitter logic from Android, as it looked like I used old documentation when I did it and it seems that Android works well with NativeDeviceEmitter (as long as you don't provide a `_nativeModule`)
- Added a `kjd.reactnative.RCTEventEmitter` interface containing default functionality which is a one to one of the `RCTEventEmiter.m` on IOS.  this allows adding the `_nativeModule` without having to do a test for Android

BluetoothClassicExample was up and running after that, but didn't have a chance to test that the `READ` event still worked correctly, will do that this evening.  Once I'm sure that this all still works I can move on.   

- Update Android with `supportedEvents()` to mimic IOS
- Update RNBluetoothClassic javascript to override the functions of NativeDeviceEmitter and replace `addListener` and `Subscription.remove()` with wrapped calls to the `_nativeModule` so that if the READ event is requested or removed, the READ event is either (a) removed from the 'supportedEvents' set (b) a `readEventEnabled` flag is updated

tpettrov, 2020-01-08:
In answer to your question - yes I have random in length responses on same commands from my devices, and I prefer to follow the 'ask and receive' strategy pretty similar to basic API calls flow, because I plan to control the flow from saga. I don't want to read everything from the device whenever the device decides to respond and then decide what to do with the information. Downside is that I have to manually delay the read...but for now the strategy works :)

kenjdavidson, 2020-01-08:
Right on, how do you know when to stop polling? If you have random length responses for a command, and you're manually requesting the "API call" wouldn't it be possible to go one message to far?  If there is some method you're using, we can implement something like the `once()` function, something along the lines of a `sendAndReadUntil(command, untilFunction, handler)`?

Something to think about.   Once I get the Android version done, I can post it as a release candidate so you can give it a go before I start doing the IOS side of things.

kenjdavidson, 2020-01-09:
Added `setReadListening(boolean)` on the RNBluetoothClassic Native side.  It's updated each time:
- A listener is added
- All listeners are removed
- A subscription is removed

whenever there are no more READ listeners.   I updated the BluetoothClassicExample to perform polling on a 3 second interval, and validated the results are what I expected.  This is a pretty poor test, but for the time being its we have. (Todo - write tests!!)

    App.js:77 Checking for available data
    App.js:79 There are 46 bytes of data available

    App.js:82 Attempting to read the next message from the device
    App.js:85 985141001159976200108220633,Temp below range

    App.js:77 Checking for available data
    App.js:79 There are 0 bytes of data available

kenjdavidson, 2020-01-09:
Updated IOS with the same `setReadObserving` logic as Android.  Also made a few other fixes:
- Refactored `BluetoothRecievedDelegate` to `BluetoothDataReceivedDelegate` to fix spelling and better name
- Refactored the `BluetoothDataReceivedDelegate` function to `extension RNBluetoothClassic : BluetoothDataReceivedDelegate`

Requires testing when I get my ipod charged.

kenjdavidson, 2020-01-11:
Hey @tpettrov and @yatagan  I just published 0.9.5 and 0.10.2 with fixes to allow for polling.  It's not as optimal as I'd like it to be, but it gets the job done without needing to maintain a bunch of flags all over the place.  I wish the standard RCTEventEmitter accepted the EventName within the unsubscribe function, but it doesn't, so I went this route.

I've tested it on both IOS and Android using 0.9.5.  I haven't tested 0.10.2 (although I can't see why it wouldn't work) I just don't feel like updating the BluetoothClassicExample to React Native 0.60.0 for right now.

Let me know how it goes.

closed 2020-01-11 by kenjdavidson