kenjdavidson/react-native-bluetooth-classic #30: Review connection to multiple devices

kenjdavidson, 2020-01-16:
Review multiple devices on both Android and IOS.  Will require some changes to the current functionality:

1. BluetoothDevice (ios) and RNBluetoothService (android) will need to be massaged a little to ensure that they can be created multiple times by the module.  IOS is close to this (if not there) but Android needs to be modified to bring it more inline.
2. Module would need to update with the functionality to: manage a number of devices (Set most likely)
3. Javascript side should be updated to manage a BluetoothDevice object as well, which would be used for writing to.  A sample workflow would be:

```
let device1 = await RNBluetoothClassic.connect("address1");
device1.readUntilDeilmiter()
device1.read();
device1.write("Some data to write");
```

From what I saw (in IOS anyhow) there are ways to have multiple device emitters.  This may mean that each `BluetoothDevice` on javascript is also an emitter, although I'm not sure i saw anything on the native side that splits this well.  Otherwise, we could catch events and re-route them to their applicable `BluetoothDevice` since it should be known and maintained within the `RNBluetoothClassic` module.

Jose-Gallegos, 2020-07-26:
I need to read from multiple devices on Android. How can I contribute?

kenjdavidson, 2020-07-26:
I'm half way through converting the android version to allow for multiple devices and better configuration. I haven't pushed the branch yet though as it's still a rough cut. If you're interested I can push what I have and you can continue it.  Note that I had lumped a number of changes in with the branch, I just haven't had much time due to covid and my day job (and I don't do much extra programming in the summer). 

Or you can run with it yourself.  

But if ios isn't completed to match I won't accept a pull request. I don't want to have ios and Android maintained separately, it would make my whole purpose for this moot.

kenjdavidson, 2020-07-26:
There is a fork that has the ios side done I believe, although I don't like parts of it. I'd have to review both forks to ensure the api is 100% matching.

kenjdavidson, 2020-11-10:
This is currently available in `release/1.0.x` branch - serving as pre-release for `release/1.60.x`.

closed 2020-11-10 by kenjdavidson