kenjdavidson/react-native-bluetooth-classic #38: Importing always returns undefined

zameschua, 2020-02-02:
Hey there!

Firstly, thank you for creating and actively maintaining this package. I am very excited to use it!

Unfortunately, I am getting undefined whenever I try to import from `react-native-bluetooth-classic`.

# Code ran
```
import RNBluetoothClassic, {
  BTEvents,
  BTCharsets,
} from 'react-native-bluetooth-classic';

const MyComponent = () => {
  console.log(RNBluetoothClassic.isEnabled());
  return (
    <View />
  );
};
```

# Result 
```
Unhandled JS Exception: TypeError: Cannot read property 'isEnabled' of undefined
```

## Note that
1. Importing the file directly from `node_modules` also yields the same error
2. Directly `console.log`ging RNBluetoothClassic, BTEvents, and BTCharsets all yield `undefined`

# Versions
React native: 0.61.5
I also used `yarn add` instead of `npm install`, not sure if it makes a difference, but I can't use `npm install` for other reasons.

kenjdavidson, 2020-02-02:
Is the Bluetooth example app working?

zameschua, 2020-02-02:
Hold on lemme give it a try

zameschua, 2020-02-02:
I'm getting this error with the example app 
```
Fatal error: Unexpectedly found nil while unwrapping an Optional value: file /Users/zames/Development/react-native-bluetooth-classic/ios/RNBluetoothClassic.swift, line 53
2020-02-03 07:00:05.274862+0800 BluetoothClassicExample[356:42273] Fatal error: Unexpectedly found nil while unwrapping an Optional value: file /Users/zames/Development/react-native-bluetooth-classic/ios/RNBluetoothClassic.swift, line 53
```

I tried both `npm install` and `yarn add` this time and they're both giving the same errors.
The build succeeds but I only see a blank screen on the device (iPad) after the splash screen.

kenjdavidson, 2020-02-02:
Have you added in your mfi protocols for the device you're using? There is info on the example Readme on how to add it (I think).

Should be a **protocol-strings-example.plist** you need to use.

zameschua, 2020-02-02:
Hey thank you so much, adding the plist tag fixed the problem :)
(It's not in the readme, you should probably add it in too !)

Excited to try the package out, thank you again!

closed 2020-02-02 by zameschua

kenjdavidson, 2020-02-02:
Right on. I'll throw it into both Readmes next time I fire up the project.

hungdev, 2020-12-30:
thank for your asking, for who want to know what things we have to add:
```
	<key>UISupportedExternalAccessoryProtocols</key>
  <array>
    <string>com.apple.m1</string>
  </array>
	<key>NSBluetoothAlwaysUsageDescription</key>
	<string>Access bluetooth device over serial connection to document connection features</string>
```

kenjdavidson, 2020-12-30:
It's not in the readme, but it's literally the second thing on the document page

https://www.kenjdavidson.com/react-native-bluetooth-classic/ios/#mfi-protocols

Gra55h0pper, 2021-08-21:
Although my app supports both Android and iOS, I (for now) only support Bluetooth Classic on Android.
To prevent the "found nil" Fatal Error I currently have to define an empty array:
```
<key>UISupportedExternalAccessoryProtocols</key>
        <array>
        </array>
```
Would be nice if that wouldn't be required.

kenjdavidson, 2021-08-21:
It's a requirement for mfi.

You can use disable the autoload of native modules by updating your configuration, there is documentation on the react native site. Lookup autolaoding.

Otherwise I feel like adding an empty list is a satisfactory way to resolve the issue.

I'll update the docs.