kenjdavidson/react-native-bluetooth-classic #39: Cannot run build on iOS device – TypeError: null is not an object

AapoPitkanen, 2020-02-04:
Hi,

glad to see progress on the library, the Android version is working like a charm.

Unfortunately we've run to an issue running our application using this library on iOS, we're getting the following error when building an running the app on iOS 13.3:

[image]

Our Podfile is configured with the line "require_relative '../node_modules/@react-native-community/cli-platform-ios/native_modules'" and "use_native_modules!". 

Have you encountered the same error anywhere on development, or have you gotten the BTEvents working also with iOS on React Native version >= 0.60?

Thank you for the great library!

kenjdavidson, 2020-02-04:
Hey thanks.  Yes, the `BluetoothClassicExample` works perfectly (with my environment) on IOS.  The application is setup with the most up to date version of MacOSX, XCode and React Native 0.61.3 (I believe).

The most common issue with IOS is that the MFi protocols are missing in the PLIST file.  Without these, the module will build, but the IOS module will not actually start (the RNBluetoothClassic INIT method will fail).  I could probably clean it up a little to display some better messages, but MFi protocols are required for External Accessory, so they should be the first thing to check.

Let me know if you have issues with the `BluetoothClassicExample` as I always ensure it's working with my companies scanner before publishing.

kenjdavidson, 2020-02-04:
https://github.com/kenjdavidson/react-native-bluetooth-classic/tree/master/BluetoothClassicExample#ios-mfi-protocols

AapoPitkanen, 2020-02-05:
Thanks for the quick reply!

We actually got it working now, we had forgotten to add the empty swift file in Xcode. After adding the swift file, the bridging headers were automatically created by Xcode. We also cleaned our node_modules and installed all dependencies and pods again, after which the build started working.

closed 2020-02-05 by AapoPitkanen

kenjdavidson, 2020-02-05:
Awesome man. Ill have to create a Lil faq of these things at some point.  Thanks for the update/resolution.