kenjdavidson/react-native-bluetooth-classic #40: Can't established connection between two Android's devices

yeswell, 2020-02-07:
Hello!

I cloned the **_react-native-bluetooth-classic_** project and built **_BluetoothClassicExample_** for Android. The app was successfully installed on my Android device and I tried to connect it to my second Android device, but the connection failed with the message "Сonnection failed" and the code "EUNSPECIFIED".

If I connect to an iOS device, the connection is successfully established.

Before attempting to connect the devices, they were successfully paired together in the bluetooth menu in both cases.

Below I fully present the error I caught when trying to connect one Android device to another Android device.
```json
{
  "framesToPop": 1,
  "nativeStackAndroid": [
    {
      "methodName": "onConnectionFailed",
      "lineNumber": 740,
      "file": "RNBluetoothClassicModule.java",
      "class": "kjd.reactnative.bluetooth.RNBluetoothClassicModule"
    },
    {
      "methodName": "connectionFailed",
      "lineNumber": 230,
      "file": "RNBluetoothClassicService. java",
      "class": "kjd.reactnative.bluetooth.RNBluetoothClassicService"
    },
    {
      "methodName": "access$300",
      "lineNumber": 28,
      "file": "RNBluetoothClassicService.java",
      "class": "kjd.reactnative.bluetooth.RNBluetoothCMassicService"
    },
    {
      "methodName": "run",
      "lineNumber": 320,
      "file": "RNBluetoothClassicService.java",
      "class": "kjd.reactnative.bluetooth.RNBluetoothClassicServiceSConnectThread"
    }
  ],
  "userinfo": {
    "class": 524,
    "id": "9C:28:40:9F:90:08",
    "address": "9C:28:40:9F:90:08",
    "name": "SecondAndroidDevice"
  },
  "message": "Connection unsuccessful",
  "code": "EUNSPECIFIED",
  "line": 18,
  "column": 1111,
  "sourceURL": "index.android.bundle"
}
```
I looked carefully at the library code and found that the error is thrown because of a failed execution of [this line](https://github.com/kenjdavidson/react-native-bluetooth-classic/blob/1703787d53330797447624b4708f79553a261467/android/src/main/java/kjd/reactnative/bluetooth/RNBluetoothClassicService.java#L303). But I'm not good at Java and I don't understand why this is happening.

What am I doing wrong? How do I connect one Android device to another Android device correctly?

| Version | React Native | Android |
| --- | --- | --- |
| 0.10.5 | 0.61.5 | 8.1.0 |

kenjdavidson, 2020-02-07:
Did you have the other Android device setup and waiting for a connection?  The line you posted

`mmSocket.connect();` 

is the `BluetoothSocket` attempting to connect.  This requires that the other Android device has an active `BluetoothServerSocket` that is listening.  In order to test this you're going to need:

1) Two devices running BluetoothClassicExample
a) one device waiting for connection (click the button at the bottom)
b) one device attempting to connect to said connection

Can you confirm:
- That you had the other device waiting for the bluetooth connection?
- Which app were you using to open the bluetooth connection on the other device?

Without the other device waiting for a connection, you're going to get that error message.   The reason you don't get this on IOS is that IOS bluetooth is always accepting connections.  You don't need to place the app into a "listen" mode, as it's done by the system.   With that said, if you're actively sending data from IOS to Android, I'd love to see how your'e doing it - as it's not possible using this library (unless you magically have the Android MFi protocols for the device you're using).

Disclaimer - since this isn't the primary use case I can't guarantee when I'll get around to testing it.  This was provided by @tpettrov - I tested quickly using two **BluetoothClassicExample** apps running on separate devices and it seemed good.

tpettrov, 2020-02-08:
@yeswell You can use Serial Bluetooth Terminal from play store to test connections.

kenjdavidson, 2020-02-10:
Hey @yeswell, any update on your testing?

closed 2020-02-13 by kenjdavidson