kenjdavidson/react-native-bluetooth-classic #43: getConnectedDevice

AndriiBoiko, 2020-02-27:
await RNBluetoothClassic.getConnectedDevice(); - returned an error - No devices connected ((
But the device is already connected!
Device: iPad Air
```
index.tsx:274 Error: No bluetooth device connected
    at RNBluetoothClassic.fn [as getConnectedDevice] (http://192.168.0.107:8081/index.bundle?platform=ios&dev=true&minify=false:2165:36)
    at App.initialize$ (http://192.168.0.107:8081/index.bundle?platform=ios&dev=true&minify=false:133436:88)
    at tryCatch (http://192.168.0.107:8081/index.bundle?platform=ios&dev=true&minify=false:29290:19)
    at Generator.invoke [as _invoke] (http://192.168.0.107:8081/index.bundle?platform=ios&dev=true&minify=false:29465:24)
    at Generator.prototype.(anonymous function) [as next] (http://192.168.0.107:8081/index.bundle?platform=ios&dev=true&minify=false:29333:23)
    at tryCatch (http://192.168.0.107:8081/index.bundle?platform=ios&dev=true&minify=false:29290:19)
    at invoke (http://192.168.0.107:8081/index.bundle?platform=ios&dev=true&minify=false:29366:22)
    at http://192.168.0.107:8081/index.bundle?platform=ios&dev=true&minify=false:29376:15
    at tryCallOne (http://192.168.0.107:8081/index.bundle?platform=ios&dev=true&minify=false:3244:14)
    at http://192.168.0.107:8081/index.bundle?platform=ios&dev=true&minify=false:3345:17

index.tsx:279 initializeDevices::list
index.tsx:280 []
```

[image]

kenjdavidson, 2020-02-27:
Just a couple things here.

1) Do you have the MFI protocols setup properly
2) Although Apple says `Connected` this is just Apples version of `Paired`.  `getConnectedDevice` ONLY returns once you've performed a `connect(device.address)` and the RFCOMM connection is established.

I get that it's a terrible name for the function, since the Apple device says `Connected` on the screen. But it's an important distinction in how the module works.

You should get that device back with `list()` if the Mfi stuff is setup right.

AndriiBoiko, 2020-02-27:
```
          let devices = await RNBluetoothClassic.list();

          console.log('initializeDevices::list');
          console.log(devices);
```
Returned
```
index.tsx:279 initializeDevices::list
index.tsx:280 []
```

kenjdavidson, 2020-02-27:
And the MFI protocols?

kenjdavidson, 2020-02-27:
If you don't have MFi Protocols you will never get a device back on IOS.   This is a limitation of the External Accessory framework, not the module.

AndriiBoiko, 2020-02-27:
Thx for the answer.
Give me time - I'll check it

kenjdavidson, 2020-02-27:
Right on.

AndriiBoiko, 2020-02-27:
No, unfortunately
I added all that can found on the internet
[image]

AndriiBoiko, 2020-02-27:
As I understand correctly - they (Garmin) use BLE protocol in the new devices
But  this Garmin GLO 2 - uses something else

kenjdavidson, 2020-02-27:
Those are not standard protocol values, they should look like a fully qualified name `com.garmin.glo1` or something along those lines.   They aren't actually the protocol in terms of `RFCOMM`.   To give you a quick run down:

Garmin would need to register with apple for the MFI (money and NDAs)
They would pass this down to you (the GPS being 105 dollars means this is unlikely).  If you look at Barcode/Microchip scanners you'll notice that the MFi versions are substantially more expensive than the BLE versions.  

The MFI protocols are essentially a secret.  In order to get them you need to sign a bunch of NDAs.  They are like fight club, you can never talk about what they are exactly. 

Every version of your app you release, you now need the Vendor to validate that this app is using the correct protocols and they say that "yes this app and version is allowed to use us".  It's super annoying for bug releases and minor version changes on IOS.

AndriiBoiko, 2020-02-27:
))) it's terrible
Thank you, man! I thinking we should dig deeper

kenjdavidson, 2020-02-27:
Feel free to shoot me an email, I've learned way to much about IOS and bluetooth in the last year trying to get this working.

closed 2020-02-27 by kenjdavidson

zachariahtimothy, 2020-03-19:
@kenjdavidson are MFi codes generally available online, or is that something we must obtain from the manufacturer?

kenjdavidson, 2020-03-19:
I'm not an expert, but I'd say 99% from the manufacturer.  The only company I've seen RELEASE their MFi codes is Zebra (scanners/printers) but those are only for test, in order to release the app to the store you need to have the manufacturer sign off on each iteration of your app (name,version,reason for use).  The vendor we use has been fantastic in this regards, but for just playing around with bluetooth classic, it doesn't seem realistic.  

MFi compliant devices seem to be 3-4x the cost of non MFi compliant devices (with regards to Zebra anyhow).  It makes sense, since it costs these companies so much to go through the MFi processes, they need to recoup it somehow.

Again, I may be completely wrong - so take this with a grain of salt.

zachariahtimothy, 2020-03-19:
Thanks for the info @kenjdavidson! Our app is for enterprise / government use so they are used to paying $1500+ for RFID scanners. I reached out to a manufacturer so will see how that goes. I confirmed the device I am using works with iOS because I tested with SerialMagic Keys iOS app. 

Thanks for the shoutout on mustache! My wife hated it at first then when I shaved it she missed it and I have had it ever since. Haha!