kenjdavidson/react-native-bluetooth-classic #46: Second attempt of connection isn't established between two devices

DarkDracoon, 2020-04-01:
Good afternoon.

After trying to implement a connection between two devices - **Device A** <-> **Device B** - on the attempt of getting it to work under normal circumstances of connecting multiple times without closing the application, I've figured out that there is an issue, that seems, according to other libraries that also use Bluetooth, either normal or LE, that only happens on Android related to the RFCOMM port not being found by the **Device B** (destination) or appearing to be closed on the **Device A** (origin) device.

This issue only happens on the second and next attempts of establishing a connection, and doesn't happen on the first time using the application. It's happening on the example application and on the project application I'm developing.

Is there any solution pending?

The logcat attached to the issue is representing normal condition of the steps executed to test this issue under Example app.

1. Open the app.
2. Set the **Device B** to be Accepting connections.
3. Connect to the **Device B** from **Device A**.
4. Disconnect from both devices.
5. Set the **Device B** to be Accepting connections again.
6. Try to connect to the **Device B** from **Device A**.

If the app is closed and re-opened, it works back again.
 
[logcat.log](https://github.com/kenjdavidson/react-native-bluetooth-classic/files/4416622/logcat.log)

kenjdavidson, 2020-04-01:
Didn't know this was an issue until now. So at this point there is not solution being worked on. And with the current world issues I can't guarentee there will be, as I don't have any time for recreational development. 

If you can fix and pull request or point me to a specific bug I can do it quickly and release.

I tested this connecting from my phone to my accepting Mac. I don't have a second android device to test this, I can see if I can borrow one at some point.

DarkDracoon, 2020-04-02:
Good morning and thank you for your quick answer.
I don't know how to implement this in a proper way, however, by this solution and comparing to the object creation and exportation, what could work was the object creation outside the `src/index.js`, by having a class reference, and creating it where it is being imported to, and adding a method to destroy the instance or communication port that is being instanced natively.

The issue that has been created on a BLE library is mentioned below, however with workarounds adapted to the way BLE has this implemented.
https://github.com/Polidea/react-native-ble-plx/issues/501#issuecomment-527485202

kenjdavidson, 2020-04-02:
I played around with it last night using the latest `master` and was able to:
- Repeatedly connect to my Mac OSX and display messages using `screen /dev/cu.BuetoothIncomingPort` I was able to disconnect and re-connect without any changes to the library or the BluetoothClassicExample
- Repeatedly connect and disconnect from my companies microchip scanners without issue - without having to restart the application or re-create the React Native module (JS).

It looks like Polidea's destroy just drops all the connections, unregisters all the listeners and re-creates the RXBluetooth object.  Re-creating the object on JS is a little overkill, as it looks like it doesn't anything on initiation but re-create the Native service.  It should be doable by just ensuring that all the connections are closed properly.  

I can see some places where it may be possible for an exception to skip closing a socket, just odd that it's happening for you, but not me.  Could you let me know what exact versions you're working with?  I can see you're connection is using the `Fallback` from the connect method, which is in there as a backup for Android 4.*

That may be the difference, as my new phone uses the direct method to connect.

filipef101, 2020-04-02:
@kenjdavidson Extra info, for me it also happens with 2 android devices using the example app, Always happens, and crashes after trying to reconnect a few times.
Connect to a device, disconnect, connect again -> fails, one or two retries and crashes

DarkDracoon, 2020-04-02:
So that would be strange at happening with the example application also. I've also tested to connect repeatedly to my Mac either, and the messages show up normally, disconnecting and reconnecting consecutively.

My Android devices are:
- **Device A** - Redmi 7, running Android 9
- **Device B** - Mi A1, running Android 10

Then I've tested trying to connect other devices, and the issue is still the same:
- Samsung J7 2017, running Android 9
- Nexus 5X, running Android 8.1
- Qilive Q8S6IN4G, running Android 7

Everyone of them are running through Fallback connect method, could log that Qilive is also running in Fallback connection method, and they fail.

kenjdavidson, 2020-04-02:
@filipef101 a couple things:
- Are your devices using the `fallback` like @DarkDracoon is? Or is it using the actual Connect method? What happens if you DON'T connect to another Android device, does this happen when you're connecting to a regular device?

Are you saying that it seems to be happening ONLY when using the RNBluetoothModule.accept() method?  When you disconnect from the other device, does it need to be restarted?  For example:

Device1.connect()
Device2.accept()
// Connection established

disconnect

Device2 - restart
Device1.connect()
Device2.accept()
// Connection established

Or do you need to restart BOTH the connecting and accepting device?  Essentially, something I noticed is that I'm only looking at the ACTION_ACL_DISCONNECT_REQUEST in order to disconnect - which may not be called for ACCEPT based devices.   You should be able to debug that pretty easily if you have chance, make sure that receiver is firing and disconnecting/shutting down the socket correctly.

kenjdavidson, 2020-04-02:
@DarkDracoon thanks, that is starting to lead me to believe that using the Accept method causes something to happen during the disconnect.   Especially since when you connect to NON android devices (hardware and mac) this works properly.  (ie. Connecting as client works without issue)

Sadly, I still don't have a second Android device in order to try this. :(

kenjdavidson, 2020-04-02:
https://github.com/kenjdavidson/react-native-bluetooth-classic/blob/0353fd1078f7a3e50fdaf24f8fc7ee075bd4c9dc/android/src/main/java/kjd/reactnative/bluetooth/BluetoothConnectionReceiver.java#L23

If you throw a break point around this line, on the device which is performing the `.accept()` we can see if something is up with accepting connections and disconnections being performed differently.

filipef101, 2020-04-02:
@kenjdavidson I just tried the example app, from master codebase, and ran on two devices, I indeed need to close both "host" and "client" apps, for it to start working again. 

Let's call a device the "host" and another "client"

From the example what I do is, on host, press accept connections, then on the client connect to host,then press the X button(Doesn't wich devices closes it first, always the same)

Then on the Host, I press, again, accept connections, on the client I try to connect and it fails (little toast on the bottom)

kenjdavidson, 2020-04-02:
Right on, I'm working on borrowing another phone from someone.  I'm probably the only person in the world without two android devices (although I have 2 LGs and a tablet, but their batteries have completely died :( haha).

DarkDracoon, 2020-04-02:
> https://github.com/kenjdavidson/react-native-bluetooth-classic/blob/0353fd1078f7a3e50fdaf24f8fc7ee075bd4c9dc/android/src/main/java/kjd/reactnative/bluetooth/BluetoothConnectionReceiver.java#L23
>
> If [...]

Tomorrow I'll be able to debug this issue out, and then I'll report back here. Although, without accept, what's the possibility of getting it to work on trying to connect?

kenjdavidson, 2020-04-02:
There's an Android app called [Serial Bluetooth Terminal](https://play.google.com/store/apps/details?id=de.kai_morich.serial_bluetooth_terminal&hl=en) it's essentially what I ripped off for the BluetoothClassicExample app.  You can try connecting/accepting with that

BCE.accept
SBT.connect

disconnect

BCE.accept
SBT.connect
// Best case scenario is that this doesn't work, then we know its related to disconnect of the Accept socket.
then disconnect and try again.  It doesn't have the `accept` functionality though, it can only connect out.  But it will be a good test of whether the `accept` method is just not disconnecting properly, or whether it's a combination of the two.

DarkDracoon, 2020-04-03:
Hello Ken. Good morning.

I've tested the Serial Bluetooth Terminal with the Bluetooth Classic Example, and the issue is really persistent.

The app even shows that there is an connection still established.
[image]

kenjdavidson, 2020-04-03:
OK so the receiver isn't getting fired, which means that:
- `connIntent.addAction(BluetoothDevice.ACTION_ACL_DISCONNECT_REQUESTED);` isn't the ACTION being fired when disconnecting from the device.  Although we know this works when it's a NON android related device, so it seems unlikely that's the issue.  You can add the other action receivers to line https://github.com/kenjdavidson/react-native-bluetooth-classic/blob/0353fd1078f7a3e50fdaf24f8fc7ee075bd4c9dc/android/src/main/java/kjd/reactnative/bluetooth/RNBluetoothClassicModule.java#L868
```java
connIntent.addAction(BluetoothDevice.ACTION_ACL_DISCONNECTED);
```
to see if it is picked up.
- All the receiver does is emit the disconnection, this hopes that the disconnecting socket is managed on both sides by the appropriate `ConnectedThread`.  This is really unexpected since closing one side should throw an IOException on the other during the next read attempt.  After you've connected are you still hitting this line with a breakpoint:
https://github.com/kenjdavidson/react-native-bluetooth-classic/blob/0353fd1078f7a3e50fdaf24f8fc7ee075bd4c9dc/android/src/main/java/kjd/reactnative/bluetooth/RNBluetoothClassicService.java#L466

DarkDracoon, 2020-04-03:
If Device 1 gets to host anything, the device that connected to it stops being able to connect to anything, although Device 1 is able to connect to anything, but once it connects to another device, it stops being able to connect to anything else.

It means:
Device 1 <- Device 2
Device 2 stops being able to connect to anything else

Device 1 -> Device 3
Device 1 stops being able to connect to anything else

Device 3 -> Device 2
Device 3 stops being able to connect to anything else

kenjdavidson, 2020-04-03:
Tried this again with my custom scanner and the BluetoothClassicExample app.  I'm seeing some duplicate calls to `RNBluetoothClassicService.stop()` from the application, but at that point the `mConnectedThread` is already disconnected, so I don't see it being a problem.  I'm able to switch between my two devices:
- Microchip Scanner
- Barcode scanner

without issues.  But again, neither of these uses the `accept()` method to connect.  So the positive is that this seems to be working exactly as intended for how the library was initially planned.  I'd be interested in what happens to a non Android device which connects to the `accept()` method.  @tpettrov any comments, I assume this is working for you with a non Android device?

kenjdavidson, 2020-04-04:
Found the problem, the `mAcceptThread` wasn't being cleared when a new connection was made.  On a successfull connection we clean up and remove the `mConnectThread`, but we do not do the same with `mAcceptThread` when a connection is established.

Due to this, the next time an `accept()` is requested, the test on the Service level fails, but the exception is not bubbled up and handled correctly on the module.

All in all this isn't the best solution - but it'll work for now.   I'm completely re-designing for the next release `0.11.1` to allow for multiple devices.  I'll clean it up for that.

kenjdavidson, 2020-04-04:
Give `0.10.8` a go and let me know if it's any better.

DarkDracoon, 2020-04-06:
Good morning, hope you had a good weekend.

I'm giving it a shot, and I'll let you know whenever I have a result.

DarkDracoon, 2020-04-06:
It works properly under normal conditions, and didn't found for now any related connection issues.
Thank you very much and keep up with the good work 😃
I hope I can contribute more in the future.

closed 2020-04-06 by kenjdavidson