kenjdavidson/react-native-bluetooth-classic #48: isConnected doesnt work after app reload

Zlatanius, 2020-04-27:
**Mobile Device Environment**
Provide a list of operating systems on which this issue is relevant.
- Device: LG K11

**Application Environment**
Provide information about your development environment:
- React Native version: v0.61.5
- RN Bluetooth Classic version: v0.10.9

**Describe the bug**
Im using the HC-06 bluethooth module with arduino. Connecting to the device works no problem, but if I reload the app while still connected my phone stays connected, but the isConnected function returns flase.

**Expected behavior**
For isConnected to return true.

kenjdavidson, 2020-04-27:
What do you mean your phone stays connected?
- The RFCOMM socket stays open? This definitely should not be happening, it's definitely not happening using the two test devices that I've been using. 

I would expect `isConnected` to return false when the app closes and re-opens.  Since sockets cannot be serialized into a Bundle, there isn't any way to manage them across Activity lifecycle events. 

In my companies app, I save the last known connected device on shutdown, then on restart I check to see if that device is paired, then connect back to it automatically.

Zlatanius, 2020-04-27:
The HC-06 module has an indicator light that changes based on wether its connected or not, when I reload the light indicates that the module is connected. Im kind of a newbee when it comes to app development so I dont know how to check if the phone thinks its connected as well. So I gues this could be related to the module itself. Also worth noting is that I've experinced simular behaviour when using a BLE librirary and a different BLE module (AT-10).

kenjdavidson, 2020-04-27:
To be honest, I've never worked with the module, but I can state for sure that when working with the hardware I have the connection/disconnection works as intended on both the Android and the two devices.  I did a quick google about what the purpose of that LED is, and there isn't much information on it.

You need to remember that with regards to this library **isConnected** refers to an open RFCOMM socket.  It does NOT refer to either being paired or being in some state between paired and an open RFCOMM socket.  For example, if you go to your phone and in the bluetooth settings it says its paired AND connected, but there is no RFCOMM socket, then this is not actually connected.

This might be an over site from the original library, and definitely something I can look into fixing in version v0.11 (which will include multiple devices) but for the purpose of v0.10 you may need to (sadly/I'm sorry) work under those rules.

kenjdavidson, 2020-04-27:
http://www.martyncurrey.com/hc-05-and-hc-06-zs-040-bluetooth-modules-first-look/
http://www.martyncurrey.com/?wpdmdl=2436

It looks like the data sheet says the light is on:

**LED**
LED Status Indicator
Flashing
- Waiting to Connect/Pair
Steady ON
- Connected/Paired|

when it's connected OR paired?

Zlatanius, 2020-04-27:
Ok its not a huge problem as everyhing else works well. I gues Ill have do more research on RFCOMM socket to understand this better. Also Ive just done some more testing and this only happens when you reload the app, if you close it (completly not just go to home screen) it disconnect fine. Either way thank you very much for such qucik response.

Zlatanius, 2020-04-27:
> **LED**
> LED Status Indicator
> Flashing
> 
> * Waiting to Connect/Pair
> 
> Steady ON
> 
> * Connected/Paired|
> 
> when it's connected OR paired?

mine stays off and flashes twice every few seconds when its connected. It might not be an original but a copy that is branded as an original HC-06. It happens with these kind of thing so it might vary from the data sheet.

kenjdavidson, 2020-04-27:
Oh, you mean pause/resume not just kill it completely.  That may be an issue with the logic, although there might not be anything I can do.  `onPause` isn't closing the socket, but it could definitely be killing the service.  Meaning that the ReactNative Bluetooth module is shut down, but the underlying socket is remaining open.

There are possibly two options:
- Kill the whole thing on pause - and attempt to automatically reconnect when it's brought back up.
- Move the connection functionality to a service, allowing it to stay open when the activity is closed

I've never noticed this while testing my hardware, but should be easy enough to do. I'll see if I can test this specific case out with my hardware.

Zlatanius, 2020-04-27:
Honestly my knowladge on the subject of services and sockets isnt that great, so I would need to do more research here. From what I can tell the connection doesnt shut down completlly when I reaload the app (using the reload button in the development menu), but it does when I kill the app. I'll now go and research this subject and maybe I discever some more information.

kenjdavidson, 2020-04-27:
So this is specific to the React Native hot reload?  That will never happen in real life, so how big of a deal is this?

Zlatanius, 2020-05-04:
Hey sorry for not responding, I took a break from this project for a bit. Yeah i dont think it will ever happen in real life, but its not connected to hot reload. It happens when you do a regullar reload. Im not too sure how to test this properly, but from what I can tell it looks like the app stays "half connected" to the device. [This](https://stackoverflow.com/a/41457893/13281752) seems to be indicative of whats happening. Perhaps the bluethooth module disconnects but the phone doesnt.

kenjdavidson, 2020-05-04:
I don't understand what you mean by "regular reload".  If you're causing onPause() or onDestroy() to be called by switching or shutting down and then re-opening the application then the connection/socket will be closed.

The link you sent is specific to the React Native reload.

Zlatanius, 2020-05-07:
Im just refering to when you pres the reload buton in the development menu or press "r" in the metro bundler.

kenjdavidson, 2020-05-07:
So soemthing that will never happen in real life? I can't see myself worrying about this in the foreseeable future. As long as it works across pause/resume and destroy I think it's good.

closed 2020-05-07 by kenjdavidson