kenjdavidson/react-native-bluetooth-classic #50: How to get hex when READ ？

preflower, 2020-06-08:
how can i get hex when addListener BTEvent.READ

closed 2020-06-08 by preflower

sunilkumarkm, 2020-10-01:
Open the RNBluetoothClassicModule.java file and paste the below function to get hex from bytes array

private static final char[] HEX_ARRAY = "0123456789ABCDEF".toCharArray();
public static String bytesToHex(byte[] bytes) {
      char[] hexChars = new char[bytes.length * 2];
      for (int j = 0; j < bytes.length; j++) {
          int v = bytes[j] & 0xFF;
          hexChars[j * 2] = HEX_ARRAY[v >>> 4];
          hexChars[j * 2 + 1] = HEX_ARRAY[v & 0x0F];
      }
      return new String(hexChars);
  }

and goto onReceivedData method and replace the below line
mBuffer.append(new String(data, mCharset));   // comment this line
  
with (add these two lines)
String hexData = bytesToHex(data);
mBuffer.append(hexData);

kenjdavidson, 2020-10-01:
Thanks Sunil.  I'm working on getting more customizable `BluetoothConnection` objects available so that this doesn't need modification of the code.  Like when you create the package:

```java
RNBlutoothClassicPackage.builder().withConnection("HEX", new HexBluetoothConnectionFactory());
```