kenjdavidson/react-native-bluetooth-classic #51: How to send hex to device?

preflower, 2020-06-08:
[image]
i find write param type is String, but i need to send hex to device? how to do it?

preflower, 2020-06-08:
[image]
i find write that support hex, and i send Buffer but not response

kenjdavidson, 2020-06-08:
Yes. At this point in time the response is solely string. There was an issue opened early in the project where someone retired a byte array. You'd need to do the same thing as that yourself.

I have the plan to allow writing custom parsers in the bootstrapping, but don't have the time to complete it at this point.

preflower, 2020-06-10:
> Yes. At this point in time the response is solely string. There was an issue opened early in the project where someone returned a byte array. You'd need to do the same thing as that yourself.
> 
> I have the plan to allow writing custom parsers in the bootstrapping, but don't have the time to complete it at this point.
> 
> Since the result is a string, you should be able to convert it to hex representation yourself.

i use method of Self-Transmitting and Self-Receiving test; but i find the data received is inconsistent with the data sent;  the data is correct in writeToDevice function, but onReceivedData i get a incorrect data；can you help me check it?

preflower, 2020-06-10:
> i use method of Self-Transmitting and Self-Receiving test; but i find the data received is inconsistent with the data sent; the data is correct in writeToDevice function, but onReceivedData i get a [...]

send file:
```
let buf = new Buffer('3aa30000000000010002800183a8', 'hex');
RNBluetoothClassic.write(buf)
```
RNBluetoothClassicModule.java
```
@Override
  public void onReceivedData(BluetoothDevice device, byte[] data) {
    String msg = String.format("Received %d bytes from device %s", data.length, device.getName());
    Log.d(TAG, msg);

    mBuffer.append(new String(data, mCharset));

    if (!mReadObserving.get()) {
      Log.d(TAG, "No BTEvent.READ listeners are registered, skipping handling of the event");
      return;
    }

    char[] hexArray = "0123456789ABCDEF".toCharArray();
    char[] hexChars = new char[data.length * 2];
    for (int j = 0; j < data.length; j++) {
        int v = data[j] & 0xFF;
        hexChars[j * 2] = hexArray[v >>> 4];
        hexChars[j * 2 + 1] = hexArray[v & 0x0F];
    }

    String message = new String(hexChars);
    Log.d(TAG, 'Check Data:' + message);
    if (message != null) {
      BluetoothMessage bluetoothMessage
        = new BluetoothMessage<>(new NativeDevice(mBluetoothService.connectedDevice()).map(), message);
      sendEvent(BluetoothEvent.READ.code, bluetoothMessage.asMap());
    }
  }
```

preflower, 2020-06-10:
[image]
i find reason, bytes[] change UTF-8 type in write function, and this operation will result in the hex code inconsistency required for the machine.

kenjdavidson, 2020-06-10:
Thanks for looking into it, that would definitely cause you issues attempting to send HEX.  Although I'm not sure how receiving hex will work, since it's doing the exact same conversion to string, although it will be easier for you to convert back.

I don't know how that will affect users already using as it is (as mentioned somewhere, most of the Android stuff I just pulled over from react-native-bluetooth-serial).  I've slowly been re-writing to include more configurability, like adding middleware/handlers to the on and receive that can be applied to the device during the `connect` request.

I just haven't had time to finish.  also to get the same functionality on Apple, I need to rewrite a bunch of that so that its either All Swift or All Objective C. 

Hopefully you can work off your own changes for the time being, but I will definitely leave this open so I have another instance of configured data.

closed 2021-01-20 by kenjdavidson