kenjdavidson/react-native-bluetooth-classic #52: unable to connect to media devices

Dheeeraj, 2020-07-07:
**Mobile Device Environment**
- Device: Android
- OS: 9.0

**Application Environment**
- React Native version: 0.61.2
- RN Bluetooth Classic version: 0.10.9

**Describe the bug**
Not able to connect to media devices such as bluetooth headphones or speakers
device is getting paired but not able to connect to it

Tried with other native apps and bluetooth manager of my device it connects fine(problem is not at bluetooth device)
this library works fine with normal bluetooth devices problem is with media devices any suggestions 

**Additional context**
logs from android studio
```
D/BluetoothClassicService: Connecting to SPEAKER
D/BluetoothClassicService: setState() DISCONNECTED -> DISCONNECTED
D/BluetoothClassicService: setState() DISCONNECTED -> CONNECTING
D/BluetoothClassicService: BEGIN mConnectThread
D/BluetoothClassicService: Connecting to socket 4C:11:A1:12:17:2B
W/BluetoothAdapter: getBluetoothService() called with no BluetoothManagerCallback
I/BluetoothClassicService: Trying fallback...
W/BluetoothAdapter: getBluetoothService() called with no BluetoothManagerCallback
D/BluetoothSocket: close() this: android.bluetooth.BluetoothSocket@96639f5, channel: 1, mSocketIS: android.net.LocalSocketImpl$SocketInputStream@ead838a, mSocketOS: android.net.LocalSocketImpl$SocketOutputStream@4b13dfbmSocket: android.net.LocalSocket@b7c0e18 impl:android.net.LocalSocketImpl@ead8071 fd:java.io.FileDescriptor@5684e56, mSocketState: INIT
D/RNBluetoothClassicModule: Connection to device SPEAKER has failed
    java.io.IOException: read failed, socket might closed or timeout, read ret: -1
        at android.bluetooth.BluetoothSocket.readAll(BluetoothSocket.java:762)
        at android.bluetooth.BluetoothSocket.waitSocketSignal(BluetoothSocket.java:715)
        at android.bluetooth.BluetoothSocket.connect(BluetoothSocket.java:404)
```

kenjdavidson, 2020-07-07:
Hey, right now the library is only attempting to connect on a specific profile, which is probably why you're getting this error.  https://en.wikipedia.org/wiki/List_of_Bluetooth_profiles#Serial_Port_Profile_.28SPP.29

In order to connect to headsets and other profile based devices, there is a relatively different method for doing so:
https://developer.android.com/guide/topics/connectivity/bluetooth#Profiles

This project, at this point in time, will only connect as server/client https://developer.android.com/guide/topics/connectivity/bluetooth#ConnectDevices

I can't guarantee that I'll ever get around to doing this, but I will definitely keep the task open and maybe look into what is required at some point (post-covid).

Dheeeraj, 2020-07-07:
> Hey, right now the library is only attempting to connect on a specific profile, which is probably why you're getting this error. https://en.wikipedia.org/wiki/List_of_Bluetooth_profiles#Serial_Port_Profile_.28SPP.29
> 
> In order to connect to headsets and other profile based devices, there is a relatively different method for doing so:
> https://developer.android.com/guide/topics/connectivity/bluetooth#Profiles
> 
> This project, at this point in time, will only connect as server/client https://developer.android.com/guide/topics/connectivity/bluetooth#ConnectDevices
> 
> I can't guarantee that I'll ever get around to doing this, but I will definitely keep the task open and maybe look into what is required at some point (post-covid). This is also NOT possible on IOS, which means that this request would add more disparity between Android and IOS, which is not what the point of this project was.

ok got it thank you for the library sir

closed 2020-11-16 by kenjdavidson