kenjdavidson/react-native-bluetooth-classic #57: Basic example not working

nriccar, 2020-08-03:
**Mobile Device Environment**
Provide a list of operating systems on which this issue is relevant.
- Device: iPhone XS
- OS: IOS 13.6

**Application Environment**
Provide information about your development environment:
- React Native version: 0.61.5 (From Example)
- RN Bluetooth Classic version: 0.10.9

**Describe the bug**
[image]

When I try to discover devices, I get this error. I tried do not press discover devices button (I have read the documentation and It seems that its not supported on IOS apparently - then why is this button on IOS and not filtered by a Platform logic?), and instead call list() func at component will mount. This is what I get:

``` LOG  App.render()
 LOG  {"bluetoothEnabled": true, "connectedDevice": undefined, "deviceList": [], "devices": [], "isAccepting": false, "isDiscovering": false, "scannedData": []}
 LOG  DeviceList.render()
 LOG  []```
 
 Thank you for your time

kenjdavidson, 2020-08-03:
Discover devices isn't supported in IOS, but this should definitly be rejecting gracefully and catchable in Javascript. I'll take a look at it when I can. Not sure if flat out rejections or just undefined functions is better, I thought the former, but now I'm thinking the latter.

nriccar, 2020-08-03:
Thank you for your time @kenjdavidson. I guess that undefined would be better.. Nevertheless, list func returns an empty array. How can I get to find my devices? 

Thank you

kenjdavidson, 2020-08-03:
Do you have your plist configured with the allowed mfi protocols?

nriccar, 2020-08-03:
[image]
You mean this one?

kenjdavidson, 2020-08-03:
Yes. Are you using the project example? Or your own app? If the example the plist will merge during building, so if you add the direct protocols they may get overwritten. Look for errors in the ios logs.

I'm honestly not handling missing entries very well, but since they are required I didn't think it a priority.

nriccar, 2020-08-03:
I tried example project and own app both. Example project does not have a supported protocols item, though.. Should I add SPP?

kenjdavidson, 2020-08-03:
Not that type of protocols, which leads me to believe you don't have the device protocol list form the vendor, and why you're getting a bunch of issues.  They will be like

com.vendor.protocol

And probably a couple of them per device.

nriccar, 2020-08-06:
The microchip from the device is RN4677, here's a description:
https://www.microchip.com/wwwproducts/en/RN4677#additional-features

And here's the user guide:
http://ww1.microchip.com/downloads/en/DeviceDoc/50002388A.pdf

I have been searching the protocols (it's not that I don't read) but I'm having a hard time finding it.

Do you have any tip that could help me understand this better?

Thank you for your time
Nicolas

kenjdavidson, 2020-08-06:
I doubt it's mfi compliant. The page says that it's ios version is ble not Bluetooth classic.  You can contact the vendor and ask, but you'll likely have to pay. I suggest using the device in ble mode, which is not what this library is for.

nriccar, 2020-08-06:
@kenjdavidson Thank you for your time, again. 

Actually its rare, because we already have an Ionics app that works with SPP serial Bluetooth. 

The strange thing is that with every library for react native serial bluetooth, I can't get to see the device and connect to it. 

I tried with BLE, but for some reason I write in every characteristic from all of the services and it doesn't write at all on the device.

kenjdavidson, 2020-08-06:
Ya, every React Native library I found for Bluetooth Classic defaulted to BLE on IOS devices.  This is the reason for which I modified the original `react-native-bluetooth-serial` because my company required that we use the IOS ExternalAccessory (MFi) version as our standardized scanner didn't support BLE.

Wish I could provide more assistance, if I get bored I'll take a look at the manuals you posted and see if I can figure something out.  I won't be closing this issue until I resolve the function problem, so if something comes up I'll post here for you.  Sorry again man.

HKSOLDIER, 2020-10-10:
it's just showing number of unpaired devices when performing "Discover Devices". Can you please tell me how to display the device details and how to use them for further. please reply me.

kenjdavidson, 2020-10-10:
@HKSOLDIER could you provide a little more information? Like what exactly you're doing and code you've tried?  All discover devices is supposed to do is find unpaired devices. You need to use that data in your own way to display all the rest of the details.

The example is just that, a basic example to start you on your app. Look at the api and the code, you should be able to see what data is available and display it in your own app. 

If you want more detailsz I need more details. I also suggest opening a new issue, this one was specifically for ios not working which I've already explained, and will be closing.

closed 2020-10-10 by kenjdavidson

HKSOLDIER, 2020-10-10:
Sir, my requirement is when a person enters his car then we have to make a
notification like "you are in your car" by using Bluetooth. Please give
some knowledge for this requirement

kenjdavidson, 2020-10-10:
First of all you probably need to connect tot he car first an d have someone say (this is my car) so you have the address for next time. So discover devices might not be the best option.

If I were doing it Id:
- get the user to pair with the car (pair or native) 
- get the user to select their car as the device to remember (list and select) 
- then you can listen to the on connect event and check against the stored device (connect event)

Since it's unlikely you have the car mfi, I assume this is just Android. You'd then need to write the headless js (service logic) which I don't know if this library will work with. I haven't attempted it in background processes. 

If you need discover devices, it will still return a list of devices. You then need to have the user select their car, then pair it, then save it for subsequent connects.

HKSOLDIER, 2020-10-10:
Sir i need to develop in both android and iOS.
One thing I have to know
When I click on discover the devices near by my phone it is just showing
number of "unpaired devices". I have to know the details about those
devices.
In discover function there is "unpaired" object which is having the details
of " const unpaired = await RNBluetoothClassic.discoverDevices();" but it
is having the "length" attribute but I want to know details of those
devices.

kenjdavidson, 2020-10-10:
I get what you need, but you should be able to figure that out with the api and the code. 

The unpaired is an array of BluetoothDevice objects. You can use them to get the information you want. I get that the documentation needs a little love, but you have access to the code so you should be able to make due.

I have faith in you.

kenjdavidson, 2020-10-10:
Also, this library won't work with ios. Unless you have the mfi protocols for every car company you want to work with.  And they are probably all ble.

HKSOLDIER, 2020-10-10:
Sir
Can you suggest me any other way for this feature because if I did this I
will get a job.

HKSOLDIER, 2020-10-10:
Sir
tell me that how to get the details of the scanned devices which are stored
in "unpaired" object

kenjdavidson, 2020-10-10:
Hey man. No offense but I'm not doing your job for you. The code is available and you can debug things. You have enough available to figure out how to do the things you want. If there is a bug or an improvement, please let me know.

HKSOLDIER, 2020-10-10:
Sorry sir
Thanks for replying to me.

hungdev, 2020-12-30:
I can run example, but it can not get list devices, list device always return [], though it's other bluetooth devices are open
@kenjdavidson

kenjdavidson, 2020-12-30:
If you just posted the comment about MFi protocols on the other page, those aren't real protocols.  So unless you're working with a manufacturer or paid a ridiculous amount of money to get your hardware into the MFi program, you're never going to see devices in that list.

I suggest reading through the MFi links posted on the document page.