kenjdavidson/react-native-bluetooth-classic #58: How can I use discoverDevices in Ios (wanted to help)

hongsa, 2020-09-02:
In ios platform, how can I discover devices??

This library can't use discoverDevices function in IOS

Please help me

kenjdavidson, 2020-09-02:
You cannot, this functionality is not provided in the External Accessory library.  

I know in IOS12 (or whatever) they opened up CoreBluetooth to provide more options (I'm actually using CoreBluetooth for the isEnabled() functionality).  I just don't know if it provides the discover-ability, connectivity functionality that we need for Classic Bluetooth.

You're more than welcome to do some investigation on your own, I'll leave this task open.

hongsa, 2020-09-04:
So how can I know deviceId??
Becase connect funcion is required deviceId!

hongsa, 2020-09-04:
So this library only available in android platform??
Sorry. I am not good at classic bluethooth.

kenjdavidson, 2020-09-04:
You can use `list()` in order to get the currently connected devices.  This will give you access to the deviceId/address so that you can connect to it.  But the MFi protocols are required upfront, and the OS takes care of discovery and connection.   The whole reason I ported it was to get it working with IOS, so I can guarantee it works - although as with Android it might need some minor changes to work with specific hardware.

Mhde96, 2020-10-26:
well what you will do for ios 

> So this library only available in android platform??
> Sorry. I am not good at classic bluethooth.

what you will do for ios

kenjdavidson, 2020-10-26:
IOS MFi requires that the IOS operating system takes care of pairing.  Once you have devices paired, you can use the `list()` function to see all the MFi compliant devices using your provided protocol.  This is how Bluetooth Classic (MFi) on IOS works, there is no way around it.

The library 100% works for what it was designed to do on IOS.   

Like I said, if you want to read about it and see if it's possible, you can look into the CoreBluetooth library which has been opened up a little, but again, it's pretty unlikely due to how locked down MFi is.

Mhde96, 2020-10-27:
> IOS MFi requires that the IOS operating system takes care of pairing. Once you have devices paired, you can use the `list()` function to see all the MFi compliant devices using your provided [...]

Thanks  this is best library for bluetooth classic
if i go   manually to bluetooth screen in ios and scanning on devices  and don't paired to any device 
and reopen my app 
can i get list phone scanned 
Because i just need name and address of bluetooth device

kenjdavidson, 2020-10-27:
No, `list()` displays PAIRED devices, so you will need to pair it through the IOS Bluetooth setting screen.  There is no way around this.  In the next version (currently in progress) I've renamed the function `list()` to `getBondedDevices()` to make it a little more obvious that the device actually needs to be paired/bonded with the Android/IOS system before it will be available.

Also thank you.

closed 2020-11-10 by kenjdavidson