kenjdavidson/react-native-bluetooth-classic #59: this._nativeModule.addListener is not a function. (In 'this._nativeModule.addListener(eventType)', 'this._nativeModule.addListener' is undefined)

incognos, 2020-09-04:
**Mobile Device Environment**
- Device:Android
- OS: API28

**Application Environment**
- React Native version: v0.61.5
- RN Bluetooth Classic version:  v0.10.9

**Describe the bug**
It was working perfectly. I deleted my node_modules folder, rebuilt it using `yarn` and ran `yarn android` and for some reason, I started getting this error whenever I tried to access the listener - whether it was add or remove

**To Reproduce**
```
  useEffect(() => {
    RNBluetoothClassic.addListener(
      BTEvents.BLUETOOTH_CONNECTED,
       device => onConnected(device),
     )
    return () => {
       RNBluetoothClassic.removeAllListeners(BTEvents.BLUETOOTH_CONNECTED)
     }
  }, [])
```

**Expected behavior**
To add and remove listeners without this error

**Screenshots**
[image]

**Additional context**
Tried re-installing, changing versions and searching the web, nothing seems to solve the issue - any help would be appreciated

kenjdavidson, 2020-09-04:
I'm going to take a wild guess and say that something in 0.61 changed from 0.60 in how the `NativeDeviceEmitter` implementation works.  You'll need to post the whole stack trace in order for me to provide more information.  I suggest (if possible) going back to 0.60 and trying it again (as confirmation).

There is no `this._nativeModule.addListener` anywhere in the library code.

Can you debug it and see if the following is being set?

```java
    // Set the _nativeModule for Android, which isn't actually done in NativeEventEmitter
    if (Platform.OS === 'android') 
      this._nativeModule = nativeModule;
```

this needs to happen for Android, or else it's just all going to blow up.

kenjdavidson, 2020-09-04:
If you debug the module, what does it looks like in Javascript? at the point in which it's setting to _nativeModule?  Do other methods on the native module work?

incognos, 2020-09-07:
Hi @kenjdavidson , thanks for the quick reply.

Yes, the other modules seem to work properly - I can still print. But any listener crashes the app. 

I am afraid a return to 0.60 is a bit too much right now. But I have been at 0.61.5 for quite sometime.

Whether it is 
``` javascript
RNBluetoothClassic.removeAllListeners(BTEvents.BLUETOOTH_CONNECTED)
```

or 

``` javascript
    RNBluetoothClassic.addListener(
      BTEvents.BLUETOOTH_CONNECTED,
      device => onConnected(device),
    )
```

it looks like
``` javascript
    // Set the _nativeModule for Android, which isn't actually done in NativeEventEmitter
    if (Platform.OS === 'android')
      this._nativeModule = nativeModule;
```

is being set properly and is not null. and `this._nativeModule` contains

```
{ accept: { [Function: fn] type: 'promise' },
09-07 12:16:59.993 20899 20968 I ReactNativeJS:   available: { [Function: fn] type: 'promise' },
09-07 12:16:59.993 20899 20968 I ReactNativeJS:   cancelAccept: { [Function: fn] type: 'promise' },
09-07 12:16:59.993 20899 20968 I ReactNativeJS:   cancelDiscovery: { [Function: fn] type: 'promise' },
09-07 12:16:59.993 20899 20968 I ReactNativeJS:   clear: { [Function: fn] type: 'promise' },
09-07 12:16:59.993 20899 20968 I ReactNativeJS:   connect: { [Function: fn] type: 'promise' },
09-07 12:16:59.993 20899 20968 I ReactNativeJS:   disconnect: { [Function: fn] type: 'promise' },
09-07 12:16:59.993 20899 20968 I ReactNativeJS:   discoverDevices: { [Function: fn] type: 'promise' },
09-07 12:16:59.993 20899 20968 I ReactNativeJS:   flush: { [Function: fn] type: 'promise' },
09-07 12:16:59.993 20899 20968 I ReactNativeJS:   getConnectedDevice: { [Function: fn] type: 'promise' },
09-07 12:16:59.993 20899 20968 I ReactNativeJS:   isConnected: { [Function: fn] type: 'promise' },
09-07 12:16:59.993 20899 20968 I ReactNativeJS:   isEnabled: { [Function: fn] type: 'promise' },
09-07 12:16:59.993 20899 20968 I ReactNativeJS:   list: { [Function: fn] type: 'promise' },
09-07 12:16:59.993 20899 20968 I ReactNativeJS:   pairDevice: { [Function: fn] type: 'promise' },
09-07 12:16:59.993 20899 20968 I ReactNativeJS:   printTaggedText: { [Function: fn] type: 'promise' },
09-07 12:16:59.993 20899 20968 I ReactNativeJS:   printerReset: { [Function: fn] type: 'promise' },
09-07 12:16:59.993 20899 20968 I ReactNativeJS:   readFromDevice: { [Function: fn] type: 'promise' },
09-07 12:16:59.993 20899 20968 I ReactNativeJS:   readUntilDelimiter: { [Function: fn] type: 'promise' },
09-07 12:16:59.993 20899 20968 I ReactNativeJS:   requestEnable: { [Function: fn] type: 'promise' },
09-07 12:16:59.993 20899 20968 I ReactNativeJS:   selectCodetable: { [Function: fn] type: 'promise' },
09-07 12:16:59.993 20899 20968 I ReactNativeJS:   setAdapterName: { [Function: fn] type: 'promise' },
09-07 12:16:59.993 20899 20968 I ReactNativeJS:   setDelimiter: { [Function: fn] type: 'promise' },
09-07 12:16:59.993 20899 20968 I ReactNativeJS:   setEncoding: { [Function: fn] type: 'promise' },
09-07 12:16:59.993 20899 20968 I ReactNativeJS:   setReadObserving: { [Function: fn] type: 'promise' },
09-07 12:16:59.993 20899 20968 I ReactNativeJS:   unpairDevice: { [Function: fn] type: 'promise' },
09-07 12:16:59.993 20899 20968 I ReactNativeJS:   writeToDevice: { [Function: fn] type: 'promise' },
09-07 12:16:59.993 20899 20968 I ReactNativeJS:   BTEvents: 
09-07 12:16:59.993 20899 20968 I ReactNativeJS:    { READ: 'read',
09-07 12:16:59.993 20899 20968 I ReactNativeJS:      CONNECTION_FAILED: 'connectionFailed',
09-07 12:16:59.993 20899 20968 I ReactNativeJS:      CONNECTION_SUCCESS: 'connectionSuccess',
09-07 12:16:59.993 20899 20968 I ReactNativeJS:      BLUETOOTH_DISABLED: 'bluetoothDisabled',
09-07 12:16:59.993 20899 20968 I ReactNativeJS:      ERROR: 'error',
09-07 12:16:59.993 20899 20968 I ReactNativeJS:      CONNECTION_LOST: 'connectionLost',
09-07 12:16:59.993 20899 20968 I ReactNativeJS:      BLUETOOTH_DISCONNECTED: 'bluetoothDisconnected',
09-07 12:16:59.993 20899 20968 I ReactNativeJS:      BLUETOOTH_CONNECTED: 'bluetoothConnected',
09-07 12:16:59.993 20899 20968 I ReactNativeJS:      BLUETOOTH_ENABLED: 'bluetoothEnabled' },
09-07 12:16:59.993 20899 20968 I ReactNativeJS:   BTCharsets: { UTF16: 'UTF16', UTF8: 'UTF8', ASCII: 'ASCII', LATIN: 'LATIN' },
09-07 12:16:59.993 20899 20968 I ReactNativeJS:   getConstants: [Function] }
```

[... 20 lines cut ...]
09-07 11:57:05.864 20899 20968 E ReactNativeJS:     in StaticContainer (at SceneView.tsx:115)
09-07 11:57:05.864 20899 20968 E ReactNativeJS:     in EnsureSingleNavigator (at SceneView.tsx:114)
09-07 11:57:05.864 20899 20968 E ReactNativeJS:     in SceneView (at useDescriptors.tsx:150)
09-07 11:57:05.864 20899 20968 E ReactNativeJS:     in RCTView (at NativeStackView.tsx:109)
09-07 11:57:05.864 20899 20968 E ReactNativeJS:     in RNSScreen (at createAnimatedComponent.js:151)
09-07 11:57:05.864 20899 20968 E ReactNativeJS:     in AnimatedComponent (at src/index.native.js:111)
09-07 11:57:05.864 20899 20968 E ReactNativeJS:     in Screen (at NativeStackView.tsx:57)
09-07 11:57:05.864 20899 20968 E ReactNativeJS:     in RNSScreenStack (at NativeStackView.tsx:37)
09-07 11:57:05.864 20899 20968 E ReactNativeJS:     in NativeStackView (at createNativeStackNavigator.tsx:68)
09-07 11:57:05.864 20899 20968 E ReactNativeJS:     in NativeStackNavigator (at root-navigator.tsx:31)
09-07 11:57:05.864 20899 20968 E ReactNativeJS:     in RootStack (at root-navigator.tsx:54)
[... cut ...]
09-07 11:57:05.864 20899 20968 E ReactNativeJS:     in RNCSafeAreaView (at src/index.tsx:28)
09-07 11:57:05.864 20899 20968 E ReactNativeJS:     in SafeAreaProvider (at app.tsx:64)
09-07 11:57:05.864 20899 20968 E ReactNativeJS:     in App (at renderApplication.js:40)
09-07 11:57:05.864 20899 20968 E ReactNativeJS:     in RCTView (at AppContainer.js:101)
09-07 11:57:05.864 20899 20968 E ReactNativeJS:     in RCTView (at AppContainer.js:119)
09-07 11:57:05.864 20899 20968 E ReactNativeJS:     in AppContainer (at renderApplication.js:39)
```

do you need additional details?

kenjdavidson, 2020-09-07:
the `this._nativeModule` looks like it's being set.  Sadly it looks like non of the `@ReactMethod` methods are being added from the Android `public interface RCTEventEmitter`.   If they've changed something within Android that stops the Interface `@ReactMethod` from being pulled into the bridged module, then this is going to have to be completely re-written sadly.

incognos, 2020-09-07:
So option 1) drop back down to pre-60 or 2) live without listeners ?

kenjdavidson, 2020-09-07:
or 3) try to debug the React Native bridge and see why the `@ReactNative` methods are not being added.   

I duplicated the IOS functionality in Android since it seemed to be designed better, worst case if this was done on purpose, I can move the methods from `RCTEventEmitter` into the `BluetoothClassicModule` but that seems bad.  Maybe converting the Interface to a Class would do it, and hope that they are differentiating between Interface and parent class?  But that seems like a wild chance.

I admit that none of these options are the best.  But my day job is pretty busy, and with school starting back up for my son, the likelihood of me spending time on this is not high at this point in time.

I appologize.

incognos, 2020-09-08:
No apologies are ever necessary, I am extremely grateful for the library and the time you invested and are affording me here. 

I am using your library to integrate functionality for Datecs BT printers (they have their own sdk and do not work with the modern BT libs) but I did not touch any of the Emitter code. As a matter of fact, to make sure I did not f anything up, I re-installed your lib, but the bridged methods were still not available. (I would not mind re-contributing the code for the Datecs printers, but it is Android only)

What makes this all weird, is the fact that I did the upgrade from 59 to 61.5 more than a month ago, why it decided to appear all of a sudden is illogical at best.

I would be happy to try and contribute to fixing this, but I am a bit rusty on the Android side of things and not sure about the best path to take. Any advice would be very much appreciated.

kenjdavidson, 2020-09-08:
Ya, that's essentially why I converted it, I needed to connect with custom scanners for both Android and IOS.  Last winter I started re-writing the whole thing to make customization easier - I've been tracking all the forks and customization and am hoping that this winter I'll be able to release 0.11.x which allows for far better configuration.

The **BluetoothClassicExample** was last tested with 0.60.5 I believe, when I released 0.10.x.   This is how I do most of my testing, comparing this app to the Bluetooth Terminal app.  I haven't had a chance to write unit tests yet.

Looking at `JavaModuleWrapper#findMethods()` this is the only place that `@ReactMethod` is used.  This builds the `Set<String> methodNames` that are provided to React Native JS.  The weird thing is, it's using `classForMethods.getDeclaredMethods()` which **doesn't include inherited methods** which would explain why the Interface methods are not being brought in.

I can't see any changes in the last 3 years, which would have caused this to:
- Recursively lookup inherited methods

So I can't even give an answer as to why this works with older versions.

incognos, 2020-09-08:
I cannot confirm that it works with older versions - I did not try a downgrade as it would break a bunch of other things. It is likely something else I am using buggered the thing. I tried adding the full implementation of Detox, which buggered my build. So I went back to a previous commit, deleted the node_modules folder and rebuilt, and this is when I seem to have lost the listeners. Been racking my brain in re to what happened between and this is truly the only thing where I could declare a "guilty" verdict.

incognos, 2020-09-08:
Gonna try rummaging through the code and see what I can find. Cheers...

kenjdavidson, 2020-09-08:
Oh it definitely works, it's live in my companies Android and IOS apps.  The apps have been live for over a year now with no issues.   Now that school is starting back up, I will probably have some more time in the evenings for this project.  Keep me posted on your findings, I will post all my notes here if I get a chance to look into it.

kenjdavidson, 2020-09-08:
I found it Commit 678679e from 3 years ago.

Original code:
```java
    for (Map.Entry<String, NativeModule.NativeMethod> entry :
          getModule().getMethods().entrySet()) {
```

which is why it was able to find public methods from the Interface.

kenjdavidson, 2021-01-14:
This shouldn't be an issue any more in the current version, I've removed and reworked the Android native code so that the `EventEmitter` logic is all within the `Module`.  I'm not a big fan of it, but Facebook seems to not give a shit with regards to the issue that was opened there.

So we'll leave it as an internal fix.

closed 2021-01-14 by kenjdavidson