kenjdavidson/react-native-bluetooth-classic #61: Discover Bluetooth devices in background mode in android

rahulkd, 2020-09-28:
I am planning to use this library for a BT Application for my company. The requirement is to discover nearby BT devices based on the RSSI values when the app runs in background. I am able to run your code and discover devices when the state of the app is foreground.

For background, I have tried react native background timer but it seems the BT discovery is not running is background.

Is there any way I can run this library in background mode for BT discovery in android ??

Thanks for your contribution.

kenjdavidson, 2020-09-28:
At this point no, if you feel confident in writing the headless service in a manner that is generic and would fit well with the project, I'd definitely accept a pull request.   If not, I'll need to look into it a little more and when I get time take a run at it.

https://reactnative.dev/docs/headless-js-android

rahulkd, 2020-09-29:
Thanks for your quick response.

I am very new in react native and still learning but definitely will give it a try.
Will really appreciate if you can look into this.

Actually I am planning to build a social distancing app for Covid for my organisation. 
Need to discover nearby BT devices in background.

closed 2024-04-30 by kenjdavidson