kenjdavidson/react-native-bluetooth-classic #62: Encoding problem

hongsa, 2020-10-08:
In ios, I use write function.

And I send string data. And I want to use EUC-KR encoding

But this is not working well..

I think 
```
func writeToDevice(_ message:String) {
        NSLog("(BluetoothDevice:writeToDevice) Writing %@ to device %@", message, accessory.serialNumber)
        if let sending = message.data(using: .utf8) {
            outBuffer.append(sending)
        }
        
        // If there is space available for writing then we want to kick off the process.
        // If all the data cannot be fully written, then the hasSpaceAvailable will be
        // fired and we can continue.  In most cases, we shouldn't be sending that much
        // data.
        writeData()
    }
```

because of utf8!

How can I solve this problem?

kenjdavidson, 2020-10-08:
You'll need to do what most are doing for the Android version, you'll need to fork and update accordingly.   

I'm working on being able to provide customized connection properties during the connection request, but I have limited spare time these days and can't give you a timetable for completing this.

hongsa, 2020-10-12:
I solve this problem.

This is solution

`
     func writeToDevice(_ message:String) {
         NSLog("(BluetoothDevice:writeToDevice) Writing %@ to device %@", message, accessory.serialNumber)
-        if let sending = message.data(using: .utf8) {
+        let encoding = String.Encoding(rawValue: CFStringConvertEncodingToNSStringEncoding(0x0422))
+        if let sending = message.data(using: encoding) {
             outBuffer.append(sending)
         }
`

closed 2020-10-12 by hongsa