kenjdavidson/react-native-bluetooth-classic #65: connect to paired device

mannat26, 2020-10-14:
async initialize() {
    try {
      console.log('inilialize called');
      let devices = await RNBluetoothClassic.list();
      console.log(devices, 'list');
      // this.setState({connectedDevices: devices});
      await RNBluetoothClassic.setEncoding(BTCharsets.ASCII);
      let connectedDevice = await RNBluetoothClassic.connect(devices[4].id);
      console.log(connectedDevice, 'check device list');
      // this.setState({connectedDevices: devices});
    } catch (err) {
      console.log(err, 'errror while getting paired');
    }
  }

react native version is 0.63.3,

Issue on connect to paired device, here i get the list of paired device, but when i tried to connect with device using id it throws an error showing Connection unsuccessful.
I am trying to connect with android phone version 10 and smart watch.

kenjdavidson, 2020-10-14:
Have the trace? Debug details? Or any outgrew information to help reproduce this? You haven't given much yo go on.

Does the example app connect?

mannat26, 2020-10-14:
Error: Connection unsuccessful
    at RNBluetoothClassic.promiseMethodWrapper [as connect] (NativeModules.js:103)
    at Eanmple.connectToDevice$ (Example.js:299)
    at tryCatch (runtime.js:63)
    at Generator.invoke [as _invoke] (runtime.js:293)
    at Generator.next (runtime.js:118)
    at tryCatch (runtime.js:63)
    at invoke (runtime.js:154)
    at runtime.js:164
    at tryCallOne (core.js:37)
    at core.js:123 "error on connectToDevice"
  
  I am getting this error while connecting to device.
 no example app throws the same above error

kenjdavidson, 2020-10-14:
Have you attempted to debug the native code (Android)? You're not really providing much to go on.

I'll try to upgrade it to 0.63.3 when I get a chance and see if it works with my devices.

mannat26, 2020-10-14:
My requirement is to tracing the heartRate and steps count from the fitness band or smart watch. so I am using above code.

kenjdavidson, 2020-10-14:
That sounds like you should be using a BLE profile?

mannat26, 2020-10-14:
no device support BLE profile

kenjdavidson, 2020-10-14:
You're going to need to provide a lot more information then, some do diligence on your part:
- The name of the device
- Documentation on the device bluetooth information
- Your code repository for me to review
- Some more information from you debugging the native code and getting Native exceptions

Pretty much anything else that might help (if you want my help) to try and resolve this.  I don't have a lot of time, and with the information you've given so far, it'll be impossible for me to do anything remotely useful.

saba-coding, 2020-11-03:
Hi,

Following the thread i am using react-native 0.63.3 running on ios 14.0.1 with react-native-bluetooth-classic 0.10.10

const list = async () => {
    const device = await RNBluetoothClassic.list();
    console.log('App -> device', device); // []
 };

I get back an empty array even though i am paired with a device any suggestions?

kenjdavidson, 2020-11-03:
Do you have your MFi protocol list configured correctly?  This is the number one reason most people have issues with IOS.

saba-coding, 2020-11-03:
I added following lines in my info.plist file as explained in documentation
```
<key>UISupportedExternalAccessoryProtocols</key>
<array>
	<string>com.apple.m1</string>
</array>
<key>NSBluetoothAlwaysUsageDescription</key>
<string>We need access to bluetooth in order to pair with device</string>
```

kenjdavidson, 2020-11-03:
Ya, that's not a **real** protocol string.  That's only an example of how and where to put it in.  I suggest you spend a bit of time going through the MFi documentation.  Unless you're working with a manufacturer/vendor that you've signed an NDA with, you're probably never going to get the MFi protocol for what you're trying to do.  

You can't just use Bluetooth classic on IOS like you can on Android, Apple's implementation of it is (even though I get why they do things their own way) a complete pain in the ass.

saba-coding, 2020-11-03:
Thanks for your response

closed 2020-11-10 by kenjdavidson

AzranAzwer, 2022-10-02:
Hi @kenjdavidson , How Can I get MFi documentation to refer the protocol ?

kenjdavidson, 2022-10-02:
You need to get it from the hardware vendor.

AzranAzwer, 2022-10-02:
Thank you for the info.