kenjdavidson/react-native-bluetooth-classic #68: Doesn't "await RNBluetoothClassic.write(message);" returns a boolean?

leoP0, 2020-11-11:
**Mobile Device Environment**
- Device: Google Pixel, kindle fire 8, Nexus 9
- OS: API >= 16

**Application Environment**
- React Native version: 0.63.3
- RN Bluetooth Classic version: 0.10.10

**Describe the bug**
When setting `let messageSent = await RNBluetoothClassic.write(message);` and then I do `console.log(messageSent);` shows up as:  "undefined"

I can confirm that the message was sent and received in the other device.

**To Reproduce**
1. From the example provided just add `let messageSent =`  to  `await RNBluetoothClassic.write(message)`
2. add a line bellow `console.log(messageSent);`
3. Connect to a BT device and send data 
4. Take a look at the result of `console.log(messageSent)`

**Expected behavior**
Shouldn't this promise return either true or false?

**Screenshots**
N/A

**Additional context**
What am I missing here? any thoughts?

kenjdavidson, 2020-11-11:
Ya, you're right.  The javascript is messed up and not actually passing back the Promise from the native side

```
/**
   * Write data to the device.  Eventually this will be updated to accept data and type, 
   * allowing the sending of different data elements to the device.  From the issues on
   * bluetooth-serial it seems like images and hex values are the top priorties, but method
   * to send any data would be preferable.
   * 
   * @param {string|buffer} data to be sent to the device as base64 string 
   * 
   * TODO modify for byte[] instead of string
   */
  write = (data) => {
    if (typeof data === 'string') {
      data = new Buffer(data);
    }
    this._nativeModule.writeToDevice(data.toString('base64'));
  }
```

should be:

```
/**
   * Write data to the device.  Eventually this will be updated to accept data and type, 
   * allowing the sending of different data elements to the device.  From the issues on
   * bluetooth-serial it seems like images and hex values are the top priorties, but method
   * to send any data would be preferable.
   * 
   * @param {string|buffer} data to be sent to the device as base64 string 
   * 
   * TODO modify for byte[] instead of string
   */
  write = (data) => {
    if (typeof data === 'string') {
      data = new Buffer(data);
    }
    return this._nativeModule.writeToDevice(data.toString('base64'));
  }
```

kenjdavidson, 2020-11-12:
Released v0.10.11 with the update to `write(data: String|Buffer): Promise<Boolean>`

closed 2020-11-12 by kenjdavidson

leoP0, 2020-11-12:
Thanks for pointing it out!