kenjdavidson/react-native-bluetooth-classic #71: Invalid Connector Type

shaanhershey, 2020-11-22:
Using Samsung Galaxy S10.

I am using Bluetooth to communicate with an HC-05 module attached to an Arduino. I can connect to the module through the Android Bluetooth Terminal and an insecure connection, but when I try to connect through the package it gives the error "Invalid connector type: rfcomm." After some research, I am fairly certain rfcomm should be the correct type.

The line that has the issue:
const device = await RNBluetoothClassic.connectToDevice("98:D3:31:F4:25:E3", {
                CONNECTOR_TYPE: "rfcomm",
                SECURE_SOCKET : false
            });
or 
await device.connect"98:D3:31:F4:25:E3", {
                CONNECTOR_TYPE: "rfcomm",
                SECURE_SOCKET : false
            });

I just want to be able to connect to the device the same that the android Bluetooth terminal would allow me.

kenjdavidson, 2020-11-22:
Rfcomm should be the default type. Can you try to remove the connector type parameter and just pass secure parameter?

I might have boned something in the determination of the connector/acceptor types. Although I'm fairly certain that "rfcomm" should still work when set explicitly.

You're using the default package? Or are you providing a custom implementation?

shaanhershey, 2020-11-22:
I am using the default package and I tried without the connector type parameter and it still gives the error.

shaanhershey, 2020-11-22:
> Rfcomm should be the default type. Can you try to remove the connector type parameter and just pass secure parameter?
>
> I might have boned something in the determination of the connector/acceptor [...]

I am using the default package and I tried without the connector type parameter and it still gives the error.

kenjdavidson, 2020-11-22:
I just fired up the development app and attempted the following:

```javascript
      let connectionOptions {
        CONNECTOR_TYPE: 'rfcomm',
        SECURE_SOCKET: false,
        DELIMITER: '\n'
      };

      let connection = await this.props.device.isConnected();
      if (!connection) {
        connection = await this.props.device.connect(this.state.connectionOptions);

        this.addData({
          data: `Connection successful`,
          timestamp: new Date(),
          type: 'info'
        });
      }

      this.setState({connection});
```

and the connections looks good; data is being processed as expected.   I'm creating a new app from scratch and will attempt to pull it in from NPM instead, maybe there is something missing on the released version.

kenjdavidson, 2020-11-22:
Could you give me some info on what how you created the app?
- React native version?
- Android version?

All the things in the ISSUE_TEMPLATE that you deleted?

shaanhershey, 2020-11-22:
Yes sorry, I am using React native version 0.63.3 and the Bluetooth version of 1.60.0-rc.3. My android version is 10. 

The code in question is this function:
async connectAndSetup () {
        try{
            const granted = await PermissionsAndroid.request(PermissionsAndroid.PERMISSIONS.ACCESS_FINE_LOCATION);
            console.log(granted);
            const device = await RNBluetoothClassic.pairDevice("98:D3:31:F4:25:E3");
            await device.connect({
                SECURE_SOCKET : false
            });
            this.setState({device : device});
            console.log(device.name);
        } catch (err){
            console.log(err.message);
        }
    }

which returns the error described above.

shaanhershey, 2020-11-22:
Yes sorry, I am using React native version 0.63.3 and the Bluetooth version of 1.60.0-rc.3. My android version is 10.

The code in question is this function:
`async connectAndSetup () {
        try{
            const granted = await PermissionsAndroid.request(PermissionsAndroid.PERMISSIONS.ACCESS_FINE_LOCATION);
            console.log(granted);
            const device = await RNBluetoothClassic.pairDevice("98:D3:31:F4:25:E3");
            await device.connect({
                CONNECTOR_TYPE: "rfcomm",
                SECURE_SOCKET : false
            });
            this.setState({device : device});
            console.log(device.name);
        } catch (err){
            console.log(err.message);
        }
    }`

which returns the error described above.

shaanhershey, 2020-11-22:
Yes sorry, I am using React native version 0.63.3 and the Bluetooth version of 1.60.0-rc.3. My android version is 10.

The code in question is this function:
async connectAndSetup () {
        try{
            const granted = await PermissionsAndroid.request(PermissionsAndroid.PERMISSIONS.ACCESS_FINE_LOCATION);
            console.log(granted);
            const device = await RNBluetoothClassic.pairDevice("98:D3:31:F4:25:E3");
            await device.connect({
                SECURE_SOCKET : false
            });
            this.setState({device : device});
            console.log(device.name);
        } catch (err){
            console.log(err.message);
        }
    }

(I apologize my code linking is not working)

which returns the error described above.

shaanhershey, 2020-11-22:
This code will output the following to the log:
"granted"
"HC-05"
" Invalid connector type: rfcomm"

The HC-05 is the correct name of what I would like to connect to which is connected to using rfcomm.

kenjdavidson, 2020-11-22:
Right on, thanks.   It might be something with `0.63.3` that's the issue.   Have you attempted to run the example app with your device?   I'll try to setup a new project with your configuration later today.

shaanhershey, 2020-11-22:
I have not attempted to run the example app on my device.

kenjdavidson, 2020-11-22:
Are you able to debug it? What happens in the native connect to device method when you're running it. Can you provide the contents of:

- mconnectors (class) 
- properties (method)

kenjdavidson, 2020-11-23:
Looks like the released version has an issue (I've already fixed locally, hence why this was working fine for me) just haven't pushed that change yet.  Sorry!!

https://github.com/kenjdavidson/react-native-bluetooth-classic/blob/1d0155166a4c47a574142ddbfa3bdf586c8f99b6/android/src/main/java/kjd/reactnative/bluetooth/RNBluetoothClassicPackage.java#L78

It's applying the default CONNECTION_TYPE **delimiter** to the CONNECTOR_TYPE.   It's using the right type though.  I'm going to release this in a second, but you should be able to use:

```javascript
device.connect({
  CONNECTOR_TYPE: 'delimited'    // The issue here is that it's named wrong.
});
```

and you should be able to continue working until I publish the next RC.

kenjdavidson, 2020-11-23:
After making the appropriate change:

```
Running "Issue71InvalidConnector" with {"rootTag":21}
App.js:37 Bluetooth enabled: true
App.js:41 Paired devices:
App.js:42 (9) [BluetoothDevice, BluetoothDevice, BluetoothDevice, BluetoothDevice, BluetoothDevice, BluetoothDevice, BluetoothDevice, BluetoothDevice, BluetoothDevice]
App.js:48 {timestamp: "2020-11-22T19:35:56.584Z", data: "985141001092814↵"}
```

shaanhershey, 2020-11-23:
I believe that worked, thank you!

closed 2020-11-23 by shaanhershey

kenjdavidson, 2020-11-23:
`1.60.0-rc.4` should be available for you with the appropriate `rfcomm` name.

ipolotsky, 2021-04-01:
> ```js
> CONNECTOR_TYPE: 'delimited'
> ```

Helped me a lot. Thank you =)