kenjdavidson/react-native-bluetooth-classic #73: cannot connect to a device with message "Invalid connector type: rfcomm"

william-schlegel, 2020-11-24:
I created an app tested on android with react native (0.63) and react-native-bluetooth-classic (1.60.0)

I try to connect to a raspberry. It works fine with the demo application but when I try to do it with my app, I have this message "Invalid connector type: rfcomm"
Here is my code for connection:
`
export async function connectToDevice(device) {
  console.log(`Attempting connection to device ${device.id}`);
  try {
    let connection = await device.isConnected();
    if (!connection) {
      connection = await device.connect();
    }
    return connection;
  } catch (error) {
    console.log(error.message);
    throw new Error(error.message);
  }
}
`
What is wrong?
thank you!

kenjdavidson, 2020-11-24:
See issue #71 and let me know if it's the same issue/resolution. It should be resolved with **1.60.0-rc.4**. Let me know.

william-schlegel, 2020-11-24:
yes, that was it! works perfectly with rc4 (I was using rc3)
Thank you so much, you saved my day 😃

kenjdavidson, 2020-11-24:
Right on.  Shoot me an email with the finished product, love to see how this is getting used.

closed 2020-11-24 by kenjdavidson