kenjdavidson/react-native-bluetooth-classic #74: How to read the AT command sent by the Bluetooth Classic?

kidfrom, 2020-12-05:
**Mobile Device Environment**
- Device: Android 4.4, iPhone 6
- OS: API19, iOS8.1

**Is your feature request related to a problem? Please describe.**
My goal is to read the AT command sent by the Bluetooth Classic. The AT command will be used to trigger a JavaScript method.

**Describe the solution you'd like**
Android source code: https://github.com/kidfrom/SimpleBluetoothTerminal
Android app: https://play.google.com/store/apps/details?id=de.kai_morich.serial_bluetooth_terminal

**Additional context**
Could you teach me on how to write natively and integrate it with React Native? I would pay $50 for the 1:1 pair programming. I have an experience in Java but as a Back End. I will not waste your time.

kenjdavidson, 2020-12-05:
Hey man. I don't necessarily mind helping. No need for the 50 bucks. But my time is limited and you'll need to give me way more information than just a forked repository and a brief description.

At commands are generally on the micro device side? Not actually sent to the controlling device, no?  Do you have model numbers? Detailed messages? Etc? 

Let me know and I'll do my best.

kidfrom, 2020-12-06:
From what I understand after using the SerialBluetoothTerminal app, the Bluetooth Microphone sent a AT commands to the phone, these are the information I gathered.

## Serial Port(SPP)

```
UUID: 00001101-0000-1000-8000-00805f9b34fb
```

## Push To Talk Button

```
# Press
$ C:BRGIN*^M+GPIO=1
# Release
$ C:END*^M+GPIO=0
```

## SOS Button

```
$ C:SOS*
```

## Group Switch / Channel Switch

```
# Switch to Group / Channel above.
C:VM*
# Switch to Group / Channel below.
C:VP*
```

kenjdavidson, 2020-12-06:
Alright. Well I mean you just need to use my sample app to test that the onRecievedMessage is being called. Then you just need to parse the specified message, you just linked, with either a regex or manually. On the appropriate message call he correct feature of the record library.

I assume the At commands are /n delimited messages that are easily listened to using the library.

kenjdavidson, 2020-12-06:
I'm confused where you're having issues? Are you using this library? Or are you writing your own custom library?

kidfrom, 2020-12-07:
@kenjdavidson I believe we don't have `BluetoothClassicExample` project folder yet for the master branch. The link you provided on README.md is broken.

Anyway, react-native-bluetooth-classic manage to read some AT commands `C:BRGIN*` and `C:END*`. I am not sure why it did not read `C:SOS*`, `C:VM*`, `C:VP*`. 

I will compare react-native-bluetooth-classic's source code and [SimpleBluetoothTerminal's source code](https://github.com/kai-morich/SimpleBluetoothTerminal) after I finished my work (5 PM GMT+7).

The listener
```
  const bluetoothConnect = async (address: string) => {
     ...
    bluetoothReadSubscription = bluetoothDevice.onDataReceived(data => bluetoothOnDataReceived(data));
    ...
  }

  const bluetoothOnDataReceived = (event: BluetoothDeviceReadEvent) => {
    console.log(event)
  };
```

I also need to update the BluetoothEvent.d.ts if I follow the example on [onDataReceived](https://kenjdavidson.com/react-native-bluetooth-classic/react-native/rn-bluetooth-device/#ondatareceived)
```
<<< Current Change
export declare type BluetoothEventListener<T extends BluetoothEvent> = (event: T) => undefined;
===
export declare type BluetoothEventListener<T extends BluetoothEvent> = (event: T) => undefined | void;
>>> Incoming Change
```

The output
```
[Mon Dec 07 2020 23:36:10.890]  LOG      connected to bluetooth device with address '00:13:04:84:03:1F'
+GPIO=1c 07 2020 23:36:12.475]  LOG      {"data": "C:BRGIN*
", "timestamp": "2020-12-07T23:36:11.829Z"}
+GPIO=0c 07 2020 23:36:12.845]  LOG      {"data": "C:END*
", "timestamp": "2020-12-07T23:36:12.199Z"}
```

kidfrom, 2020-12-07:
I also will take a look on how to read the audio sent by the Bluetooth Microphone tomorrow (after 5PM GMT+7), currently the library on read the AT commands.

kidfrom, 2020-12-07:
Update:

This is a major bug for me, I will try to solve it tomorrow.

To make things easier:
1. There is 4 button that send AT commands.
  1. Microphone button on Press sends `C:BRGIN*^M+GPIO=1` and on Release sends `C:END*^M+GPIO=0`
  2. Group Switch Up button sends `C:VM*`
  3. Group Switch Down button sends `C:VP*`
  4. SOS button sends `C:SOS*`

**The source of problem:**
react-native-bluetooth-classic` onDataReceived will not be triggered if the data does not end with ASCII Char Number 10 or Char LF.

To make things easier:
1. ${CR} is String.fromCharCode(13)
2. ${LF} is String.fromCharCode(10)

Example 1 (end without `${LF}`):
1. The bluetooth device send `C:SOS*${CR}`
2. The onDataReceived will not be triggered.

Example 2 (end with `${LF}`):
1. The bluetooth device send `C:BRGIN*${CR}+GPIO=1${CR}${LF}`
2. The onDataReceived will be triggered.

**Severity of the bug:**
1. The onDataReceived will only be triggered if the bluetooth device send data with endings `${LF}`, which is not a normal behavior. The onDataReceived should be triggered regardless of the endings.

kenjdavidson, 2020-12-07:
The standard connection type of `DelimitedStringDeviceConnection` allows you to configure a termination string

```javascript
// This is effectively what the default config device.connect() does.
let connected = await device.connect({
  delimiter: '\n'
})
```

is the standard functionality.  You should be able to configure this based on what you want:

```javascript
let connected = await device.connect({
  delimiter: String.fromCharCode(10)
});
```

If you've tried to configure the connection this way, and it doesn't work, then yes, that is an issue.  I've tried, and confirmed this working with::

- `\n` for android devices using scanner
- `\r` for IOS devices using scanner

so the ability to change the delimiter definitely works, it's just a combination of character set.

kenjdavidson, 2020-12-07:
```
<<< Current Change
export declare type BluetoothEventListener<T extends BluetoothEvent> = (event: T) => undefined;
===
export declare type BluetoothEventListener<T extends BluetoothEvent> = (event: T) => void;
>>> Incoming Change
```

This should definitely be merged in.

kenjdavidson, 2020-12-07:
I updated the readme with the path to [react-native-bluetooth-classic-apps](https://github.com/kenjdavidson/react-native-bluetooth-classic-apps) where I've moved the development app.  This was done due to some annoyances I ran into while attempting to debug/write tests for the different versions that I was maintaining.

kidfrom, 2020-12-07:
> The standard connection type of `DelimitedStringDeviceConnection` allows you to configure a termination string
> 
> ```js
> // This is effectively what the default config device.connect() does.
> let connected = await device.connect({
>   DELIMITER: '\n'
> })
> ```
> 
> is the standard functionality. You should be able to configure this based on what you want:
> 
> ```js
> let connected = await device.connect({
>   DELIMITER: String.fromCharCode(10)
> });
> ```
> 
> If you've tried to configure the connection this way, and it doesn't work, then yes, that is an issue. I've tried, and confirmed this working with::
> 
> * `\n` for android devices using scanner
> * `\r` for IOS devices using scanner
> 
> so the ability to change the delimiter definitely works, it's just a combination of character set.
> 
> The list of options currently available for connect/connection are documented here https://kenjdavidson.com/react-native-bluetooth-classic/react-native/rn-bluetooth-device/#connect

Is this correct way to import it?
I could not import { BluetoothDevice } from "react-native-bluetooth-classic". I imported BluetoothDevice from "react-native-bluetooth-classic/lib/BluetoothDevice".

I need to change the types `react-native-bluetooth-classic/lib/BluetoothDevice`
```
<<< Current Change
export default class BluetoothDevice implements BluetoothNativeDevice {
    ...
}
===
export interface BluetoothDevice implements BluetoothNativeDevice {
    ...
}
>>> Incoming Change
```

The types also does not accept `connect( { DELIMITER: '\n' } )`. My TypeScript version is "~3.9.5".

react-native-bluetooth-classic/lib/BluetoothDevice
```
<<< Current Change
export default class BluetoothDevice implements BluetoothNativeDevice {
    ...
    connect(options: Map<string, object>): Promise<boolean>;
    ...
}
===
interface connectOptions {
    DELIMITER: string
}

export default class BluetoothDevice implements BluetoothNativeDevice {
    ...
    connect(options: connectOptions): Promise<boolean>;
    ...
}
>>> Incoming Change
```

kenjdavidson, 2020-12-07:
It definitely needs a lot of cleaning up, hence why it's still being released as a release candidate.   Again, this is all working with the example application that I use for testing with my companies devices.   The types definitely need to be cleaned up a little, but it would more likely be something like

```
<T extends ConnectOptions> connect(options: T);
```

So that if someone needed to implement their own Acceptor, Connector or DeviceConnection they could.  

I appreciate the suggestions, I haven't updated the Bluetooth Classic Example app to typescript, so I wasn't seeing any of these issues during development.

kidfrom, 2020-12-07:
Anyway, I tried to use the `DELIMITER` but it does not make any difference. I read from the example app you provided earlier and it use lower case `delimiter` it does not work either.

```
  const bluetoothConnect = async (address: string) => {
    const enabled = await bluetoothIsEnabled();
    if (enabled === true) {
      try {
        const connected = await bluetoothIsConnected(address);
        if (connected === false) {
          await RNBluetoothClassic.connectToDevice(address);
          const result = await bluetoothIsConnected(address);
          if (result === true) {
            bluetoothDevice = await RNBluetoothClassic.getConnectedDevice(
              bluetoothDeviceAddress
            );
            bluetoothDevice.connect({ DELIMITER: String.fromCharCode(10) }); // This is where I
            bluetoothReadSubscription = bluetoothDevice.onDataReceived((data) =>
              bluetoothOnDataReceived(data)
            );
            console.log(
              `connected to bluetooth device with address '${address}'`
            );
          }
        }
      } catch (error) {
        console.error(`bluetoothConnect(): ${error}`);
      }
    }
  };
```

kenjdavidson, 2020-12-07:
Hey ya sorry, it looks like it should be lowercase, but it definitely should work.  Here is the line in the sample app that I've used to make sure that it parses messages based on delimiter:

https://github.com/kenjdavidson/react-native-bluetooth-classic-apps/blob/c0abf9bc0e00970e1f5c16e58c301983de0fea40/BluetoothClassicExample/src/connection/ConnectionScreen.js#L26

This also takes into account the encoding that is set on the device.    It looks like the default for charset is Ascii, so that shouldn't be an issue:

https://github.com/kenjdavidson/react-native-bluetooth-classic/blob/73bb0d1cb89e5c0ca258d6dc29da50f00ffd43a1/android/src/main/java/kjd/reactnative/bluetooth/conn/StandardOption.java#L50

You're going to need to debug the Android side and see why exactly it's not deliminating the message and sending the appropriate data.

kidfrom, 2020-12-07:
Regarding the types generated automatically by TypeScript. I think you should create a NPM package with `@types/react-native-bluetooth-classic` and make a dependency for the `react-native-bluetooth-classic` package if the project use TypeScript. I have never done it though, not sure if you can upload a package with @types.

Sorry if I could not help too much. I only have 4 months worth of experience and mainly focus as a Java (Spring Boot) developer.

kenjdavidson, 2020-12-07:
Ya, that isn't going to happen anytime soon.  What's the point of having Typescript if the definitions it builds doesn't work with Typescript?  I won't get started of of my opinions on Javascript as a whole.   I'm glad I spend most my time (like you) on the back end side.

kidfrom, 2020-12-07:
I appreciate you open source this. I think you should continue to use JavaScript to speed things up.

**I think this could be the problem:** Can you change the delimiter after the bluetooth device is already connected?

The current logic of method `bluetoothConnect()` is: Connect the bluetooth device => Change the delimiter.

I tried to use random char number. e.g. 21, it should not trigger onDataReceived listener, but it does. **I think the issue is with how I wrote bluetoothConnect(). I will refactor it tomorrow, it's 3 AM now.**

The connect()
```
bluetoothDevice.connect({ delimiter: String.fromCharCode(21) });
```

The output
```
[Tue Dec 08 2020 03:03:00.812]  LOG      bluetooth is enabled: true
[Tue Dec 08 2020 03:03:00.859]  LOG      bluetooth is connected: false
[Tue Dec 08 2020 03:03:01.225]  LOG      bluetooth is connected: true
[Tue Dec 08 2020 03:03:01.282]  LOG      connected to bluetooth device with address '00:13:04:84:03:07'
+GPIO=1c 08 2020 03:03:05.728]  LOG      C:BRGIN*

+GPIO=0c 08 2020 03:03:06.138]  LOG      C:END*
```

@kenjdavidson anyway does this library support bluetooth profile A2DP? I just aware that Bluetooth Microphone use A2DP to send the audio data haha.

Thank you for the time.

kenjdavidson, 2020-12-07:
I don't think I understand what you're doing.   To answer your questions:

No, you cannot change the delimiter after connection.  The Connection Options are immutable once they get to the Native side (java / ios).   The process goes like this:

- Javascript `device.connect`
- Java - ConnectionConnector picks out it's options and that's it
- Java - DeviceConnection picks out its options and thats it

So once you tell the the Native side to connect to your device with delimiter `\n` it will only ever use that.  If you change the connection options on Javascript side, you need to `disconnect` and `connect` again with the new options.

This was done deliberately, as it made no sense to change.  If you're connected to a bluetooth device, the communication is pretty static and strict.  You can re-configure the device sure, but at that point you should open a new connection.

With regards to the `BluetoothProfiles` no, it does not yet support them.   I have a task open to try and get this working.   Now, you could get created and write your own set of:

- Acceptor
- Connector
- DeviceConnection

that would allow you to manage the connection that way.  For example:

A2DPConnectionConnector
- Create receiver
- attempt to connect
- on connect return a NULL bluetooth Socket (since I'm not actually checking)
- Set some singleton wrapper on your native side with the AD2P connection

A2DPDeviceCOnnection
- accept the null bluetooth Socket
- Read the singleton wrapper for the `HeadsetProfile`

Then you'd use `read` and `write` to send messages back and forth and your `A2DPDeviceConnection` would manage those appropriately.

It's super hacky, but at this point I just don't have time to add the `BluetoothProfile` connection pieces.

kidfrom, 2020-12-08:
Good morning.

I am unable to find method to get the unpaired BluetoothDevice. The method I could found is `RNBluetoothClassic.getConnectDevice(address: string)`, and `BluetoothDevice.connect(options: Map<string, Object>)` but it returns boolean.

I am looking for method to get BluetoothDevice without connecting it first, similar to the `BluetoothClassicExample` project. If I am not wrong, in the `BluetoothClassicExample` project, you have a state with key value `device` and somehow the device can run `connect(options: Map<string, Object>)` method, which I am assuming is a BluetoothDevice. Am I missing something?
```
const device: BluetoothDevice;
```

I read the `connectToDevice(address: string, properties: Map<string, Object>)` method which connect the bluetooth device to the app and returns the BluetoothDevice.

```
    /**
     * Attempts to connect to the requested device.  Defaults the properties to an empty
     * map.
     *
     * @param address the address to which we are connecting
     * @param properties extra properties required for the connection.
     */
    connectToDevice(address, properties = new Map()) {
        return __awaiter(this, void 0, void 0, function* () {
            let connected = yield this._nativeModule.connectToDevice(address, properties);
            return new BluetoothDevice(connected, this);
        });
    }
```

This is the method I wrote to connect. I set the delimiter to `aaaaa` or String.fromCharCode(2) which I assume will not trigger the `BluetoothDevice.onDataReceived()`, but it did not. The `BluetoothDevice.onDataReceived()` still got triggered with `String.fromCharCode(10)` / LF only.

```
  /**
   * // To Do: fix bluetoothDevice.connect({ delimiter: String.fromCharCode(13) }) is ignored.
   * // Potential cause of error: the connected device can't change delimiter.
   * @param address
   */
  const bluetoothConnect = async (address: string) => {
    const enabled = await bluetoothIsEnabled();
    if (enabled === true) {
      try {
        const connected = await bluetoothIsConnected(address);
        if (connected === false) {
          bluetoothDevice = await RNBluetoothClassic.connectToDevice(address, { delimiter: 'aaaaa' }); <==
          const result = await bluetoothIsConnected(address);
          if (result === true) {
            // bluetoothDevice = await RNBluetoothClassic.getConnectedDevice(
              // bluetoothDeviceAddress
            // );
            bluetoothReadSubscription = bluetoothDevice.onDataReceived((data) =>
              bluetoothOnDataReceived(data)
            );
            console.log(
              `connected to bluetooth device with address '${address}'`
            );
          }
        }
      } catch (error) {
        console.error(`bluetoothConnect(): ${error}`);
      }
    }
  };
```

kidfrom, 2020-12-08:
Thank you for the explanation regarding `BluetoothProfiles`

> With regards to the BluetoothProfiles no, it does not yet support them. I have a task open to try and get this working. Now, you could get created and write your own set of:

I will try this week, I will get back to you by the end of the weekend.

This may take sometime. Anyway, I will not be writing the iOS native module as I do not have a Mac. Any thoughts?

kenjdavidson, 2020-12-08:
For finding unpaired devices you need to use the `discovery` methods.  This will give you a List<BluetoothDevice> which are unpaired, you can then choose to pair or do what you wish.

With regards to the delimiter, it definitely isn't "ignored" although it may not be working as you're expecting, but it looks like you're doing things right.   I have a bluetooth scanner (barcode) that I've played around with:

**123123123456\n**

`device.connect()` will return 123123123456
`device.connect({ delimiter: '1' }); returns [123, 123, 123456] in separate events

so it's definitely working as I intended it to, although I haven't tried:

- With multiple character strings, although I can't see it being a difference, I'm just doing an indexOf(delimiter) to find if it exists.

You should really just start debugging the Android code and see why it's not delimiting your messages appropriately.  I'm going to guess you have the wrong charset or something.

kidfrom, 2020-12-08:
I notice that the script require `new Map()` for the ConnectionOptions.

This will return an error `[Tue Dec 08 2020 20:51:08.217]  ERROR    bluetoothConnect(): TypeError: properties.get is not a function. (In 'properties.get("delimiter")', 'properties.get' is undefined)`
```
let bluetoothDevice = RNBluetoothClassic.connectToDevice(address, { delimiter: '1' } )`
```

The culprit is [StandardOption.java](https://github.com/kenjdavidson/react-native-bluetooth-classic/blob/73bb0d1cb89e5c0ca258d6dc29da50f00ffd43a1/android/src/main/java/kjd/reactnative/bluetooth/conn/StandardOption.java#L50)
```
    DEVICE_CHARSET("charset", Charset.class, Charset.forName("ascii")) {
        @Override
        public <T> T get(Properties properties) {
            Object value = properties.get(this.name());
            if (value == null) {
                value = Charset.forName("ascii");
            } else if (value instanceof String) {
                value = Charset.forName((String) value);
            }

            return (T) value;
        }
    },
```

I use this `new Map()` for the properties.

```
let properties = new Map();
properties.set("delimiter", "1");

let bluetoothDevice = RNBluetoothClassic.connectToDevice(address, properties);
```

> You should really just start debugging the Android code and see why it's not delimiting your messages appropriately. I'm going to guess you have the wrong charset or something.

I tried to debug the [StandardOption.java](https://github.com/kenjdavidson/react-native-bluetooth-classic/blob/73bb0d1cb89e5c0ca258d6dc29da50f00ffd43a1/android/src/main/java/kjd/reactnative/bluetooth/conn/StandardOption.java#L43)
```
<<< Current Change
    DELIMITER("delimiter", String.class, "\n"),
===
    DELIMITER("delimiter", String.class, "\r"),
>>> Current Change
```

It works. But the properties from `connectToDevice(address, properties)` is ignored in my case.

I will look for the problem tomorrow after I manage to connect the bluetooth device with A2DP.

kenjdavidson, 2020-12-08:
It depends how the React Native converts those data types to native.  They eventually need to be in a `ReadableMap` so that they can be converted to `Properties` or `Dictionary` (ios).   I have zero problem with you needing to create the `new Map` in order to customize properties.  

Are you confirming that:

```
let properties = new Map();
properties.set("delimiter", "1");

let bluetoothDevice = RNBluetoothClassic.connectToDevice(address, properties);
```

is successfully "compiled" in the application?

If this works on the React Native/Typescript side, what is actually provided to the `connectToDevice` native method?  Is the same map content not there?

kidfrom, 2020-12-08:
Yes, properties using `new Map()` successfully compiled.

By `connectToDevice` native method, do you mean on the `BluetoothModule.js`?

BluetoothModule.js
````
    connectToDevice(address, properties = new Map()) {
        console.log(typeof properties);
        return __awaiter(this, void 0, void 0, function* () {
            let connected = yield this._nativeModule.connectToDevice(address, properties);
            return new BluetoothDevice(connected, this);
        });
    }
```

The output is `object` both for
```
let properties = new Map();
properties.set("delimiter", "1");
bluetoothDevice = await RNBluetoothClassic.connectToDevice(address, properties);

or

bluetoothDevice = await RNBluetoothClassic.connectToDevice(address, { delimiter: "1" });
```

Anyway, it's false positive, sorry After I reverted back any changes I made to the `BluetoothModule.js`. I could not reproduce the issue by using `bluetoothDeice = await RNBluetoothClassic.connectToDevice(address, { delimiter: "1" })`.
```
[Tue Dec 08 2020 20:51:08.217] ERROR bluetoothConnect(): TypeError: properties.get is not a function. (In 'properties.get("delimiter")', 'properties.get' is undefined)
```

kidfrom, 2020-12-08:
The problem left is the `properties` / `{ delimiter: "1" }` is ignored.

kenjdavidson, 2020-12-08:
I mean in the `RNBluetoothClassic.java` file.  There is a `@ReactMethod connectToDevice` that is where we need to ensure the properties is converted.  It really makes no sense, the `StandardOptions` accepts a `Properties` object.  How does `Properties` not have a get method?  It's Java, so if it's a `Properties` object, it MUST have the `get()` method.

I think we're crossing wires a little.

kidfrom, 2020-12-08:
RNBluetoothClassic.java
```
<<< Current Change
L729
===
L729                System.out.println(properties.get("delimiter"));
>>> Incoming Change
```

The triggerer:
```
bluetoothDevice = await RNBluetoothDevice.connectToDevice(address, { delimiter: "x" });
```

The output `from debugger`
```
I/System.out: x
```

It works as expected.

However

I tried to make change to [`DelimitedStringDeviceConnectionImpl.java` L47](https://github.com/kenjdavidson/react-native-bluetooth-classic/blob/73bb0d1cb89e5c0ca258d6dc29da50f00ffd43a1/android/src/main/java/kjd/reactnative/bluetooth/conn/DelimitedStringDeviceConnectionImpl.java#L47)

```
<<< Current Change
L47        this.mCharset = StandardOption.DEVICE_CHARSET.get(properties);
===
L47        System.out.println("a: " + this.mDelimiter + " " + properties.get("delimiter"));
L48        this.mCharset = StandardOption.DEVICE_CHARSET.get(properties);
```

The expected output is
```
I/System.out: a: x x
```

The actual output is
```
I/System.out: a:  x
```

I think the issue is `this.name()` on `StandardOption.java` L125 returns "DELIMITER" instead of "delimiter".

Case close.

kenjdavidson, 2020-12-08:
Ya, that would definitely be the issue, I was confused as to whether it was looking at `this.name()` or `this.code()`.  It's hard attempting to determine what it should be when using different languages (Java vs Javascript vs Swift) with development conventions.

Java ENUM = Caps
Swift ENUM = Camel
Javascript = Snake

It's really just super annoying to choose the best option.   From what you're saying `connect({ DELIMITER: 'a' });` should work?

kidfrom, 2020-12-08:
> Ya, that would definitely be the issue, I was confused as to whether it was looking at `this.name()` or `this.code()`. It's hard attempting to determine what it should be when using different languages (Java vs Javascript vs Swift) with development conventions.
> 
> Java ENUM = Caps
> Swift ENUM = Camel
> Javascript = Snake
> 
> It's really just super annoying to choose the best option. From what you're saying `connect({ DELIMITER: 'a' });` should work? We can always update so that it looks at both
> 
> * Name()
> * Code()
> 
> to make it a little more friendly. Although less type friendly.

I did not test `connect()` as I try to avoid Android API only (correct me if I am wrong, that there is common API to get BluetoothDevice without connecting first).

But if I take a look at `connect()`, it relies on `connectToDevice()`. I am assuming same behavior for `connect({ DELIMITER: 'a' })`. It should work.

kenjdavidson, 2020-12-08:
Yes, the JavaScript functions push through to their native counterpart.

with regards to finding UNPAIRED devices you need to use the Discovery methods.  It's all documented and I've confirmed it's all working in the sample app.

kidfrom, 2020-12-08:
> * Code()

Anyway, regarding this.

Do you happen to use `this.code()` somewhere? In my opinion, the `code` attribute make the code less readable.

> Yes, the JavaScript functions push through to their native counterpart.
>
> with regards to finding UNPAIRED devices you need to use the Discovery methods. It's all documented and I've confirmed it's [...]

Well noted. Thank you, I will be using `getBondedDevices()`.

kenjdavidson, 2020-12-08:
So where do we stand with this?  Just that we need to:

- Update documentation to DELIMITER?
- Update StandardOptions to use both name() and code() for sanity sake?
- Both (which I'm not against)

but other than that things are working as expected?

kidfrom, 2020-12-08:
> So where do we stand with this? Just that we need to:
> 
> * Update documentation to DELIMITER?
> * Update StandardOptions to use both name() and code() for sanity sake?
> * Both (which I'm not against)
> 
> but other than that things are working as expected?

From what I can see:
1. Some user will use this library and think "I can change the DELIMITER after connection". I think it need to have a note for this on https://kenjdavidson.com/react-native-bluetooth-classic/react-native/rn-bluetooth-device/#connect
2. The documentation for DELIMITER is correct. The `BluetoothExampleApp` project use `delimiter` instead of `DELIMITER`. I think it should be change.

> Update StandardOptions to use both name() and code() for sanity sake?

I don't think I have enough expertise to put an opinion on this, but if you need one:

I have different opinion regarding the usage of `code()`. I think we should use java.util.TreeMap to remove the case sensitivity instead of using java.util.Properties and remove `code()`.

> but other than that things are working as expected?

Yes. It works as expected, thank you.

kenjdavidson, 2020-12-08:
(1) I get what you're saying, I thought I did mention that it was only used during connection.   I have zero problem with the documentation being the only source of truth - that means I need to setup TSDoc or some other kind of automated sourcing (which I'd love to get done, again I just don't have the time).  I also assumed that since the properties are ONLY used during the connect method it was obvious that they couldn't be changed during processing.  This is why I removed the `setDelimiter` and `setCharset` methods from the original version.   It also makes no sense to change delimiter and charset in my mind, generally it should be a 1:1 for Device:ConnectOptions, maybe that's wrong, but I'd need to see some valid uses cases for this.  Also, it's as easy as just disconnecting and re-connecting with new options if it's really important.

This is also the reason I separated the development app out of the project, since I wanted to get more of the Tests/Documentation driving it than the example app.   I can't see me ever dropping the example app, since it's how I do my testing of current and new functionality using my companies devices.

(2) Ya, I think I was flip flopping back and forth on which to use Name() or Code().  But I honestly like the option of both, so I'm going to make those changes when I get a chance.

(3) Tree map would just add the requirement to have a `Comparator<String>` that does an ignores case compare - so I don't see much of an add there, so I'll probably just leave it as `Properties` and `Dictionary` (ios) since it's working as intended, just have to make the docs / sample consistent.

I appreciate the testing and the discussions, there are some good take aways from this that I can use to update when I get a chance (hopefully this weekend).

kenjdavidson, 2020-12-10:
Made all the changes we discussed in here `v1.60.0-rc5` should be available for you to test out.

closed 2020-12-10 by kenjdavidson