kenjdavidson/react-native-bluetooth-classic #75: Minor bug: RNBluetoothClassic.isConnected() return true when the Bluetooth was disabled (previously turned on and paired)

kidfrom, 2020-12-07:
**Mobile Device Environment**
Provide a list of operating systems on which this issue is relevant.
- Device: Android
- OS: API 29

**Application Environment**
Provide information about your development environment:
- React Native version: 0.63.3
- RN Bluetooth Classic version: v0.10.12

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Connect to a bluetooth device
2. Disable the bluetooth
3. use method RNBluetoothClassic.isConnected()
4. returned value is true.

**Expected behavior**
The returned value should be false.

**Additional context**
Severity: Minor.
This may lead to unexpected behavior if the app monitor `RNBluetoothClassic.isConnected()` / `RNBluetoothClassic.getConnectedDevice()`.

kenjdavidson, 2020-12-07:
Could you provide information on how the device was disconnected?  The only reason this would happen is if there is a disconnect and the status wasn't changed from `CONNECTED`.  It looks like in the  `ConnectedThread` it's not setting the status on Exception, so if you shut off the device (ungracefully) then there's a chance it would not have been updated accordingly.

If you're debugging, could you add the a call to `clearConnectedThread()` in on error?

This should be happening through the ACL device disconnected, but obviously more than likely it's not.  Most likely around this line

https://github.com/kenjdavidson/react-native-bluetooth-classic/blob/c78dfd0d243cc9c0792f363acceaa22cf9ed6e72/android/src/main/java/kjd/reactnative/bluetooth/RNBluetoothClassicService.java#L486

and let me know how it goes?

kidfrom, 2020-12-07:
@kenjdavidson I did not disconnect the device, I turned off the bluetooth feature from the phone, sorry for the ambiguity, I have edited the steps to reproduce, please take a look.

Regarding the debugging, I am interested to contribute, if you don't mind, do you have time for a Zoom meeting to show how to debug?

kenjdavidson, 2020-12-07:
Oh, ya, it's definitely not handling that situation well.  Right now it's using the `BluetoothStateReceiver` to handle when state changes occur.  So we'd either need to:

- Do what I propose above and hope that shutting off Bluetooth kills the socket and we manage that in the service
- Add a callback to the `BluetoothStateReceiver` which can set tell the Module to clear the connection

The first one is probably easier and will clear up some other issues with handling the state if the `ConnectedThread` fails.

Sorry man, I don't have time for a zoom.  If you're in Android studio, you should see the linked `react-native-bluetooth-classic` project, which you will be able to set breakpoints in just like you would your own application.  You can play around from there, or you can fork/clone the project (although you'd have to modify your code to use the local repository which I'm not sure you want to do).

If I have time tonight, I can add the line, it shouldn't really affect anyone negatively as at this point the `ConnectedThread` is dead anyhow, so it should always be clearing it up.

kenjdavidson, 2020-12-07:
Also, for the most part `v1.60.x` has been cleaned up a lot.   I would suggest starting to use the release candidates of that, if you're interested in helping test it out, rather than concentrating on `v0.10.x` which (although I'll release the bug fix) I don't plan on putting much time into.

kidfrom, 2020-12-07:
I need Android 4.4 to penetrate the market, some of the dedicated phone still use Android 4.4 and with 512 MB. Currently the product is in MVP.

Give me 15 minutes, I will considering to switch to v1.60.x.

kenjdavidson, 2020-12-07:
That's what I was worried about when I made the changes for the Android version.  To be honest, I may need to push it back down for my own company as well - I just can't remember which functionality I had it updated for.  There was one call in there that required a certain version that I just said `eff it I'll update`.  

I didn't think Android was allowing new application uploads below a certain version anyhow, but if they are, I should definitely revert it back to 4.4 being the lowest.

kidfrom, 2020-12-07:
> I didn't think Android was allowing new application uploads below a certain version anyhow, but if they are, I should definitely revert it back to 4.4 being the lowest.

In practice, I would do this:
1. Publish the app with minSdkVersion 19 / 21 to the Google Play.
2. If you decide to support an Android device but the Android device use unsupported API Level, you can do manual install or ask the manufacturer to install it for you.

I decided to use the rc version. I will deal with the Android 4.4 later after I have enough customer to hire a dedicated Android developer.

kidfrom, 2020-12-07:
Anyway, I think the BluetoothClassicExample project link you provided on README.md is broken.

kenjdavidson, 2020-12-07:
I get that, but I read somewhere (an email from Android) that they were not allowing the app to even be uploaded to the store.  This is why I did it.  If you're saying that is not the case, then there is no reason not to revert it back to 19.  I've have to go back and re-look at the email, it's entirely possible I read it wrong.

kidfrom, 2020-12-07:
I believe that should not be the case. My known competitor support Android 4.1 and up.

kidfrom, 2020-12-07:
The issue does not persist on v1.60.x. I will leave this open for future.

kenjdavidson, 2021-01-14:
I went through the codebase and reverted all the issues based on minSdk from `1.60.x`.
- Streams
- BiConsume (the java.util.Function package)
- Etc

There are no longer any failing/warning code analysis items based on min sdk. 

I'm going to close this now, I don't see myself back porting this logic to the `0.10.x` version since I'd like to just get `1.60.x` completely working and used.

I appreciate all the help you've provided.

closed 2021-01-14 by kenjdavidson