kenjdavidson/react-native-bluetooth-classic #76: onDeviceConnected not being triggered

kieranmaine, 2020-12-16:
**Mobile Device Environment**
Provide a list of operating systems on which this issue is relevant.
- Device: OnePlus 5
- OS: Android 10, OxygenOS 10.0.1

**Application Environment**
Provide information about your development environment:
- React Native version: 0.63.3
- RN Bluetooth Classic version: 1.60.0-rc.5

**Describe the bug**
When connecting a device the `onDeviceConnected` listener isn't called. However when disconnecting a device the `onDeviceDisconnected` listener is successfully called. I'm using the following code:
```
const onDeviceConnectedSub = RNBluetoothClassic.onDeviceConnected(
  (event) => {
    console.log('[onDeviceConnected]', event);
  },
);

const onDeviceDisconnectedSub = RNBluetoothClassic.onDeviceDisconnected(
  (event) => {
    console.log('[onDeviceDisconnected]', event);
  },
);
```
I've updated my AndroidManifest.xml with the following permissions:
```
<uses-permission android:name="android.permission.BLUETOOTH"/>
<uses-permission android:name="android.permission.BLUETOOTH_ADMIN"/>    
<uses-permission android:name="android.permission.ACCESS_BACKGROUND_LOCATION" />
<uses-permission android:name="android.permission.ACCESS_FINE_LOCATION" />
```

**To Reproduce**
Steps to reproduce the behavior:

1. Set up `onDeviceConnected` listener. Log event to console.
2. Connect bluetooth speaker (Bose Soundlink Mini in this case).

**Expected behavior**
The onDeviceConnected event will be logged to the console when the bluetooth speaker connects.

**Screenshots**

**Additional context**

kenjdavidson, 2020-12-17:
Yes. This was ignored on purpose

```
// aclreciver.java
// This is currently ignored for now, all Connections should be managed by the device
// telling the Module when it has successfully opened the connection
```

You shouldn't need to listen for connected devices since you will be getting a promise that resolves with the device. The disconnected was left in there as you do need to know if the device was disconnected from a source other then you're app. 

I thought I added the in the docs, that it was deprecated, guess not.  If you think it's absolutely needed I'll throw it back in next time I login.

kenjdavidson, 2020-12-17:
Sadly this does two different things on android and ios.

Android is acl connect/disconnect (open socket) 
Ios is connected (which is essentially state between paired and open socket) where it actually means something. 

So on ios it's meaningful, while on Android it doesn't really make sense with he changes from 0.60 where it was required.

kieranmaine, 2020-12-17:
Thanks for the swift reply. My knowledge of bluetooth is limited, so I'll have to dig into the Android/ios difference further to understand how this affects what I'm trying to achieve. In esscense do you mean that it isn't possible to trigger onDeviceConnected events on Android?

Just to give you some more context I'd like to start/stop location tracking when the user connects/disconnected their bluetooth device (ideally I want this to happen in the background, but it's fine to have it occur in the app for now). 

I also tried using the ``device.isConnected()` and `RNBluetoothClassic.getConnectedDevices()` but when a device was connected these didn't return the expected values.

kenjdavidson, 2020-12-17:
Ya that's not how it works. 

There are different levels of connected on both systems. 

**Unpaired** on both means that your phone has no idea the device exists. 

**Discovered** (Android only) is that the phone has found the device but has not bonded/paired.

**Paired/bonded** is when the phone now knows about the device. This is where most people (you) are confusing connected.  On both Android and ios the method `getBondedDevices` returns this list. At this state there is no data transfer. 

This is where the issue of DEVICE_CONNECTED comes in. This came from a combination of the base library and my inexperience at the start. 

When a device is paired on Android its always paired. You can get details about it and try to connect (with failure if out of range or offline).  On Android the DEVICE_CONNECTED was used to handle the actual connection process, which I've since simplified with promises.  Android is based off acl events which I've tested and cannot find any event that matches the ios bonding state event. 

When a device is bonded on ios it will not show up unless it is online and in range. Therefore DEVICE_CONNECTED can be used to refresh the `getBondedDevices` list.

**Connected** status is actually when a socket is open with the device.   DEVICE_DISCONNECTED is an all encompassing message that covers
- socket closed events on both ios and android
- device unbounded on ios

So you can see its these sublets differences that made me rething connected. 

So to do what you want you couldn't use the event anyhow on Android.  To monitor for a device you'd need to continually turn on and off discovery (Android) and listen to device_connected on ios. Not optimal.

Since this is still in candidate level I have no problem introducing breaking changes to get this working with something that makes sense. But device connected in my mind is not it. 

I believe  the only way to do this is to compared bonded devices with discovered devices. There was another request to discover a single device, which I'm not against.  

IOS would just work when the device came online with
Android would still require a `startDiscovery(address)` and would need to consolidate that list against bonded devices in order to send the event. Or return when that single device is found. 

Again it probably won't make everyone happy.

kenjdavidson, 2020-12-17:
It's sounds like you want ble where you can just request updates and it'll happen when you get near.

kenjdavidson, 2020-12-17:
You also don't want to leave discovery mode on for a long time. At this point discovery returns the full list. I'd have to add in individual discovered events for each device found.  Which didn't suit my purpose as I don't generally need to continually scan, I just check if a device is connects and either start communicating or show the failure.

kieranmaine, 2020-12-29:
I'll close this for now as I won't be able to look into this until later in the year. Thanks for the detailed responses and apologies for the late reply.

closed 2020-12-29 by kieranmaine