kenjdavidson/react-native-bluetooth-classic #77: [Help needed for iOS] A2DP support

kidfrom, 2020-12-18:
After some considerations:
1. I have limited knowledge of reading a Android or iOS Native module. So, it's a bit hard to add A2DP support directly to this library.
2. My barrier to market require Android 4.1 support, although the MVP does not require one.

I decided to write an Android Native Module for BluetoothProfile.SPP and BluetoothProfile.A2DP. I hope this helps the development for this library.

**Mobile Device Environment**
- Device: Android 4.1
- OS: API 16

**Is your feature request related to a problem? Please describe.**
My goal is to make a Walkie Talkie app with Bluetooth Microphone. BluetoothProfile.SPP will emit event to the React Native Bridge while BluetoothProfile.A2DP will call `AudioManager.startBluetoothSco()`.

**Describe the solution you'd like**
I would like to use this library with [react-native-recording](https://github.com/qiuxiang/react-native-recording).
The steps:
1. This library connects to a Bluetooth Microphone.
2. This library listens to `BluetoothSocket.getInputStream()`, this library emits the event to the React Native bridge.
3. The React Native bridge calls the `AudioManager.startBluetoothSco()` to start recording audio from the Bluetooth Microphone.
4. `react-native-recording` will start getting input from the Audio Manager.

**Additional context**
This is only for Android Native Module, I hope this can help.

Know bugs to fix:
- [ ] The AudioManager refuse to record audio from the Bluetooth Microphone if the app went to the Background Mode after 2 minutes.

RNBluetoothNMPackage.java
```
package com.satpam.RNBluetoothNM;

import android.app.Activity;
import android.bluetooth.BluetoothAdapter;
import android.bluetooth.BluetoothDevice;
import android.bluetooth.BluetoothProfile;
import android.content.Intent;
import android.util.Log;

import androidx.annotation.NonNull;

import com.facebook.react.bridge.ActivityEventListener;
import com.facebook.react.bridge.Arguments;
import com.facebook.react.bridge.BaseActivityEventListener;
import com.facebook.react.bridge.Promise;
import com.facebook.react.bridge.ReactApplicationContext;
import com.facebook.react.bridge.ReactContextBaseJavaModule;
import com.facebook.react.bridge.ReactMethod;
import com.facebook.react.bridge.WritableArray;
import com.facebook.react.bridge.WritableMap;
import com.facebook.react.modules.core.DeviceEventManagerModule;
import com.satpam.RNBluetoothNM.A2DP.A2DPService;
import com.satpam.RNBluetoothNM.SPP.SPPService;

import java.util.Set;

public class RNBluetoothNMPackage extends ReactContextBaseJavaModule {

    @NonNull
    @Override
    public String getName() {
        return "RNBluetoothNM";
    }

    private static final int getBondedDeviceRequestCode = 0;
    private Promise mPromise;

    private final ActivityEventListener activityListener = new BaseActivityEventListener() {

        @Override
        public void onActivityResult(Activity activity, int requestCode, int resultCode, Intent data) {
[... 245 lines cut ...]
        }
    }

    @Override
    public void run() {
        byte[] buffer = new byte[1024];
        int len;

        while (bluetoothSocket.isConnected()) {
            try {
                len = bluetoothSocket.getInputStream().read(buffer);
                byte[] data = Arrays.copyOf(buffer, len);
                if (len > 0) {
                    mReactContext
                            .getJSModule(DeviceEventManagerModule.RCTDeviceEventEmitter.class)
                            .emit("SPPService", new String(data));
                }

            } catch (IOException e) {
                e.printStackTrace();
            }
        }
    }
}
```

kenjdavidson, 2020-12-18:
Hey, this looks good, but I don't think there is anything I need to add in the library.  The current library gives you the ability to:

- Connect to SPP
- Listen to SPP
- Disconnect from SPP

which you're essentially replicating in your version.  You also have the A2DP service which you're using to control the AudoManager.  I don't see any real reason why you couldn't continue to use this library for your SPP management, and use your own custom library solely for the A2DP/AudioManager logic?  Unless I'm missing something, the only reason I see to merge them is just so you can use the same `nativeModules.BluetoothNMModule` in order to communicate.

All this should be possible by implementing your own custom `ConnectionConnector` and `DeviceConnection` classes.

kenjdavidson, 2020-12-18:
This will probably never work on IOS using Bluetooth Classic, as the framework locks down all audio communication by the operating system.  You'd also need to work with the Bluetooth vendor to get the MFi codes, which is an extremely annoying/expensive process.

In no way do I expect this library to be the be-all and end-all to Bluetooth communication, but in my opinion for the lengths that you're using it and your requirements, you should definitely be using a 100% custom module - specialized to your process.

kidfrom, 2020-12-18:
I get your point. Thank you for the earlier explanation regarding Bluetooth. It helps me to write my first Android Native Module.

I will start writing a custom module for the Bluetooth and audio recorder & player.

kenjdavidson, 2020-12-18:
Feel free to leave it open.  

I honestly love the idea of helping you get this project completed, and will do my best to answer any questions / provide any feedback that I can.   Some of the stuff in here I see positives to, I just need to figure out the best way to manage the additions/changes.

At this point I feel like creating a new `react-native-bluetooth-profiles` project would be better suited for this.   From what I've read there is no reason why the two cannot work together (from your sample code) it's definitely possible.  For example, in your sample code you:

`connectToDevice`

First you do this
```
BluetoothDevice bluetoothDevice = bluetoothAdapter.getRemoteDevice(address);
            a2DPService = new A2DPService(getReactApplicationContext());
            boolean a2dpIsBonded = a2DPService.createBond(bluetoothDevice)
```
which is an async action, and would just returns `true` if the process begun, and not actually if the device was paired.  Therefore it's possible you call the `connect` method before it's even paired.  These two things should be separated in my mind.  For example, you should only try connecting if the device is already paired/bonded.

Then you connect to SPP.  Does the order matter?  Does A2DP need to be connected before SPP in order for this to work?  If not, this could be done just as easily (as mentioned) with a customized `SPPAndA2dpDeviceConnection` since they are managed together, and a **connection** is only valid when both of them are complete.

Anyhow, definitely keep posting details and questions.