kenjdavidson/react-native-bluetooth-classic #78: Can this package upgrade OTA to device directly

CaanDoll, 2021-01-05:
Sry,I'm not good at bluetooth.I'd like to ask ‘Can this package upgrade OTA to device directly’.Does it mean that ``write`` something to device.Example write instructions to device and write ota config file to device. Thank you very much for your answer!

kenjdavidson, 2021-01-05:
The library doesn't have anything specifically related to OTA; but with that said the functionality you've described:

- write instructions to device (supported)
- write OTA config file to device (may required a custom `DeviceConnection` to write binary data, but supported)

It would be interesting to see how you make out with this, but I believe the library contains the building blocks you'd need to get started.

I'm always open to PR that might add functionality to make this work.

CaanDoll, 2021-01-05:
Thanks! 
You mean it provides all the basic functions of Bluetooth, and I need to compose it to realize OTA

kenjdavidson, 2021-01-05:
It should, I don't see why not. 

The standard functionality provides text based writing (configurable encoding):
- This should let you write instructions to the device

There are issues/docs on how to create a `DeviceConnection` for transfering binary data:
- This should let you save a file once you've used the correct command

You'll need to review your device functionality, and see what is required in the form of data format, but the base functionality should all be there.

Feel free to open any issues/PRs along the way so that we can get it happening.  

Just remember, I don't want to pull in any other `Connectors`, `Acceptors` or `Connections`.   I'd like to keep the base library as small as possible for the most generic purposes (ie. mine).   But we can definitely add as much documentation/samples as possible to describe how to replicate.

CaanDoll, 2021-01-06:
Well, thanks a lot for your timely reply.
I try it.

kenjdavidson, 2021-01-29:
Closing - as this isn't so much to do with the library, but more the custom device/application logic.  If a specific example can be provided and replicated we can look at adding functionality.

closed 2021-01-29 by kenjdavidson