kenjdavidson/react-native-bluetooth-classic #79: The official example "Bluetooth classic example" produce error when click a device what connect in advance

CaanDoll, 2021-01-12:
**Mobile Device Environment**
Provide a list of operating systems on which this issue is relevant.
- Device: Android
- OS: Android 11

**Application Environment**
Provide information about your development environment:
- React Native version: v0.63.4
- RN Bluetooth Classic version: 1.60.0-rc.5

**Describe the bug**
The official example ``Bluetooth classic example``.A bluetooth device that has been connected all the time. After clicking in, the error is not currently connected

**To Reproduce**
Steps to reproduce the behavior:
1. Connect a device in advance
2. Click on this device
3. See error

**Expected behavior**
See a normal connection

**Screenshots**
[image]

kenjdavidson, 2021-01-12:
The line in question:

https://github.com/kenjdavidson/react-native-bluetooth-classic/blob/228e623457c6ce50e450096996997757606e6e4b/android/src/main/java/kjd/reactnative/bluetooth/RNBluetoothClassicModule.java#L1060

is pretty straight forward

```
            if (!mConnections.containsKey(eventDevice)) {
                throw new IllegalStateException(String.format("Cannot read from %s, not currently connected", requestedEvent));
            }
```

which would lead me to believe that the device is not actually connected.  There may be an issue with:

- Your device being connected
- The sample application and your specific device

All I can do is confirm that the `BluetoothClassicExample` app is successfully connecting to my two test devices (described in the README).

```
      let connection = await this.props.device.isConnected();
      if (!connection) {
        connection = await this.props.device.connect(this.state.connectionOptions);

        this.addData({
          data: `Connection successful`,
          timestamp: new Date(),
          type: 'info'
        });
      }

      this.setState({connection});
      this.initializeRead();
```

I can see in the connection code that it's **assuming** the connection is successful.  I should definitely be wrapping the `this.initializeRead()` in a check for connectivity as `device.connect` could return with a failure (although any failure here SHOULD push to the following `catch` block.

With regards to your comments:

- What does `connect to device in advance` mean?  Like you're just clicking on it from the list and then the connection screen pops up? 
- What type of device are you connecting to?  Can you link to the specs?

Could you provide some more debugging of the application with respect to your own device?  You'll want to place some breakpoints on:

https://github.com/kenjdavidson/react-native-bluetooth-classic-apps/blob/3731d852d87c6b7d9642ee4b4b67c8a421b9853a/BluetoothClassicExample/src/connection/ConnectionScreen.js#L65

to see what the result of the `device.connect` is, and why it's not being thrown on error.

kenjdavidson, 2021-01-12:
With this said, I've been wrestling with the idea of NOT requiring an active connection when attempting to set the `addListener` for reading on a device.  The problem with this being, that if you accept a listener BEFORE the device is connected, you now have to:

- Maintain a list of waiting listeners
- When the device finally connects attach said listener

This really didn't fit my personal use cases, so I left it out.  If you think (and it gains traction) that this should be possible, it would make sense to add it in.

CaanDoll, 2021-01-13:
Thx!

``What does connect to device in advance mean? Like you're just clicking on it from the list and then the connection screen pops up?``

My devices are a pair of bluetooth earphones. Usually when the mobile phone is turned on Bluetooth, the headset will be automatically connected when it is taken out of the box. At this time, open the app and it will be connected in advance.

And why I am sure that it is connected is because the status bar at the top of the phone can already see the earphones lit, and the music can be played through the earphones

kenjdavidson, 2021-01-13:
Ya that means it's connected with, most likely, the a2dp profile. This library doesn't support that. This library only supports server and Bluetooth sockets that you need to open manually using the #connect method.

There are already a couple tickets open regarding this, so I'm going to close this one off.

CaanDoll, 2021-01-13:
But the official example "Bluetooth classic example" click to connect the device to succeed. And it can trigger the connection instruction of the Bluetooth headset, and there is also a voice announcement, the system is indeed connected.
I don’t know if this is supported or not QAQ

kenjdavidson, 2021-01-13:
I can reopen for later. But I have zero use for this case so I don't see me spending any time on it in the immediate future.

- if the example app initiates the connection, that's a bonus 
- if you mean it can send a command/message to the headset to initiate a message then that's cool.

But the point remains if there is not Bluetooth socket opened and passed to the module you will not be able to read from it. 

You're more than welcome to do some digging and debugging. And I'm More than happy to have discussions and manage pull requests. But we just got locked down for another month and between home school and work, I just don't have time.

I'm sorry.

CaanDoll, 2021-01-13:
Don’t say sorry, I really appreciate your response.
I was wrong before, the connected device is the system's own action.
This time I use your example and my own code to try to disconnect first by system, but keep the paired state, to execute the ``connect()``, I found that the device gave me the feedback is ``paring!``.
Very strange, I don’t know if the connect parameter is wrong or something, but I think your library should be able to support it.

kenjdavidson, 2021-01-29:
Is there any follow up on this? From what I can tell it seems like it's working as designed.
If you've got a suggestion to change how it functions overall, a new task with substantially more information might be the best plan.

closed 2021-02-03 by kenjdavidson