kenjdavidson/react-native-bluetooth-classic #81: Error java.io.IOException: read failed... when trying to connect to a device

Rm-lee, 2021-01-13:
**Mobile Device Environment**
- Device: lgg8,samsung a21
- OS: Android 10

**Application Environment**
- React Native version: 0.63.4
- RN Bluetooth Classic version: 1.60.0
- BuildToolsVersion = "29.0.2"
- CompileSdkVersion = 29
- TargetSdkVersion = 29

**Describe the bug**
When trying to connect to any device be it earbuds, phones, laptops, obdII scanners using the example project, pairing works but then immediately throws error java.io.IOException: read failed, socket might closed or timeout, read ret: -1 after pairing.

Also while having a known connection to another bluetooth device and trying to use getConnectedDevices() returns empty and throws no errors.

**Expected behavior**
Click on a bonded device or found unbonded device and be able to pair and complete connection without error.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
The getBondedDevices() works fine its only the getConnected methods and connect that dont seem to work for some reason.

kenjdavidson, 2021-01-13:
Hey man, since Bluetooth is so fickle on a device to device basis, I'm going to need you to provide some debugging information along with this.   Can you link to your the repository in which you're having this issue?  Your comment about pairing works but then immediately throws an IO exception is confusing.

- Are you using the library to pair and the pair fails?
- Are you pairing through the Android system settings and then using the `device.connect()` which is then failing?
- Are you failing when attempting to set `onDeviceRead()` after the `device.connect()` returns?

What do you mean by **having a known connection**.  The call to `getConnectedDevices()` ONLY returns devices which have an active `BluetoothSocket` open.  So if `connect` is failing then there is no way that you have an active `BluetoothSocket` and therefore you are not connected to any devices (expected empty list).

Can you confirm that we are using the same language in terms of the library (this is the cause of a large number of questions for some reason):

- Paired means that it's paired in android (getBondedDevices()
- Connected means that there is a successful `BluetoothSocket` open and listening (`getConnectedDevices()`/`getConnectedDevice()`)

 Some things that I've seen from other issues opened recently:

- Ear buds have problems, this is a known issue that I haven't really thought about.  But most of the more recent issues opened have been because of this.  I'm hoping that one of you can do some debugging and find the issue/resolution.
- I've never attempted a connection to a laptop for the purposes of communication.  Are you connecting or accepting on the device?  There is a known issue with `accepting` on the device that I need to look into.  What server software are you using to accept the socket connection on the laptop?
- The car scanner, again I've never tried, and since I don't have one, nor have a way to get my hands on one, I see this as being the least likely thing that gets looked at. 

As mentioned in other issues, you're more than welcome to debug and provide:

- Findings
- Discuss preferred resolutions
- submit pull requests

but at this point, my time is limited for debugging things that I don't use the library for myself, nor have a way in which to test.

batkhai290499, 2021-01-25:
[image]
here. maybe like me

kenjdavidson, 2021-01-25:
Thanks for the screen shot. But without any context of how or what you're doing hwne you get it it's not too helpful.

Can you show me the 
- example app screen
- the Bluetooth settings screen
- walk step by step on what you're trying

That error could be anyhting.  It's coming from the Android native side during the connect to device, but other than that I'm going to need substantially more info. 

What device are you attempting connection? 
Is it paired? 
Is it turned on? 
Have you attempted to debug the native code? 

I test with a few devices and have never seen this or had issues. So I'll need a little more.

kenjdavidson, 2021-01-29:
Any word on this or should I consider it working as intended?

Rm-lee, 2021-01-29:
Sorry for the lack of info. My issue was with getting your example app working. I gave up on it and just created my own from scratch using RNBC and things worked fine. I'm not sure what the issue was exactly with your example. The only issue i have now is with write() not working and needing to use writetodevice..this is off the top of my head I believe thats the methods name.

kenjdavidson, 2021-01-29:
Ya, the `BluetoothClassicExample` isn't the best - and a bunch of people have had problems setting it up.  This is why I moved it out of the main project, in the hopes that it would stop causing so many issues due to it's configuration and required setup.  I'm glad that the library is working with your own app.

You mean that `device#write(data)` doesn't work, but the direct `RNBluetoothClassic.writeToDevice(address, data)` is working?

Rm-lee, 2021-01-29:
Yeah I probably should haven't opened the issue there instead. But yes I had to switch to the latter. I'm working now so i cant look to see what the error with device.write was

kenjdavidson, 2021-01-29:
Right on.  Good luck and keep me posted.

closed 2021-03-03 by kenjdavidson