kenjdavidson/react-native-bluetooth-classic #82: How can I get generic available devices in IOS without pair? (Help needed)

WO-Mehul, 2021-01-19:
**Mobile Device Environment**
- Device: iPhone6s
- OS: iOS14.3

**Application Environment**
- React Native version: v0.63.4
- RN Bluetooth Classic version: 1.60.0-rc.6

**In iOS platform, How can I get generic available devices without pair?**
**Generic device like :** Windows 10 Bluetooth, Mac mini Bluetooth etc...

The Android app works as expected but iOS needs help.

If I am not wrong we don't need an MFi string for this type of device. Right?

I am asking it because I have successfully parried Windows 10 Bluetooth through the IOS Bluetooth setting screen. and after call getBondedDevices() method but I am not getting any parried device. it is always retune blank array.

kenjdavidson, 2021-01-19:
I'm not sure I understand your use case.

Let's say for example that we updated the `getBondedDevices` to use the `CoreBluetooth` library which might make this possible.  To what end?

You still cannot make a serial connection with the device due to the lack of MFi protocols.

kenjdavidson, 2021-01-19:
I've tried looking into whether the new changes to `Core Bluetooth` to allow for classic connection would be helpful:

https://developer.apple.com/documentation/corebluetooth/using_core_bluetooth_classic

But haven't had much time to actually play with the sample to see if it would fit in.   I see no point providing a list of devices to the user, which they cannot actually connect. 

```swift
func centralManagerDidUpdateState(_ central: CBCentralManager) {
		// In your application, you would address each possible value of central.state and central.authorization
		switch central.state {
		case .resetting:
			os_log("Connection with the system service was momentarily lost. Update imminent")
		case .unsupported:
			os_log("Platform does not support the Bluetooth Low Energy Central/Client role")
		case .unauthorized:
			switch central.authorization {
			case .restricted:
				os_log("Bluetooth is restricted on this device")
			case .denied:
				os_log("The application is not authorized to use the Bluetooth Low Energy role")
			default:
				os_log("Something went wrong. Cleaning up cbManager")
			}
		case .poweredOff:
			os_log("Bluetooth is currently powered off")
		case .poweredOn:
			os_log("Starting cbManager")
// It might be possible to use the BTConstant for SSP here? Although who knows if the devices support it.
			let matchingOptions = [CBConnectionEventMatchingOption.serviceUUIDs: [BTConstants.sampleServiceUUID]]
			cbManager.registerForConnectionEvents(options: matchingOptions)
		default:
			os_log("Cleaning up cbManager")
		}
	}

	func centralManager(_ central: CBCentralManager, connectionEventDidOccur event: CBConnectionEvent, for peripheral: CBPeripheral) {
		os_log("connectionEventDidOccur for peripheral: %@", peripheral)
		switch event {
		case .peerConnected:
			cbPeripherals.append(peripheral)
		case .peerDisconnected:
			os_log("Peer %@ disconnected!", peripheral)
		default:
			if let idx = cbPeripherals.firstIndex(where: { $0 === peripheral }) {
				cbPeripherals.remove(at: idx)
			}
		}
		tableView.reloadData()
	}
```

Although this would require changes to the IOS side so that the devices were managed using both Core Bluetooth and EAccessory `getBondedDevices()` which again, doesn't help much since even if you see the device, you cannot connect to it.  

Although, with that said, it is possible now to write your own connection logic, so there would be nothing stopping you from wrapping a BLE connection within this library now.  

Let me think about it.

kenjdavidson, 2021-01-29:
Any word on this, or should I consider it working as intended?

WO-Mehul, 2021-01-30:
> Any word on this, or should I consider it working as intended?

Thanks for the reply, 
I have changed my mind and used "react-native-ble-plx" library instead of 'react-native-bluetooth-classic' and it is working for me without 'MFi'.

kenjdavidson, 2021-01-30:
if you can use BLE with your devices that's probably for the best.
Thanks for the update.

closed 2021-01-30 by kenjdavidson

psyzz, 2021-03-15:
Hello @kenjdavidson , 

sorry to reopen this issue, but i'm facing a similar issue.

What i'm trying to do : get generic available devices around with iOS (and Android but it seems easier), concretely i would like to know if the user is near his car (which often uses old bluetooth versions, and assuming that he had already paired his iOS phone with his car).

if I'm not mistaken, with this librairie i can't do that ?  ( `getBondedDevices` return an empty array, and `getConnectedDevices()` return `[TypeError: this._nativeModule.getConnectedDevices is not a function. (In 'this._nativeModule.getConnectedDevices()', 'this._nativeModule.getConnectedDevices' is undefined)]` )

kenjdavidson, 2021-03-15:
`getBondedDevices` returns a list of (yup) the currently bonded devices.  These are the devices that the user has gone through the process of pairing with.  How this works is pretty straight forward:

**Android**
- User pairs device
- User sees device as paired in System / Bluetooth
- User calls `getBondedDevices()`
- User gets a list of devices

Now, where you're having the problem (and there is literally nothing I can do)

**IOS**
- User pairs the device with IOS (sweet it works)
- User sees device in System / Bluetooth (great)
- User calls `getBondedDevices()`
- Empty

This is due to how Apple manages Bluetooth Classic devices through MFi protocols.   Yes, that device is connected and paired, but Apple won't let you see / use it without the MFi Protocols.

So lets take your example:
- You pair your phone to your car (sweet it works)
- Behind the scenes if your car (Ford for example) is MFi compliant, which it probably has to be, there will be protocols that Apple is aware of, say `com.ford.apple-protocol`, `com.ford.manage-contacts`, `com.ford.manage-audio`, etc.
- As a vendor, you need to go to Ford and ask for those.  Without them you don't get to work with the connected device (hence you see an empty list)

This is just how Mfi works, it's annoying and stupid and yet another reason why I can't stand Apple as a company.  But such is life.

kenjdavidson, 2021-03-15:
Feel free to browse through:

- https://developer.apple.com/accessories/Accessory-Design-Guidelines.pdf
- https://mfi.apple.com/en/how-it-works.html

```text
How MFi Certification Works
Here is an overview of the steps required to create an MFi accessory, from conception to production.

1. Product Plan
Submit a Product Plan for an accessory that integrates MFi licensed technology.

2. Development
Design, develop, and test the accessory. Procure any MFi components as needed.

3. Certification
Use Apple’s certification tools to help ensure optimal user experience and interoperability. Submit production-ready samples and packaging materials for review.

4. Mass Production
Upon completion of certification approved by Apple, begin manufacturing and sales.
```

- https://medium.com/theengineeringgecko/external-accessory-framework-what-why-how-d0a9fad52271

Essentially, unless you go to every car manufacturer and ask/pay for their protocol string, you're out of luck.  Or you use BLE, which doesn't have these issues.

kenjdavidson, 2021-03-15:
With regards to the `getConnectedDevices`, you are correct.  I missed implementing that function when I moved the Android logic over to IOS.  My bad.  I'll try to get to it at some point, I've opened #92 for this purpose.

Although, `getConnectedDevices` will/should only return the Device information for those devices you've already established an RFCOMM socket with - which requires MFi protocols anyhow. 

It's really just a convenience method when connecting to multiple devices (which to be honest no one has really even documented that they are using).  In generally most applications connect to 1 device at a time, and therefore don't really even need it:
- They remember the Address which was connected and calls `isDeviceConnected(address)` instead
- Just loop through `getBondedDevices()` and re-connect to the device they were last connected

psyzz, 2021-03-15:
Thank you for your detailed answer , i will try to use BLE but my guess is that all cars may not have bluetooth 4.0

kenjdavidson, 2021-03-15:
No problem man, it's been a "wonderous" ride attempting to learn about Apple and how they deal with Bluetooth Classic.  So I feel your pain and sadness.

Good luck!! Hit me up anytime, I'll do my best to help.