kenjdavidson/react-native-bluetooth-classic #84: NPE during attempt connection with device

holakmateusz, 2021-01-19:
- Device: [Motorola Moto g7 Power]
- OS: [Android 10, API 29]

- React Native version: [0.63.1]
- RN Bluetooth Classic version: [1.60.0-rc.6]

**Describe the bug**
When I try to connect with device, then an error appears: NPE in native module (android).
In details error comes from method 
`public void connectToDevice(String address, ReadableMap parameters, Promise promise)
in RNBluetoothClassicModule
`

**To Reproduce**
Steps to reproduce the behavior:
1. Start discovery
2. Try to connect with device with invoking:
 `  connect = async (address: string): Promise<BluetoothDevice> => {
    return await RNBluetoothClassic.connectToDevice(address);
  };` 
3. See error as mentioned above.

**Expected behavior**
An error not appear, get default parameters or the method should require additional parameters.

I'm not sure i made something mistake so please read it and tell me am i wrong or it's just bug in code.
This error appears when i try to establish connection with real device by invoking code as mentioned in 2 step.
Thank you for any answer.

kenjdavidson, 2021-01-19:
First off, if you're attempting to connect to an unpaired device, I don't know what will happen.  I've never attempted it.   Looking at the code for `connectToDevice` it's definitely not checking that the device is paired before attempting it.

(but you just added stack trace, so it doesn't look like that)

The line in question `Properties properties = Utilities.mapToProperties(parameters);`  definitely looks like the second parameter of `public void connectToDevice(String address, ReadableMap parameters, Promise promise)` is null.

I thought that I had defaults set in both:

`BluetoothDevice.ts#connect()`
`BluetoothModule.ts#connectToDevice()`

The latter being the most important, but I guess not.   You should be able to get around it by 

`RNBluetoothClassic.connectToDevice(address, {});`

Super shady, but should be doable until I add in the defaults.

Good catch!

holakmateusz, 2021-01-19:
I think i add some parameters and check what happen. I can tell you that i try to connect with device that was paired before (it is already listed as known device).

kenjdavidson, 2021-01-19:
Here's the code that I use to connect

```
 let connection = await this.props.device.isConnected();
      if (!connection) {
        this.addData({
          data: `Attempting connection to ${this.props.device.address}`,
          timestamp: new Date(),
          type: 'error',
        });

        connection = await this.props.device.connect(
          this.state.connectionOptions
        );

        this.addData({
          data: 'Connection successful',
          timestamp: new Date(),
          type: 'info',
        });
```

And I do indeed have:

```
this.state.connectionOptions = {
    DELIMITER: '\r'
}
```

where `\n` is the default, so you probably just need to fake it out.

I'll update `BluetoothModule.ts#connectToDevice()` with the default empty options:

```
  async connectToDevice<T extends StandardOptions>(
    address: string,
    options?: T = {}
  ): Promise<BluetoothDevice> {
    let connected = await this._nativeModule.connectToDevice(address, options);
    return new BluetoothDevice(connected, this);
  }
```

Which should clean up your issue.

kenjdavidson, 2021-01-19:
Might be wise to add a null check on the native side as well

```
Properties connectionProperties = parameters != null ? Utilities.mapProperties(parmaeters) : new Properties();
```

Will take a look what the IOS side does.

kenjdavidson, 2021-01-20:
Was able to replicate and resolve this on Android.

**v1.60.0-rc.7** available.

closed 2021-01-20 by kenjdavidson