kenjdavidson/react-native-bluetooth-classic #86: How to send bytes to device

sopholtruong, 2021-01-25:
Hi all,
I using React Native Bluetooth Classic version 1.60.0,
When i sent byte array to device connected then data recieve wrong
Ex:
=> let buffer= Buffer.alloc(10, **0xFE**);
then data recive is:
=> **ef bf bd ef bf bd ef bf bd ef bf bd ef bf bd ef bf bd ef bf bd ef bf bd ef bf bd**
How to fix it?
Thanks

kenjdavidson, 2021-01-25:
Ya, there was definitely an issue that was going to mess this up.  I've updated the writing to:

- If message is a Buffer it will leave it as is
- If message is anything else it will `Buffer.from(message, encoding)` (not really your issue)
- Base64 encode to transfer to React Native bridge

I've also added `ByteArrayDeviceConnectionImpl` which you'll need to implement on the Native side so it actually sends the byte[] instead of converting to String to send.

Could you please give `1.60.0-rc.12` a run and see if you're seeing some better stuff?

sopholtruong, 2021-01-26:
> Ya, there was definitely an issue that was going to mess this up. I've updated the writing to:
>
> * If message is a Buffer it will leave it as is
> * If message is anything else it will [...]

**You’re the best. Thank you so much.**

kenjdavidson, 2021-01-26:
LOL wait until it works to say that!!  I have no way to really test it except confirm that the same 10 diamond question marks (ahha) are displaying on the other side.  Keep me posted!

kenjdavidson, 2021-01-29:
Closing and continuing discussion on #87

closed 2021-01-29 by kenjdavidson