kenjdavidson/react-native-bluetooth-classic #87: > Ya, there was definitely an issue that was going to mess this up. I've updated the writing to:

sopholtruong, 2021-01-26:
> Ya, there was definitely an issue that was going to mess this up. I've updated the writing to:
> 
> * If message is a Buffer it will leave it as is
> * If message is anything else it will `Buffer.from(message, encoding)` (not really your issue)
> * Base64 encode to transfer to React Native bridge
> 
> I've also added `ByteArrayDeviceConnectionImpl` which you'll need to implement on the Native side so it actually sends the byte[] instead of converting to String to send.
> 
> Could you please give `1.60.0-rc.12` a run and see if you're seeing some better stuff?

**You’re the best. Thank you so much.**
I updated to version 1.60.0-rc.12 but data recieved on device wrong.
When i send array buffer with data < 0x80 then data recieved is correct, but i send array buffer with data > 0x80 then data recieved is wrong.
[image]
[image]

_Originally posted by @sopholtruong in https://github.com/kenjdavidson/react-native-bluetooth-classic/issues/86#issuecomment-767216240_

kenjdavidson, 2021-01-26:
Yup. If you're using the default `DelimitedStringDeviceConnectionImpl`? If so then you're getting the correct data for that connection type. It's only going to send the utf8 version of that data, which looks correct

100 x7e 

You need to take it further now. Look at the discussion topic on how to send hex and implement a `ByteArrayDevceConnectionImpl` in order to get bytes working.

Since react native only allows strings across the bridge, there isn't much to do about that. But it should be possible now that you've got what you have now.

sopholtruong, 2021-01-26:
> Yup. If you're using the default `DelimitedStringDeviceConnectionImpl`? If so then you're getting the correct data for that connection type. It's only going to send the utf8 version of that data, which looks correct
> 
> 100 x7e
> 
> You need to take it further now. Look at the discussion topic on how to send hex and implement a `ByteArrayDevceConnectionImpl` in order to get bytes working.
> 
> Since react native only allows strings across the bridge, there isn't much to do about that. But it should be possible now that you've got what you have now.
> 
> **edit**
> 
> Just read the difference. Not sure if my previous comment holds true.
> 
> This week is going to be super busy for me and I probably won't look into it. But I'll try to take a peak at this example on the weekend.

**Thank you very much!!!!**

kenjdavidson, 2021-01-29:
Hey, any luck getting this going?

sopholtruong, 2021-01-30:
> Hey, any luck getting this going?

When i send string in spite of a buffer array with value > 127 (ascii code) and set charset utf-8, then recieve data still wrong. I think it is right when data is a ascii code (don't support ascii extended)

kenjdavidson, 2021-01-30:
Ascii is the default Charset. So this may be dumb, but are you setting the right Charset on connect?

```
Await this.props.device.connect({ charset: 'utf8' })
```

?  Or have you got got the byte array connection setup?

kenjdavidson, 2021-01-30:
You have access to all the Android standard Charset

https://developer.android.com/reference/java/nio/charset/Charset#forName(java.lang.String)

If you need more then yes. This will need development. Let me know.

sopholtruong, 2021-01-30:
> Ascii is the default Charset. So this may be dumb, but are you setting the right Charset on connect?
> 
> ```
> Await this.props.device.connect({ charset: 'utf8' })
> ```
> 
> ? Or have you got got the byte array connection setup?

All right,
I had set charset utf8 when connect, send byte array or string, so still wrong.
I see log of framework  (android), base64 when set default / utf8 is like
 :(

kenjdavidson, 2021-01-30:
Can you provide a little more information on what/where it's wrong?

Right now it should be doing this:

- `device.connect({ charset: 'utf8' })` which will connect to the device using `Charset.forName('utf8')` which I think is right.  Maybe `utf-8` I'm not 100% sure.
- Then you send data, you're saying it doesn't work regardless of what you send, so based on what we've been discussing `device.write(Buffer.alloc(10, 0xFE, 'hex')` 
- This will get Base64 encoded in order to send to native, which should be `FE FE FE FE FE FE FE FE` which is what I was seeing when I attempted sending the buffer.  Are you seeing something else in the `RNBluetoothModule#writeToDevice()` method?
- That data gets converted to `byte[] bytes = Base64.decode(message, Base64.DEFAULT)` which should result in your byte[] so it should be `[FE, FE ,FE ,FE ,FE ,FE ,FE ,FE, FE, FE]` or the <?> character in the debugger.

Ok, so at this point It might be missing something:

- Now when you send that to `DelimitedStringDeviceConnectionImpl` the `AbstractDeviceConnection` is taking over and writing the byte[] directly to the device.

You can try updating

https://github.com/kenjdavidson/react-native-bluetooth-classic/blob/9d38f0e79696b97afe342b8ea65d14ef55d9095b/android/src/main/java/kjd/reactnative/bluetooth/conn/AbstractDeviceConnection.java#L173

the `AbstractDeviceConnection#write()` method to do re-convert the bytes:

- Convert to String from bytes using Encoding
- Convert back to bytes to write to `mOutputStream`

Let me know.

kenjdavidson, 2021-01-30:
May need to create the matching 

```
    /**
     * Handle incoming data from the device.  The bytes provided are in raw form and can be modified
     * prior to saving, handling, etc.
     *
     * @param bytes the raw byte[] read from the device
     */
    protected abstract void receivedData(byte[] bytes);
```

abstract `writeData(byte[] bytes)` for the writing portion.

sopholtruong, 2021-02-01:
> May need to create the matching
>
> ```
>     /**
>      * Handle incoming data from the device.  The bytes provided are in raw form and can be modified
>      * prior to saving, handling, etc.
>      *
>      * [...]

Hey,
I using **device.write(data)** instead of **RNBluetoothClassic.writeToDevice(IdAddress, data)** then data receive on device all right.
Thank you so much :)

kenjdavidson, 2021-02-01:
Wait.....

`device.write(data)` works

but

`RNBluetoothClassic.writeToDevice(address, data)` does not?

And it's the exact same data?

sopholtruong, 2021-02-01:
> Wait.....
> 
> `device.write(data)` works
> 
> but
> 
> `RNBluetoothClassic.writeToDevice(address, data)` does not?
> 
> And it's the exact same data?
> 
> That doesn't really make any sense, all the `device.write` does is pass it's own address through to the `writeToDevice`. I'm glad it's working though!

Oh No!,
I was confused, `device.write(data)` and `RNBluetoothClassic.writeToDevice(IdAddress, data)` all right to send, maybe byte array data to init wrong before send data to device. Sorry for bothering you!

kenjdavidson, 2021-02-01:
Right on, I'm glad that it's working!! 
Thanks for letting me know.

closed 2021-02-01 by kenjdavidson