kenjdavidson/react-native-bluetooth-classic #88: Continuous bluetooth scanning mode

holakmateusz, 2021-02-03:
**Mobile Device Environment**
- OS: [Android,iOS]

**Is your feature request related to a problem? Please describe.**
I have question related to the bluetooth scanning: 
`RNBluetoothClassic.startDiscovery()`
It is asynchronous method, but what I try to suggest, it is increasing scanning time. On Android can be limited, but we can increase time, it helps in development in many devices. For example if some device is available only temporary it can be difficult to make communication between devices with this API.

**Describe the solution you'd like**
Scanning devices can be limitted for some bigger period, now its 12 seconds i guess. The best solution can be solution with unlimited scanning, and another method "stopScanning" should cancelling scanning devices.

**Describe alternatives you've considered**
Method should allow to put additionaly argument: duration or something like that, what indicate how long should work scanning devices.

**Additional context**
Please let me know if my feature request is correct or not, because maybe i used wrong API related with scanning.

kenjdavidson, 2021-02-03:
https://developer.android.com/reference/android/bluetooth/BluetoothAdapter?hl=en#startDiscovery()

Android does not support or provide a method for extending, I would suggest that in your app you listen for the response and if you don't get the device you're looking for start discovery again.  This isn't a limitation of the library, but of Android itself.

```
Start the remote device discovery process.

The discovery process usually involves an inquiry scan of about 12 seconds, followed by a page scan of each new device to retrieve its Bluetooth name.

This is an asynchronous call, it will return immediately.
```

If you've got any other suggestions, let me know.  But at this point I don't feel like this is an option.

kenjdavidson, 2021-02-03:
On the IOS side, there is no concept of discovery as that is required to be done by the IOS system itself.

closed 2021-02-03 by kenjdavidson