kenjdavidson/react-native-bluetooth-classic #89: Connection error after app reload in development mode

andreibursuc97, 2021-02-14:
**Mobile Device Environment**
Provide a list of operating systems on which this issue is relevant.
- Device: Zebra TC26
- OS: Android 10

**Application Environment**
Provide information about your development environment:
- React Native version: v0.63.4
- RN Bluetooth Classic version: 1.60.0-rc.12

**Describe the bug**
The library works as expected when I first open the app and  I connect the bonded bluetooth device. The problem appears when I reload the app during development, when I try again to create a connection I get a connection error instead of the app letting me know that the device is already connected. I checked and that happens because the map `mConnections` from the class `RNBluetoothClassicModule` is reinstantiated every time the app reloads and the application no longer knows that the connection already exists. I fixed that locally by making the map static and instantiating it outside the constructor. Now I just wanted to ask if this is the intended behavior and I should actually disconnect the device at component unmount. Thank you.

**To Reproduce**
Steps to reproduce the behavior:
1. Connect the device to app
2. Reload the app
3. Try again to connect to device
4. First time error appears

**Expected behavior**
No error should appear because the device is already connected.

kenjdavidson, 2021-02-14:
It looks like I removed (for some reason) the closing of the connections when the application is paused or destroyed.

```java
    @Override
    public void onHostDestroy() {
        if (BuildConfig.DEBUG)
            Log.d(TAG, "onHostDestroy: stop discovery, connections and unregister receivers");

        // Missing disconnect from all connections
        mAdapter.cancelDiscovery();
    }
```

Which is really weird since the device that I use seem to be doing it anyhow.  Both the devices I use disconnect the `BluetoothSocket` when the application closes.  Making it possible to automatically re-connect when I start back up.

So there is definitely room for improvement.

I don't particularly think that creating the map statically is a good idea.  It may work, but it's opening things up for trouble down the road, I also don't think the React Native framework is meant to do that.

In my opinion, a better method (and the method I just haven't got around to working on) is hooking into the Android and IOS native lifecycle hooks.  I just haven't had a chance to look into the best way to do this, there is no direct access (from what I've seen):

- `onCreate()` and `onSaveInstanceState()` for accessing the `Bundle`
- https://developer.apple.com/documentation/uikit/app_and_environment/managing_your_app_s_life_cycle

But I believe this is a better way of managing, mainly because this also incorporates things into the background processing, which I also haven't had a chance to look at.  So it brings up more things:

1) This has to be configurable, since:

- Want it to persist (you)
- Want it to disconnect and auto reconnect
- Want it to disconnect and manual reconnect (me)

2) Should the connection be handed off to a background service?  In my opinion yes, this should be done as the standard so that data can continue to be received and stored even when the application is off screen (there is a task for this).   I just don't have the time to work on it at this point in time.

3) How is it customized?  Should it be part of the:

- `device.connect()` options?
- Or configured on the `package` level?

kenjdavidson, 2021-02-14:
With regards to your specific issue, I think it definitely needs to be resolved, and at this point the resolution should be to ensure that `onHostPause` and `onHostDestroy` cause all the connections to be disconnected.  Ensuring that all the devices are aware of the `BluetoothModule` not being available.

This would enabled you to re-connect on startup from React Native.

Once this is done, I think the next step is deciding which of the above makes the most sense.  Feel free to submit a PR for the issue at hand and then I'll open a new issue discussing the future of how the Application lifecycle works.

andreibursuc97, 2021-02-14:
Just out of curiosity, what kind of problems would you think the static map raises? Just asking because I've did something similar in another library and I would like to asses if this could give me headaches there, as I am already using it. Also, I will submit a PR when I get the opportunity. Thank you.

kenjdavidson, 2021-02-14:
I'm not an expert on the android classloader lifecyclez so I may be wrong. 

Generally using static to get around design questions isn't a good solution (in my opinion).  Its just the way I've been conditioned. 

 I'm not sure how long the class is loaded for? When does it get reloaded? That's how long that map stays around now.  In a web app it wouldn't be until the context is stopped.  I've seen static variables cause more problems than they solve. 

As I mentioned using the built in features

- services 
- Bundle states
- etc

I also need to keep functionality consistent across ios as well. I know that ios has lifecycle events but I'm very unsure how static is thought of in swift.

kenjdavidson, 2021-02-14:
What happens when the device disconnects. And the modules isn't running to remove the connection? It won't get garbage collected.

Now you need to check connection on startup and clear them out? Seems to be the same issue as restarting the connection automatically just with more issues. 

A background service could be shut down and cleared when Android needed too, unlike the static map/connection.

andreibursuc97, 2021-02-15:
Ok tried the fix but apparently the problem is that `onHostDestroy` is not called at app reload, issue here https://github.com/facebook/react-native/issues/9773, I guess the only solution would be to disconnect at component unmount

kenjdavidson, 2021-02-15:
That's really odd, that task says that the on Host Destroyed isn't called because the activity isn't stopped. If the activity isn't stopped then the package/module shouldn't be recreated and this shouldn't be an issue.

There is a comment in there about no way to notify native modules about the reload. So I'm confused as to why you're seeing the module get recreated of they've effectively said it shouldn't be shut down in the first place.  I'd love to see some more of you're lifecycle testing with regards to your app and device. 

Regardless. This is still an issue for production. On destroy should definitely close off all the connections. I'm curious, when you close the app does that effectively kill the connection? 

What about putting the app in the background (pause) and bringing it back.  You're changes still may be valid to resolve that problem. 

Feel free to use the static version, itll work. But I don't think it's the right solution for me to pull in.

Once I get a bunch of free time I'll have to look into services on ios. I don't think they exist, or exist in a limited fashion. So I'll have to determine how that will work across both.

kenjdavidson, 2021-02-15:
I'm curious if an easy way to manage it is to provide the module to the application / activity when it's created. So that anyone can jack into the module logic from the native side.

This way nothing app wise changes on ios but would allow the state (connected device address) to be saved and restored across pauses. 

They didn't make it easy to work with though, probably because like 99% of the time apps don't need to to be managed at such a low level.

andreibursuc97, 2021-02-15:
Yes, closing the app actually closes the connection, putting the app in background and bringing it back does not close the connection. Turning off the bluetooth device while app is running in background actually removes the connection from the map, not sure how this works but it does.

kenjdavidson, 2021-02-15:
That's the `ACL_DISCONNECT` event being listened to by the `AclBroadcastReceiver` (I think it's called).

kenjdavidson, 2021-03-03:
Closing this since it seems it's only a development issue and there are comments about it being a problem with React in general.  Will continue along the Service implementation using the other open task.

closed 2021-03-03 by kenjdavidson