kenjdavidson/react-native-bluetooth-classic #90: onDeviceRead listener never call.

NoumanSakhawat, 2021-03-02:
**Mobile Device Environment**
Provide a list of operating systems on which this issue is relevant.
- Device: [e.g. Android]
- OS: [e.g. API29, Android 10]

**Application Environment**
Provide information about your development environment:
- React Native version: [e.g. v0.63.4]
- RN Bluetooth Classic version: [e.g. v1.60.0-rc.12]

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Scan device
2. connect any device 
3. RSSI will be undefined.
4. onDeviceConnected and onDeviceRead listeners never call

kenjdavidson, 2021-03-02:
Going to need a bunch more information than that:
- Sample code
- Full descriptions of what you're doing
- Etc.

With regards to your comments:

1. Scan device, I assume that was from `RNBluetoothClassic.startDiscovery()`??  When you do this, RSSI should be available in the Extras field (or maybe on it's own, I forget).
2. Connect to device - how? Did you pair after a scan? Or did you get this from an already bonded device? 

Note - Already bonded devices don't have RSSI values, you need to discover them.  So if you want to have RSSI values for bonded devices you need to:
- Get a bonded devices
- Start a discovery
- Merge the two lists together to get a combined bonded + rssi 

3) Depends where you're looking for RSSI.
4) `onDeviceConnected` and `onDeviceRead` are definitely working, I use them thoroughly.  So you'll need way more info.

All in all, you need to provide substantially more information.

NoumanSakhawat, 2021-03-03:
@kenjdavidson can you please share the sample code that how to connect a bonded device and as well as the device which is not bond?

kenjdavidson, 2021-03-03:
It's all in the documentation of the Bluetooth example app in the other repository. You're doing too many connections, and generally reconnecting will drop the connection.

Essentially the flow is

- getBondedDevices()
- device.connect(). Once this promise returns you don't need to check is Connected. 

And you're either good or not. 

Fo unbounded you need to pair it first. Either with the OS settings or the app (which honestly I don't use and haven't tested much, it's a remnent of the original lib) 

Also, what are you doing in setup Connected?

NoumanSakhawat, 2021-03-03:
`let connectToDevice = await RNBluetoothClassic.connectToDevice(DEVICE_ADDRESS);`
after executing this line connectToDevice RSSI value is undefined.

And in setupConnected function, we have our business logic.

kenjdavidson, 2021-03-03:
Like I said. Rssi is only available after a device is scanned or discovered. So if you want to get the Rssi you need to

- discover devices first
- then chekc if connected (isbonded should be available too)
- then connect

Or.

- connect first
- then dicover
- then listen for onDeviceDiscovered, filter for that device
- merge rssi

This is an android thing. Not a library thing. It's just how it works.

NoumanSakhawat, 2021-03-03:
> Like I said. Rssi is only available after a device is scanned or discovered. So if you want to get the Rssi you need to
>
> * discover devices first
> * then chekc if connected (isbonded should be [...]

@kenjdavidson did you check Steps to reproduce the behavior

NoumanSakhawat, 2021-03-03:
simple is that can we get RSSI value any moment after the device connected? @kenjdavidson

kenjdavidson, 2021-03-03:
Yes. The example app has the following finctionality

- discovery where rssi is available 
- connecting
- listing bonded

All of which you need to use. 

With regards to discovery the app only listens for all devices discovered and merges the list. But I have put in the ability to receive single discovery devices. 

Feel free to let me know if it doesn't work. But provide details. Like exact code and maybe some debugging. I don't have a lot of time these days so I can't do it from scratch.

kenjdavidson, 2021-03-03:
> simple is that can we get RSSI value any moment after the device connected? @kenjdavidson

That's great. But that's not how Android works. Rssi is only available after discovery. That's in the android docs man. You need to do some research as well. 

There is no rssi available if you just get paired devices or connect. 

And its not wise or common enough use case for me to add the discover merge logic on connect or listing bonded. When it's easy enough for you to do yourself and no one else has asked for it.

kenjdavidson, 2021-03-03:
> @kenjdavidson did you check Steps to reproduce the behavior

1. Scan Device
I have absolutely no idea what you mean by this, since you've provided no context, but if you mean have I tested `RNBluetoothClassic.startDiscovery()` then yes, I have tested it and it works.

https://github.com/kenjdavidson/react-native-bluetooth-classic-apps/blob/1cba89aaa1181998dbcab7a9549a88a9bb5071a1/BluetoothClassicExample/src/device-list/DeviceListScreen.js#L159

2. Connect any Device
Yes, I've connected to multiple devices:
- Barcode scanners
- Other phones/tablets

3. RSSI will be undefined
This isn't actually an issue in terms of something I can replicate, but the RSSI is definitely available after you perform a discovery.

https://github.com/kenjdavidson/react-native-bluetooth-classic-apps/blob/1cba89aaa1181998dbcab7a9549a88a9bb5071a1/BluetoothClassicExample/src/device-list/DeviceListScreen.js#L174

This section of code is terrible and doesn't do what I want it to do, but if you debug that line you'll see that `device.rssi` or `device.extras.rssi` is available.

4. onConnectedDevice and onDeviceRead are not called
As I mentioned `onConnectedDevice` will never be called, I've removed it from the native side, I just haven't had a chance to remove it from the React side.  There was too much confusion around what a `connected device` was to make it worth while for me to maintain.

`onDeviceRead` is definitely being called when data comes in and you have a listener attached, this is the only method in which I received data from my scanners:

https://github.com/kenjdavidson/react-native-bluetooth-classic-apps/blob/1cba89aaa1181998dbcab7a9549a88a9bb5071a1/BluetoothClassicExample/src/connection/ConnectionScreen.js#L134

You can see in this function I either setup polling or read listener based on the parameter/state defined.

Like I said, you'll need to augment all of these things to get them working:

```js
// This gets me a list of bonded/paired devices (currently no RSSI)
const bonded = await RNBluetoothClassic.getBondedDevices();
setBondedDevices(bonded);  // Set state

// Now start discovery
const discovered = await RNBluetoothClassic.startDiscovery();

// Now update the state with the discovered version of the device.
// this version will have RSSI and a bunch of other "extras" available
const bondedRSSI = discovered.filter(d => bonded.find(b => b.address === d.address));
setBondedDevices(discovered)

// Now you can connect to this device
// Or you could connect first, and do the discovery after, it really doesn't matter as all you're doing is managing BluetoothDevices by address.
```

Anyhow, unless you provide more information and specific examples I'm not going to worry about this, what you've explained you need is all doable.

closed 2021-03-03 by kenjdavidson