kenjdavidson/react-native-bluetooth-classic #95: writeToDevice message string does not use "encoding" param

philpettican, 2021-03-25:
**Mobile Device Environment**
Provide a list of operating systems on which this issue is relevant.
- Device: Android

**Application Environment**
Provide information about your development environment:
- React Native version: 0.61.2
- RN Bluetooth Classic version: 1.60.0-rc.13

**Describe the bug**
According to the docs

```
The `message` can be provided
     * as either:
     * - A String which will be encoded (using the optional `encoding`)
     * - A Buffer
```

However. calling RNBluetoothClassic.writeToDevice(address, message, 'hex') ignores the encoding param.

**To Reproduce**
Steps to reproduce the behaviour:

```js
import RNBluetoothClassic from 'react-native-bluetooth-classic';

RNBluetoothClassic.writeToDevice(address, message, 'hex');
```

**Expected behaviour**
The message string should be encoded using the "encoded" param.

kenjdavidson, 2021-03-25:
Could you provide a little more information on what you're sending in:

```
// writeToDevice
    let data = Buffer.isBuffer(message)
      ? (message as Buffer)
      : Buffer.from(message, encoding);
    return this._nativeModule.writeToDevice(address, data.toString("base64"));
```

If you're message is already a buffer, it will just be used as is, and will not be encoded.  If you're message is not `Buffer.isBuffer()` then you're saying that `Buffer.from(message,'hex')` doesn't work?  

There are a large number of issues/comments regarding sending HEX data to devices, have you gone though those?

philpettican, 2021-03-25:
Thanks for the quick response.

Apologies but I think I must be tired. I was looking at the code thinking it was:

```
let data = !Buffer.isBuffer(message)
            ? message
            : Buffer.from(message, encoding);
```

I'm trying to send the following:

`
RNBluetoothClassic.writeToDevice(address, '02000306FFF703', 'hex');
`

02000306FFF703 is supposed to be an acknowledgement. When the scanner sends a barcode, it will try to send the data 3 times unless an acknowledgement is sent.

The issue I'm seeing is that when I have more than one onDataReceived listener in the app (2 different screens for example), the scanner seems to ignore the acknowledgement and tries to send the data 3 times. The issue doesn't occur if I only have one subscription to onDataReceived in the app.

I'll close this issue as it could be an issue with the scanner.

closed 2021-03-25 by philpettican

kenjdavidson, 2021-03-25:
Haha I feel ya man. It's been a crazy couple weeks myself.