kenjdavidson/react-native-bluetooth-classic #97: onDataReceived native module listener not removed when calling .remove()

philpettican, 2021-03-31:
**Mobile Device Environment**
Provide a list of operating systems on which this issue is relevant.
- Device: Android

**Application Environment**
Provide information about your development environment:
- React Native version: 0.61.2
- RN Bluetooth Classic version: 1.60.0-rc.13

**Describe the bug**

After subscribing to onDataReceived and then removing the subscription, if I subscribe again to onDataReceived, onDataReceived is called twice. 

**To Reproduce**
```js
let subscription = this.device.onDataReceived(this.onReceivedData);
subscription.remove();
subscription = this.device.onDataReceived(this.onReceivedData);
// now trigger a BluetoothDeviceReadEvent e.g. a barcode scan
// this.onReceivedData is called twice
```

**Expected behaviour**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
Add any other context about the problem here.

philpettican, 2021-03-31:
// BluetoothModule.js

```js
onDeviceRead(address, listener) {
        let eventType = `${BluetoothEventType.DEVICE_READ}@${address}`;
        this._nativeModule.addListener(eventType);
        let subscription = this._eventEmitter.addListener(eventType, listener);
        return {
            remove() {
                this._nativeModule.removeListener(eventType); //  <-- remove native module event listener as well
                subscription.remove();
            },
        };
    }
```

kenjdavidson, 2021-04-27:
Merged (manually, sorry I was doing something else and just figured I'd make the change locally) and released `1.60.x-rc14`.

closed 2021-04-27 by kenjdavidson