kenjdavidson/react-native-bluetooth-classic #100: Missing data while utf-8 transmission

maslovay, 2021-04-07:
I'm trying to send the jpg file via BTSerial on ESP-32. 
It works fine on python scripts and Serial Bluetooth Terminal app for android. But with react-native-bluetooth-classic it lose some data in each chunck. I tried to use both 
```
 this.state.btDevice.onDataReceived((data) => {this.onDataReceived(data.data)})
 this.state.btDevice.read();
```
But every of them cut the the chunks. For example first chunks is 
���� on react-native-bluetooth-classic
\xff\xd8\xff\xe0\x00\x10JFIF\x00\x01\x01\x01\x00\x00\x00\x00\x00\x00\xff\xdb\x00C\x00\ in python

connection options are
`      connectionOptions: {
        CONNECTOR_TYPE: "rfcomm",
        DELIMITER: '\r',
        READ_SIZE: 4096,
        DEVICE_CHARSET: "utf-8",
      },
`
please help

kenjdavidson, 2021-04-07:
With regards to the code you posted, you provided both

```js
 this.state.btDevice.onDataReceived((data) => {this.onDataReceived(data.data)});  // Sets an OnRead event
 this.state.btDevice.read(); Manual Read
```

You should not be using both of these at the same time, it should be an **either or**.  So if you're doing both, I can see why you would be missing data.   

Secondly, if you're sending a JPG file, you don't want to be sending UTF8, you want to be sending Binary data, no?  You'll need to implement your own Binary data connection for parsing the information correctly.  Or you'll have to write your own encoding/decoding to map it.

Anyhow, need a bunch more information about what you're actually doing (more than just 2 lines).

I have a couple issues I need to look at this weekend, if you can provide more details, I will try to look at this one in more detail as well.

maslovay, 2021-04-07:
Thank you for fast response. I tired to use two methods, each separate, not both in one time.

Yes, I'm sending the JPG file. I think its byte array, but in Serial Bluetooth Terminal app utf-8 was chosen. That's why it make me think that that is utf-8. 

How can I implement my Binary data connection for parsing the information correctly with react-native-bluetooth-classic

Thank you a lot for help

kenjdavidson, 2021-04-07:
I know that **Serial Bluetooth** has a charset setting, but I'm unsure about how that's actually applied to the connection.  I've seen the repository for that project on Github but haven't actually looked into the source code. 

By default **Serial Bluetooth** receive is set to `CR-LF` I believe, but you only have the delimiter set to `LF` is it possible that part of the Image is matching the `/r` and therefore the first chunk is being received?  Or have you updated **Serial Bluetooth** to match the delimiter?

There are a bunch of issues and documentation showing the starting point for dealing with Binary data.  Since it was really an after thought of both this (and the original library) I really haven't put much time into worrying about it, but it does work as seen from others.

If your peripheral is sending UTF8 encoded image data, then you should be fine accepting it, you'll just need to match up the delimiters.

I suggest you do some debugging of the Android code and provide your findings.  I can help with suggestions from there.

maslovay, 2021-04-07:
I tried different delimiters (\r, \n, \r\n, null, empty), the result was the same - some data was not received.

I will try to deal with Java code to understand how to fix it.

kenjdavidson, 2021-04-07:
Sounds good. 

There isn't much to it.

- Data is read
- Data is added to buffer
- Buffer is decoded based on charset
- If there is an OnRead listener it will look for a delimiter and send the content of the message.  Then return the remaining data to the Buffer
- If there is no OnRead listener, data just builds up in the buffer until either a Read or an OnRead listener is added.

So there are limited spots where the issue could be occurring:

- When the data comes in do you get all the bytes?
- When it's decoded from the buffer and looked for the Delimiter is it found?
- When the message is sent to React Native side, what format is it in?

Feel free to keep me posted.

kenjdavidson, 2021-04-16:
Hey, any luck with debugging the Java side to see if anything is apparent?

maslovay, 2021-04-16:
0

maslovay, 2021-04-16:
Eah, it works well with Java code!

kenjdavidson, 2021-04-16:
So we've got no idea where things are going wrong?

maslovay, 2021-04-16:
Unfortunately I can't say what is wrong with react-native-bluetooth-classic, I will be happy to use it ir there will be no losses. I just took the Bluetooth serial android library and it works. https://developer.android.com/reference/android/bluetooth/BluetoothSocket

kenjdavidson, 2021-04-18:
Right on. Sorry it wasn't working for you, but glad your eup and running.

closed 2021-04-18 by kenjdavidson

maslovay, 2021-04-18:
Thank you! Good luck!