kenjdavidson/react-native-bluetooth-classic #101: read failed, socket might closed or timeout, read ret: -1

LinAction, 2021-04-16:
Error: java.io.IOException: read failed, socket might closed or timeout, read ret: -1
    at Object.promiseMethodWrapper [as connectToDevice] (D:\Project\hshdablegather\node_modules\react-native\Libraries\BatchedBridge\NativeModules.js:103)
    at BluetoothModule._callee3$ (D:\Project\hshdablegather\node_modules\react-native-bluetooth-classic\lib\BluetoothModule.js:129)
    at tryCatch (D:\Project\hshdablegather\node_modules\regenerator-runtime\runtime.js:45)
    at Generator.invoke [as _invoke] (D:\Project\hshdablegather\node_modules\regenerator-runtime\runtime.js:274)
    at Generator.prototype.<computed> [as next] (D:\Project\hshdablegather\node_modules\regenerator-runtime\runtime.js:97)
    at D:\Project\hshdablegather\node_modules\react-native-bluetooth-classic\lib\BluetoothModule.js:7
    at tryCallTwo (D:\Project\hshdablegather\node_modules\promise\setimmediate\core.js:45)
    at doResolve (D:\Project\hshdablegather\node_modules\promise\setimmediate\core.js:200)
    at new Promise (D:\Project\hshdablegather\node_modules\promise\setimmediate\core.js:66)
    at __awaiter (D:\Project\hshdablegather\node_modules\react-native-bluetooth-classic\lib\BluetoothModule.js:3)

kenjdavidson, 2021-04-16:
Your troubleshooting and issue opening skills are on point.
Re-open this when you actually provide worth while information.

closed 2021-04-16 by kenjdavidson

woodrufs, 2021-06-18:
I was getting this same error on an Android 9 and Android 10 device. Sounds silly, but if you come across this make sure the bluetooth device you're attempting to connect to did not turn off.

kenjdavidson, 2021-06-18:
haha.. I hope we can all laugh about it.  Because I know we've all done it @woodrufs at one point in time.

thgh, 2021-12-29:
Reproduction
* `expo run:android`
* Press "a" in terminal (Open app on Android device)
* Connect to a bluetooth device in app on Android
* Press "r" in terminal (Reload app)
* Connect to the same bluetooth device in app on Android

Expected: connects to the device
Actual: `[Error: java.io.IOException: read failed, socket might closed or timeout, read ret: -1]`

It seems like the "Reload app" command does not cleanup connection resources. I see on my BT device that it stays connected after the app reloaded.

kenjdavidson, 2021-12-29:
Yup, that's an issue.  This type of functionality has been an issue (and is well documented in other native modules) when using local development and performing reloads of the app.

I haven't spent (nor plan to spend) much time looking into it as it doesn't seem to pose an issue when you're in a live/installed environment.

If you can find any information on the React Native side on whether they've added development only callbacks for the app hot/reloading we can definitely look into implementing them.

kenjdavidson, 2021-12-29:
As long as the Android/IOS side of things aren't shut down during a reload then the device should still be connected and active.   For example, instead of just:

- App starts
- Attempt to connect

you can

- App starts
- See if device is already connected
- If so, then just `getConnectedDevice`
- If not, then just connect again

From what I remember the app reload is solely a "recompile" of JS and doesn't actually affect the native application on either platform (I may be way wrong now).   Obviously in production it's an extra call that would never be true, but if it helps you during development.

One of the other differences is the device itself, the devices that my company uses will automatically disconnect if a re-connect is attempted.  So for example:

- Connect
- Do stuff
- Reload app
- Connect
* Disconnects already connected device
* Fails connection
- Connect 
- Do stuff

It's not optimal, and not something that we developed ourselves, but was useful in not having this issue to worry about in most cases.

thgh, 2021-12-31:
I tried to use `getConnectedDevices()` on app start, but that gave an empty array.

The workaround is to disconnect before reload and when forgetting to disconnect, kill the app and start over. So it's really a blocking issue.

kenjdavidson, 2021-12-31:
On app start yes, it's empty.

But I thought you said this was only during development reload? At which time the native app is not restarted, only the react Bridge.

I'm glad it's not blocking. Annoying yes. But I still don't see any dev/reload life cycle hooks to plug into. I remmeber in pause an Don resume not being called, maybe that's changed?

If you sort it out, feel free to submit a pr.