kenjdavidson/react-native-bluetooth-classic #115: [Help] Cannot read from device, but can write to it

hemanth-kumarv, 2021-05-19:
**Mobile Device Environment**
Provide a list of operating systems on which this issue is relevant.
- Device: Android
- OS: Android 11

**Application Environment**
Provide information about your development environment:
- React Native version: v0.63.4
- RN Bluetooth Classic version: 1.60.0-rc.16

**Describe the bug**
I have a Raspberry Pi 3, on which a bluetooth server (Bluedot btcomm) is running. I can send and receive data perfectly from the Android Serial Bluetooth Terminal application. But with my React Native app, I can only send data. I cannot read data from the device. My code is shown below:

```
  let status = await RNBluetoothClassic.requestBluetoothEnabled();
  if (status) {
    try {
      let unpaired = await RNBluetoothClassic.startDiscovery();
      let rpi = unpaired.find((o) => o.name === "raspberrypi");
      if (rpi) {
        let conn = await RNBluetoothClassic.getBondedDevices();
        let connStatus = conn.find((o) => o.name === "raspberrypi");
        if (!connStatus) {
          console.log("Pairing....");
          await RNBluetoothClassic.pairDevice(rpi.id);
        }
        try {
          device =  await RNBluetoothClassic.connectToDevice(rpi.id, {});
          await device.connect({});
          await device.write("Foo", "ascii");  // 'Foo' is received in raspberry pi
          console.log(await device.available());  // Logs '0'
          await device.onDataReceived((res) => console.log(res));  // Does not log anything
          let readData = await device.read();
          console.log(readData);  // Logs 'null'
        } catch (e) {
          console.log(e);
        }
      } else {
        console.log("Bluetooth error!");
      }
    } catch (e) {
      console.log(e);
    }
  }
```

Please let me know if I am doing anything incorrectly. 

Thank you.

kenjdavidson, 2021-05-19:
Looking at your code:

- Status looks good
- Check for nearby devices looks good 
- Bonded devices looks good
- Pairing request looks good
- connect to device looks good (using standard options)
- Writing to device doesn't really need "ASCII" since that's the default but obviously fine
- `device.onDataReceived` does not return a promise, it returns a `BluetoothEventSubscription` which is used to remove the subscription when you're done.  So I'm not sure what awaiting that will do
- `device.read()` should not be used at the same time as `onDataReceived` as they could cause problems for each other.  I believe I've documented it numerous times to only use one method.

I'm not sure what `await device.onDataReceived()` will do, but it should be `let subscription = device.onDataReceived()` so that you can eventually call `subscription.remove()` on unload.

I also don't know how your Bluetooth Terminal or your device is configured.  The settings you're using are:
- ascii
- \n delimiter

So if your device is actually sending \r (which I believe is the default for bluetooth terminal) you will need to change that.

`let connection = device.connect({ DELIMITER: '\r' })`

or something along those lines.

hemanth-kumarv, 2021-05-20:
Thank you for the quick response and detailed reply.

> `device.read()` should not be used at the same time as `onDataReceived` as they could cause problems for each other. I believe I've documented it numerous times to only use one method.

I tried all possibilities, just `device.read()`, just `device.onDataReceived()` and both. None of them worked.

>So if your device is actually sending \r (which I believe is the default for bluetooth terminal) you will need to change that.
`let connection = device.connect({ DELIMITER: '\r' })`
or something along those lines.

This did not work either. But I went through the Standard Options list and passed `{ SECURE_SOCKET: false }` to the `connect()` function. This worked and I am receiving data now. 

Thank you very much for taking the time to solve this problem of mine.

P.S. I was also wondering if it is possible to enable Bluetooth Tethering with this package. Please let me know if it can be done, as I cannot find any docs related to Bluetooth Tethering.

kenjdavidson, 2021-05-20:
Probably not.  Tethering would require background communication which this library doesn't support yet.  React Native uses HeadlessJS instead of actual Background Services which I've never actually looked into.

You can connect from Android to Android without issue, so you may be able to hack it if it's a small amount of networking, but It seems unlikely.

closed 2021-05-20 by kenjdavidson

kenjdavidson, 2021-05-20:
Hey, just a thought as well to make the connection a little quicker.  Instead of

```
let unpaired = await RNBluetoothClassic.startDiscovery();
      let rpi = unpaired.find((o) => o.name === "raspberrypi");
```

You could take advantage of the `onDeviceDiscovered` and `BluetoothEvent.DEVICE_DISCOVERED` event so stop discovery early as soon as you get the device for which you're expecting.

deepalisrivastava25, 2022-02-09:
@hemanth-kumarv @kenjdavidson I have implemented this library in my app to establish bluetooth connection with another device.(which is an android phone in my case)
However I am unable to connect to another device.
In the above code snippet shared by @hemanth-kumarv , 
          device =  await RNBluetoothClassic.connectToDevice(rpi.id, {});
          await device.connect({});

connectToDevice accepts 2 params which are address and options. 

This is the definition of this method in the react-native-bluetooth-classic lib.

    connectToDevice(address, options) {
        return __awaiter(this, void 0, void 0, function* () {
            **// Comming from the Java world this is nuts - not being able to assign anything to
            // options because it's a <T extends StandardOptions> even with something that matches
            // the StandardOptions interface**
            let connected = yield this._nativeModule.connectToDevice(address, options || {});
            return new BluetoothDevice(connected, this);
        });
    }

What do we mean by the bold comment highlighted above.? What should I pass as **options** in connectToDevice method.

Can you share a proper example of how can we connect to a device?

deepalisrivastava25, 2022-02-10:
@kenjdavidson thank you for a quick reply.

I have attempted to connect like this :
await RNBluetoothClassic.connectToDevice(address);

When this line executes , I get a timeout error soon after :
Error: java.io.IOException: read failed, socket might closed or timeout, read ret: -1
    at Object.promiseMethodWrapper [as connectToDevice] (C:\BT-bugs\commandernext-ui\node_modules\react-native\Libraries\BatchedBridge\NativeModules.js:106:51)
    at BluetoothModule._callee3$ (C:\BT-bugs\commandernext-ui\node_modules\react-native-bluetooth-classic\lib\BluetoothModule.js:139:54)
    at tryCatch (C:\BT-bugs\commandernext-ui\node_modules\regenerator-runtime\runtime.js:63:40)
    at Generator.invoke [as _invoke] (C:\BT-bugs\commandernext-ui\node_modules\regenerator-runtime\runtime.js:294:22)
    at Generator.next (C:\BT-bugs\commandernext-ui\node_modules\regenerator-runtime\runtime.js:119:21)
    at C:\BT-bugs\commandernext-ui\node_modules\react-native-bluetooth-classic\lib\BluetoothModule.js:7:71
    at tryCallTwo (C:\BT-bugs\commandernext-ui\node_modules\promise\setimmediate\core.js:45:5)
    at doResolve (C:\BT-bugs\commandernext-ui\node_modules\promise\setimmediate\core.js:200:13)
    at new Promise (C:\BT-bugs\commandernext-ui\node_modules\promise\setimmediate\core.js:66:3)
    at __awaiter (C:\BT-bugs\commandernext-ui\node_modules\react-native-bluetooth-classic\lib\BluetoothModule.js:3:12)

I was trying to check if my device is still connected like this :
await RNBluetoothClassic.isDeviceConnected(address);

This returns me false. 

This is a link with similar problem. 
https://stackoverflow.com/questions/67904223/react-native-bluetooth-connection-failed-java-io-ioexception-read-failed-socke

You suggested to use accept() before connectToDevice. The problem is accept() method does not work the way you have suggested in this link. 

Why am I getting this timeout error ? Please help.

kenjdavidson, 2022-02-10:
Not sure what to tell ya man, the error is right there

```
Error: java.io.IOException: read failed, socket might closed or timeout, read ret: -1
```

The socket on the other side (`ServerSocket`) is:
- closed
- timeout
- etc

as I mentioned before, in order to make a connection, you need something waiting on the other side to accept this connection.  

- How are you doing that?
- You say that `accept` doesn't work, but how doesn't it work? you have provided no errors.

You also need to have an understanding of the language used:

- Paired = Paired
- Connected = RFCOMM socket open

You're not connected (ie. the error above) so you'd never get the device back from `isDeviceConnected()`.  I can tell you that the `accept` method works just fine, it's being used in the sample/development application and I've used it to communicate between two Android devices.