kenjdavidson/react-native-bluetooth-classic #117: How to start discovery devices in IOS?

HudsonJunior, 2021-06-15:
I noticed that the method `startDiscovery` is available only for Android. Is there any other function in IOS that do the same thing? Or in IOS I only can search for paired/connected devices?

kenjdavidson, 2021-06-15:
Negative, the MFi protocol does not support the scanning feature.

kenjdavidson, 2021-06-15:
Sorry, closed by accident.  I'll leave open for further discussion for a bit if you want to look into it.

kenjdavidson, 2021-06-15:
Here is a sample project showing how to use Core Bluetooth with 'Classic' although I don't believe this covers RFCOMM which this library is primarily for.  The library already uses Core Bluetooth to determine whether bluetooth is enabled, so I'm not against pulling some features in that make sense - but there are super strict requirements to the MFi protocol that I'd prefer to stick to - ie I don't want this becoming a BLE library by chance (although it's more than likely using manual implementation of the Module and connections.

https://developer.apple.com/documentation/corebluetooth/using_core_bluetooth_classic

HudsonJunior, 2021-06-16:
Ok. Thanks for the feedback.

kenjdavidson, 2022-09-26:
It looks like `CoreBluetooth` still doesn't support this.  At this point I'm going to leave it as a known issue between Android and IOS.  IOS controls a number of things internally which we just don't have access to using the `EAAccessory` library.

closed 2022-09-26 by kenjdavidson