kenjdavidson/react-native-bluetooth-classic #118: Doesn't work readFromDevice

Luanfv, 2021-07-20:
I'm writing code where I need to pass a request and receive a response, but when I try to use the readFromDevice function it returns null.

I've already replicated this functionality in an app called "serial bluetooth" and it worked, which suggests that either the read isn't working or I'm doing something wrong. Can someone help me? please.

"react-native": "0.64.0"
"react-native-bluetooth-classic": "^1.60.0-rc.16"

[image]

kenjdavidson, 2021-07-20:
How are you connecting to the device? What are you setting with regards to:
- Encoding
- Delimiter
- Etc.

I'm guessing that you're not setting the same parameters during connection that you're using with `Serial Bluetooth Terminal`.   I use `Serial Bluetooth Terminal` to test Android to Android communication, so it should definitely work as long as you've setup your communication protocols the same.

For example:

Serial Bluetooth terminal defaults to \r while this defaults to \n - without this happening, chances are the buffer is building up.  I would suggest that you actually run `#isAvailable()` before attempting to read to ensure you actually have messages.

I would also suggest attempting to debug the Java code, it should be available to you.

Luanfv, 2021-07-20:
I'm connecting in the standard way, via the `RNBluetoothClassic.connectToDevice(id)` . But I'm not setting any parameters in this connection... How could I do that?

Whenever I run the command `RNBluetoothClassic.availableFromDevice(id)` it returns 0.

Luanfv, 2021-07-20:
I started a connection with the delimiter being \r, but the problem continues.

[image]

kenjdavidson, 2021-07-20:
I'm not sure man, everything you're doing looks good; but I'm still fairly confident it's a data/configuration issue.  In order for me to help more I'm going to need you to provide a lot more debugging information:

- Give me a little higher level of what you're actually doing?  Devices? Sample Messages? Etc.
- Have you tried to debug the Java implementations of `availableFromDevice` to see if it's delimiting the messages properly?
- Have you tired to debug the Java implementations of `DelimitedStringDeviceConnectionImpl` to determine what is being added to the buffer?
- You're not actively listening for the READ event as well as manually requesting data?

Without, at a minimum this information, there isn't much I can do to help.

Maybe take a look through some of the closed issues with regards to reading from a device and see if any of them are similar to your issue?

Luanfv, 2021-07-20:
I'm developing an application that will communicate with a Raspberry Pi 3, from which I send a string in JSON format to the Raspberry that will list the nearby wifis. I'm just calling a `readFromDevice` function, I'm not calling `onDataReceived`.

I don't know how to use Java, so unfortunately I can't answer the two Java-related questions.

Note: I found an issue similar to what I'm going through, but I couldn't solve it, even following the steps.
https://github.com/kenjdavidson/react-native-bluetooth-classic/issues/115

Ps: This image is how the code looks after the changes. Still returns 0
[image]

kenjdavidson, 2021-07-21:
Unless you're willing to give me access to your project (and maybe I can find some way to simulate the data that you're sending) there isn't much I can do to help.

It could be time to start learning Java.

Luanfv, 2021-07-27:
I managed to solve my problem through this issue https://github.com/kenjdavidson/react-native-bluetooth-classic/issues/120

closed 2021-07-27 by Luanfv

kenjdavidson, 2021-07-27:
Good to hear!

Curious why both of you using Pis have issues with /r and /n within the application though.  I'm personally using both of those based on the two types of microchip scanner that my company uses.  But I'm glad you sorted it out!

Luanfv, 2021-08-12:
Hi, after a while with the same question I found out from the engineering team that the problem was actually in the python code that didn't have a delimiter defined.

ELBEQQAL94, 2021-12-15:
I have the same issue, readFromDevice always returns null. Any help, please.

kenjdavidson, 2021-12-15:
> I have the same issue, readFromDevice always returns null. Any help, please.

I suggest opening a new ticket with substantially more information of you are having an issue.  Expecting help with one sentence and no details is essentially pointless.

ELBEQQAL94, 2021-12-20:
Hi @kenjdavidson thanks for your response,

I changed my code from readFromDevice to onRecievedData now it's working!!

`RNBluetoothClassic.connectToDevice(deviceId, { delimiter: ''})`

`const event = device.onDataReceived(data => console.log(data)` // data recieved

If i removed delimiter from `connectToDevice` i got `null`

Thanks

ELBEQQAL94, 2021-12-20:
If you have any extra questions please let me know.

Thanks

kenjdavidson, 2021-12-20:
This looks like it's working as expected.  The default delimiter is '\n' and it looks like you've changed that to ''.

With the latest change it means that an `empty` delimiter will ALWAYS send data, ie. It will NOT use a delimiter.  I really don't know how I feel about a `delimited` connection ignoring delimitation, but it seems to be working for some of you guys.  so I'll leave it in there since it's easier than adding a whole new connection type.

If you remove the delimiter it's reverting back to '\n' and it would seem that you're not using `\n` as a delimiter.  So again, you need to:

- Fully understand how you're sending data
- Fully understand the API  with which you're reading
- connect to the device using the appropriate delimiter

From what I can see, this is working as it's supposed to.