kenjdavidson/react-native-bluetooth-classic #120: Can't read from the device but I can write to it

slim-abid, 2021-07-25:
Hello,
I'm connecting my react native app to MC60 MCU using react-native-bluetooth-classic,
after connecting succesully with 
myDevice.connect({CONNECTOR_TYPE: "rfcomm",
            DELIMITER: "\r",
            DEVICE_CHARSET: Platform.OS === "ios" ? 1536 : "utf-8",
            SECURE_SOCKET: false });
I tried to send data with myDevice.send(data) this step is done successfully but when trying to read data with myDevice.read() I'm getting null everytime.

slim-abid, 2021-07-26:
The problem was the delimiter,
I changed it to a customized one (using any alphabetical character instead of \n or \r)
the problem was that when dealing with AT-commands to send data to the app the terminal cuts the string into the given number of characters without adding a delimiter to the string

closed 2021-07-26 by slim-abid

kenjdavidson, 2023-06-05:
This is a `hack` around just sending the constant stream of data as it comes in.  This is good if you can guarantee that each message is a full message, and you don't actually need to chunk them together (which is probably most cases to be honest).   You don't want to use this if you are expecting numerous messages which get applied and handled together.

mmestiyak, 2023-06-05:
@kenjdavidson thank you so much for writing,  I'm keeping it in the state, and merging the prev state with current data that's coming from the device but could you tell me that if it's numerous data what could be the best practices to follow

kenjdavidson, 2023-06-05:
I'd need to see examples of what your data is in order to give a realistic answer.  Essentially, the standard connection is the `DelimitedStringConnectionImpl` which as you see in the name expects a delimited string.  So for example, if your device sends something like this:

`{ "name": "value",`
`"age": 43,`
`"gender": "male" }\n`

That's three messages which combined create a single message.  With the throughput of devices, this is pretty uncommon, which is why you'd usually get that object serialized in one message.   It comes down to where you want the messages to be concatenated:

- Within Native Java then you need to add a valid delimiter.
- Within react, then you just do things your way.

It all comes down to what your data looks like.

mmestiyak, 2023-06-05:
I'm actually building a apk generator that allows users to program with any kind of data communication between the app and the device, if there's any specific situation in which I'm stuck, surely I'll be seeking your guidance, thank you brother for being quick & instant, you are awesome  ✨

brianwachira, 2024-07-24:
@kenjdavidson,
I am sending data object that looks something like this 
`{"gps":{"lat":-1.3729792,"lng":36.9360896},"survey_id":"9c74c0be-e671-466b-ab9d-725c9003cbe3","question1":false,"question2":true}`

I stringify it and  send it between two mobile devices within my app. However I am unable to receive the data from the other device.

How I connect to device?

```
const connectToDevice = async (device: BluetoothDevice) => {
		try {
			const connected = await device.connect();
			if (connected) {
				setConnectedDevice(device);
			}
		} catch (error) {
			console.error("Failed to connect to device:", error);
		}
	};
```

How I accept connections?

```
const acceptConnections = async () => {
		try {
			const device = await RNBluetoothClassic.accept({});
			if (device) {
				setConnectedDevice(device);
			}
		} catch (error) {
			console.error("Failed to connect to device:", error);
		}
	};
```

How I subscribe to onDataReceived?
```
useEffect(() => {
		if(connectedDevice) {
			console.log('Reading')
			const subscription = connectedDevice.onDataReceived((data) => {
				console.log(data)
			})
			return () => {
				subscription?.remove();
			}
		}
	},[connectedDevice])
```

Please help, is there something I am missing?